lvgl/lvgl #4411: Use 3.5 inches IPS screen color exception

BestStark, 2023-07-26:
### Introduce the problem

Use 3.5 inches IPS screen & LVGL UI rendering results appear color difference, do not know why, need to do any special configuration?

Use LVGL 8.3
Use ESP32S3
Use Arduino
Use **3.5(inch)  TFT IPS ILI9488 & FT6236  480*320 (Pixel) screen**

### Examples and cases

[image]

### Suggested solution

/*Color depth: 1 (1 byte per pixel), 8 (RGB332), 16 (RGB565), 32 (ARGB8888)*/
#define LV_COLOR_DEPTH 16

/*Swap the 2 bytes of RGB565 color. Useful if the display has an 8-bit interface (e.g. SPI)*/
#define LV_COLOR_16_SWAP 1

**How does LVGL UI use IPS enabled screens??**

thanks...

kisvegabor, 2023-07-26:
How does it look like with `LV_COLOR_16_SWAP   0`?

BestStark, 2023-07-27:
> How does it look like with `LV_COLOR_16_SWAP 0`?

Still the same effect, no change, I would like to ask for the **IPS screen** LVGL is supported?

kisvegabor, 2023-07-27:
What is you call `lv_obj_set_style_bg_color(lv_scr_act(), lv_color_hex(0xff0000), 0)`? Do you see a red screen? If not, there is a  color format mismatch somewhere.

BestStark, 2023-07-28:
> What is you call `lv_obj_set_style_bg_color(lv_scr_act(), lv_color_hex(0xff0000), 0)`? Do you see a red screen? If not, there is a color format mismatch somewhere.

No，`lv_obj_set_style_bg_color(lv_scr_act(), lv_color_hex(0xff0000), 0)`  **screen background turned blue**

Yes, I also think it is a configuration problem, but this is SPI 4-wire TFT screen, the parameters are supported IPS wide color gamut, so LVGL need any special configuration?

thanks..

kisvegabor, 2023-07-28:
Please double check the color depth of your display (probably 16, 18, or 24 bit). If it's correct, try toggling `LV_COLOR_16_SWAP`.

BestStark, 2023-07-28:
> Please double check the color depth of your display (probably 16, 18, or 24 bit). If it's correct, try toggling `LV_COLOR_16_SWAP`.

18bit

kisvegabor, 2023-07-28:
Aaaah, LVGL can render only in 16 or 32 bit. Do you do any conversation from 16 bit to 18 bit in the `flush_cb`?

BestStark, 2023-07-28:
> Aaaah, LVGL can render only in 16 or 32 bit. Do you do any conversation from 16 bit to 18 bit in the `flush_cb`?

oh....this is sad news for me

but，this my `flush_cb `code 
```
void my_disp_flush(lv_disp_drv_t* disp_drv, const lv_area_t* area, lv_color_t* color_p) {
  uint32_t w = (area->x2 - area->x1 + 1);
  uint32_t h = (area->y2 - area->y1 + 1);

  tft.startWrite();
  tft.setAddrWindow(area->x1, area->y1, w, h);
  tft.pushColors((uint16_t*)&color_p->full, w * h, true);
  tft.endWrite();

  lv_disp_flush_ready(disp_drv);
}
```

BestStark, 2023-07-28:
> Aaaah, LVGL can render only in 16 or 32 bit. Do you do any conversation from 16 bit to 18 bit in the `flush_cb`?

oh，Is there a recommended driver or screen type for using lvgl on an MCU screen? For example TFT SPI, ILI9488 or ILI9431, touch FT6236?

kisvegabor, 2023-07-31:
Almost anything works, but the 18 bit color depth is not that standard. ILI9431 is a common and battle tested alternative.

closed 2023-08-22 by lvgl-bot