lvgl/lvgl #4412: [slider] Limit knob movement inside indicator

C47D, 2023-07-27:
### Introduce the problem
See [Slider Knob out of the track](https://forum.lvgl.io/t/slider-knob-out-of-the-track/11956) forum post for more information.

### Examples and cases
A feature to limit the knob in the slider is desired.

### Suggested solution
See this [proposal](https://forum.lvgl.io/t/slider-knob-out-of-the-track/11956/5)

kisvegabor, 2023-07-27:
Thank you for opening this issue.

So now we always put the center of the knob to the right edge of the indicator.  For this mode I think this is what we need to do:
- Let's say the indicator has max 100 px length
- The knob is 20 px wide
- In the left most position the indicator is at x=10 (knob_width/2)
- In the right most position the indicator is at x=90 (max_indicator_width - knob_width/2)
- The effective length is 90-10 = 80  -> max_indicator_width - knob_width
- So what we need to do is mapping the current value on in 0..80 range  and add 10 as offset
- The tricky part is when the slider is clicked, as not the indicator's end  but the knob's center should jump to the clicked position. So here some extra trick/offsetting needs to involved here.

> Would you like me to add a new slider mode for this feature? Should the knob size be always smaller than the indicator in this mode?

I think it's not a new mode as it applies to all the current modes. Instead it's a new property, e.g. `lv_slider_set_knob_inside(slider, true/false);` 

What do you think?

C47D, 2023-08-14:
Not stale

C47D, 2023-08-30:
Not stale, see PR for more information

kisvegabor, 2023-09-14:
The discussion continues in the PR.

closed 2023-09-14 by kisvegabor