lvgl/lvgl #4417: Possibility to use custom icons into lv_list_add_btn

Linzdigr, 2023-07-28:
Hello,

I exposed my problem and the tries on the forum [here](https://forum.lvgl.io/t/some-symbols-icon-dont-appear-in-list/12457) and will now expose the problem from the latest perspective I have so far:

### Introduce the problem
I'm actually trying to have custom icon inside lv_list elements in LVGL 8.2.0 but internally, LVGL doesn't seem to be able to resolve to the correct glyph descriptor, resulting into no icon displayed.

### Examples and cases
Maybe I missed something but I didn't find any solution to use custom compiled glyph icon into lv_list widget yet.
```c
list_btn1 = lv_list_add_btn(list_l1, Q_ICON_RULER, "Unit"); // Custom icon from Font Awesome not displayed
```

### Suggested solution
If not existing already (don't think so) : having the possibility to specify custom font style for the icon part of the lv_list_add_btn object.
Also, having LVGL resolving automatically the correct glyph descriptor would be great if feasible.

Thanks.

kisvegabor, 2023-07-31:
Depending on which LVGL version you are using `lv_font_montserrat_14` might be `const`.

Linzdigr, 2023-08-01:
Thanks for your reply @kisvegabor, indeed I noticed this meanwhile and choose to make my own font loader function with associated callback :)

closed 2023-08-01 by Linzdigr