lvgl/lvgl #4422: RGB888 support

krupis, 2023-07-31:
### Introduce the problem
I have been working with RGB565 displays for a while and some particular projects where I need complex color gradients I found RGB565 not to be sufficient enough.

I have recently ordered T-Display-RGB development board from the LilyGO:
https://github.com/Xinyuan-LilyGO/T-RGB/tree/main

Apparently, this display is capable of RGB888 therefore I was hoping to test it out on LVGL but from what I know it is currently not possible

### Examples and cases
Most of the times it is hard to accurately display Color pallettes and complex color gradients with RGB565 display

### Suggested solution
Would it be possible to implement 24bit color depth to the RGB to support displays such as ST7701.

kisvegabor, 2023-07-31:
Hi,

RGB888 is already supported in the master branch (v9 development). You just need to set `LV_COLOR_DEPTH 24` in `lv_conf.h`.

krupis, 2023-07-31:
@kisvegabor 

Wow thats great. I have downloaded the latest LVGL version and placed it as a driver component in my project. I had previously used LVGL version 8.3.6.

I have 2 issues:

**Issue1**
I cant seem to be able to build the project. Since I have a few images that I used on the lvgl 8.3.6, I am getting the following compilation errors:
```
D:/Programming/ESP32/RGB_Controller_Round/T-RGB/main/logo.c:2040:16: error: 'LV_IMG_CF_TRUE_COLOR' undeclared here (not in a function); did you mean 'LV_IMG_BUF_SIZE_TRUE_COLOR'?
 2040 |   .header.cf = LV_IMG_CF_TRUE_COLOR,
      |                ^~~~~~~~~~~~~~~~~~~~
      |                LV_IMG_BUF_SIZE_TRUE_COLOR
ninja: build stopped: subcommand failed.
```

```Issue2```

After I have removed all the images from my project since it was complaining about LV_IMG_CF_TRUE_COLOR,
I am now experiencing a different compilation errors:

```
D:/Programming/ESP32/RGB_Controller_Round/T-RGB/main/main.cpp:352:83: error: 'disp_drv' was not declared in this scope
  352 |   ESP_ERROR_CHECK(esp_lcd_rgb_panel_register_event_callbacks(panel_handle, &cbs, &disp_drv));
      |                                                                                   ^~~~~~~~
C:/Users/Lukas/esp/esp-idf/components/esp_common/include/esp_err.h:117:30: note: in definition of macro 'ESP_ERROR_CHECK'
  117 |         esp_err_t err_rc_ = (x);                                        \
      |                              ^
D:/Programming/ESP32/RGB_Controller_Round/T-RGB/main/main.cpp:371:26: error: 'disp_buf' was not declared in this scope
  371 |   lv_disp_draw_buf_init(&disp_buf, buf1, buf2, EXAMPLE_LCD_H_RES * EXAMPLE_LCD_V_RES);
      |                          ^~~~~~~~
D:/Programming/ESP32/RGB_Controller_Round/T-RGB/main/main.cpp:371:3: error: 'lv_disp_draw_buf_init' was not declared in this scope
  371 |   lv_disp_draw_buf_init(&disp_buf, buf1, buf2, EXAMPLE_LCD_H_RES * EXAMPLE_LCD_V_RES);
      |   ^~~~~~~~~~~~~~~~~~~~~
D:/Programming/ESP32/RGB_Controller_Round/T-RGB/main/main.cpp:383:21: error: 'disp_drv' was not declared in this scope
  383 |   lv_disp_drv_init(&disp_drv);
      |                     ^~~~~~~~
D:/Programming/ESP32/RGB_Controller_Round/T-RGB/main/main.cpp:383:3: error: 'lv_disp_drv_init' was not declared in this scope; did you mean 'lv_fs_drv_init'?        
  383 |   lv_disp_drv_init(&disp_drv);
      |   ^~~~~~~~~~~~~~~~
      |   lv_fs_drv_init
D:/Programming/ESP32/RGB_Controller_Round/T-RGB/main/main.cpp:386:23: error: 'example_lvgl_flush_cb' was not declared in this scope
  386 |   disp_drv.flush_cb = example_lvgl_flush_cb;
      |                       ^~~~~~~~~~~~~~~~~~~~~
D:/Programming/ESP32/RGB_Controller_Round/T-RGB/main/main.cpp:392:21: error: 'lv_disp_drv_register' was not declared in this scope; did you mean 'lv_fs_drv_register'?
  392 |   lv_disp_t *disp = lv_disp_drv_register(&disp_drv);
      |                     ^~~~~~~~~~~~~~~~~~~~
      |                     lv_fs_drv_register
D:/Programming/ESP32/RGB_Controller_Round/T-RGB/main/main.cpp:395:22: error: 'indev_drv' was not declared in this scope
  395 |   lv_indev_drv_init(&indev_drv);
      |                      ^~~~~~~~~
D:/Programming/ESP32/RGB_Controller_Round/T-RGB/main/main.cpp:395:3: error: 'lv_indev_drv_init' was not declared in this scope; did you mean 'lv_fs_drv_init'?       
  395 |   lv_indev_drv_init(&indev_drv);
      |   ^~~~~~~~~~~~~~~~~
      |   lv_fs_drv_init
D:/Programming/ESP32/RGB_Controller_Round/T-RGB/main/main.cpp:397:23: error: 'lv_touchpad_read' was not declared in this scope
  397 |   indev_drv.read_cb = lv_touchpad_read;
      |                       ^~~~~~~~~~~~~~~~
D:/Programming/ESP32/RGB_Controller_Round/T-RGB/main/main.cpp:398:3: error: 'lv_indev_drv_register' was not declared in this scope; did you mean 'lv_fs_drv_register'?
  398 |   lv_indev_drv_register(&indev_drv);
      |   ^~~~~~~~~~~~~~~~~~~~~
      |   lv_fs_drv_register
```

As you can see from above, it is not able to detect many functions all of a sudden. Could that be caused due to lvgl v9 or that must be some other issue with my project?

**Issue3**
Ignoring the first 2 issues and lets assume I manage to compile my project.
I can ```#define LV_COLOR_DEPTH 24``` to lvgl.h as you have suggested but is that going to override the ```CONFIG_LV_COLOR_DEPTH_16``` that is declared in sdkconfig?

C47D, 2023-08-01:
Hi, there are some changes on both display and indev handling in v9, see https://github.com/lvgl/lvgl/issues/4011

kisvegabor, 2023-08-01:
v9 is still under development and multiple things have been changed compared to v8. As @C47D mentioned in #4011 you can check out the changes. 

I've also update [lv_port_pc_eclipse](https://github.com/lvgl/lv_port_pc_eclipse) so you can try it out in a ready to use project. 

Regarding the image conversation, we are still updating the image converter, but in the meantime, in the [online image convert](https://lvgl.io/tools/imageconverter) you can select `CF_RGB565A8` and in the converted image change `LV_IMG_CF_RGB565A8` to `LV_COLOR_FORMAT_RGB565A8`.

I know it's quite cumbersome now, so thank you if take the effort to try out v9. It'd be a very useful feedback for us.

krupis, 2023-08-01:
@kisvegabor 

Hello. I am going through #4011 as well ass https://github.com/lvgl/lv_port_pc_eclipse and trying to modify my code to be compatible with v9.

I have come across a weird issue.

In the lv_port_disp_template.c the following function is declared:
```
static void disp_flush(lv_disp_t * disp_drv, const lv_area_t * area, lv_color_t * px_map)
{
    if(disp_flush_enabled) {
        /*The most simple case (but also the slowest) to put all pixels to the screen one-by-one*/

        int32_t x;
        int32_t y;
        for(y = area->y1; y <= area->y2; y++) {
            for(x = area->x1; x <= area->x2; x++) {
                /*Put a pixel to the display. For example:*/
                /*put_px(x, y, *px_map)*/
                px_map++;
            }
        }
    }

    /*IMPORTANT!!!
     *Inform the graphics library that you are ready with the flushing*/
    lv_disp_flush_ready(disp_drv);
}
```

I have created very simillar disp_flush function in my project:

```
static void disp_flush(lv_disp_t * disp_drv, const lv_area_t * area, lv_color_t * px_map)
{
        int32_t x;
        int32_t y;
        for(y = area->y1; y <= area->y2; y++) {
            for(x = area->x1; x <= area->x2; x++) {
                /*Put a pixel to the display. For example:*/
                /*put_px(x, y, *px_map)*/
                px_map++;
            }
        }
    /*IMPORTANT!!!
     *Inform the graphics library that you are ready with the flushing*/
    lv_disp_flush_ready(disp_drv);
}
```

and I am passing it to ```lv_disp_set_flush_cb``` function as following:
```
  lv_init();
  lv_disp_t * disp = lv_disp_create(EXAMPLE_LCD_H_RES, EXAMPLE_LCD_V_RES);
  void *buf1 = NULL;
  void *buf2 = NULL;
  ESP_LOGI(TAG, "Allocate separate LVGL draw buffers from PSRAM");

  buf1 = heap_caps_malloc(EXAMPLE_LCD_H_RES * 100 * sizeof(lv_color_t), MALLOC_CAP_SPIRAM);
  assert(buf1);
  buf2 = heap_caps_malloc(EXAMPLE_LCD_H_RES * 100 * sizeof(lv_color_t), MALLOC_CAP_SPIRAM);
  assert(buf2);
  lv_disp_set_flush_cb(disp, disp_flush);
  lv_disp_set_draw_buffers(disp, buf1, buf2, 1024, LV_DISP_RENDER_MODE_PARTIAL);
```

I am getting the following error:
```
D:/Programming/ESP32/RGB_Controller_Round/T-RGB/main/main.cpp:417:30: error: invalid conversion from 'void (*)(lv_disp_t*, const lv_area_t*, lv_color_t*)' {aka 'void (*)(_lv_disp_t*, const lv_area_t*, lv_color_t*)'} to 'lv_disp_flush_cb_t' {aka 'void (*)(_lv_disp_t*, const lv_area_t*, unsigned char*)'} [-fpermissive]
  417 |   lv_disp_set_flush_cb(disp, disp_flush);
      |                              ^~~~~~~~~~
      |                              |
      |                              void (*)(lv_disp_t*, const lv_area_t*, lv_color_t*) {aka void (*)(_lv_disp_t*, const lv_area_t*, lv_color_t*)}
```

I cannot fully wrap my head around this and why is it complaining

krupis, 2023-08-01:
Initially, when I was using LVGL 8.3.6 I had the following LVGL initialization code:

```
static void example_increase_lvgl_tick(void *arg)
{
    /* Tell LVGL how many milliseconds has elapsed */
    lv_tick_inc(EXAMPLE_LVGL_TICK_PERIOD_MS);
}
```

```
ESP_LOGI(TAG, "Initialize LVGL library");
    lv_init();
    void *buf1 = NULL;
    void *buf2 = NULL;

    ESP_LOGI(TAG, "Allocate separate LVGL draw buffers from PSRAM");
    buf1 = heap_caps_malloc(EXAMPLE_LCD_H_RES * 100 * sizeof(lv_color_t), MALLOC_CAP_SPIRAM);
    assert(buf1);
    buf2 = heap_caps_malloc(EXAMPLE_LCD_H_RES * 100 * sizeof(lv_color_t), MALLOC_CAP_SPIRAM);
    assert(buf2);
    // initialize LVGL draw buffers
    lv_disp_draw_buf_init(&disp_buf, buf1, buf2, EXAMPLE_LCD_H_RES * 100);

    ESP_LOGI(TAG, "Register display driver to LVGL");
    lv_disp_drv_init(&disp_drv);
    disp_drv.hor_res = EXAMPLE_LCD_H_RES;
    disp_drv.ver_res = EXAMPLE_LCD_V_RES;
    disp_drv.flush_cb = example_lvgl_flush_cb;
    disp_drv.draw_buf = &disp_buf;
    disp_drv.user_data = panel_handle;

    lv_disp_t *disp = lv_disp_drv_register(&disp_drv);

    ESP_LOGI(TAG, "Install LVGL tick timer");
    // Tick interface for LVGL (using esp_timer to generate 2ms periodic event)
    const esp_timer_create_args_t lvgl_tick_timer_args = {.callback = &example_increase_lvgl_tick, .name = "lvgl_tick"};
    esp_timer_handle_t lvgl_tick_timer = NULL;
    ESP_ERROR_CHECK(esp_timer_create(&lvgl_tick_timer_args, &lvgl_tick_timer));
    ESP_ERROR_CHECK(esp_timer_start_periodic(lvgl_tick_timer, EXAMPLE_LVGL_TICK_PERIOD_MS * 1000));

    while (1) {
        // raise the task priority of LVGL and/or reduce the handler period can improve the performance
        vTaskDelay(pdMS_TO_TICKS(5));
        // The task running lv_timer_handler should have lower priority than that running `lv_tick_inc`
        lv_timer_handler();
    }

```

Now, I have updated LVGL to v9 and have replaced the above code to the following:

```
  ESP_LOGI(TAG, "Initialize LVGL library");
  lv_init();
  lv_disp_t *disp = lv_disp_create(EXAMPLE_LCD_H_RES, EXAMPLE_LCD_V_RES);
  lv_disp_set_flush_cb(disp, (lv_disp_flush_cb_t)(disp_flush));
  static lv_color_t buf_2_1[EXAMPLE_LCD_H_RES * 10];
  static lv_color_t buf_2_2[EXAMPLE_LCD_H_RES * 10];
  lv_disp_set_draw_buffers(disp, buf_2_1, buf_2_2, sizeof(buf_2_1), LV_DISP_RENDER_MODE_PARTIAL);

   ESP_LOGI(TAG, "Install LVGL tick timer");
  const esp_timer_create_args_t lvgl_tick_timer_args = {.callback = &example_increase_lvgl_tick, .name = "lvgl_tick"};
  esp_timer_handle_t lvgl_tick_timer = NULL;
  ESP_ERROR_CHECK(esp_timer_create(&lvgl_tick_timer_args, &lvgl_tick_timer));
  ESP_ERROR_CHECK(esp_timer_start_periodic(lvgl_tick_timer, EXAMPLE_LVGL_TICK_PERIOD_MS * 1000));

    while (1) {
        // raise the task priority of LVGL and/or reduce the handler period can improve the performance
        vTaskDelay(pdMS_TO_TICKS(5));
        // The task running lv_timer_handler should have lower priority than that running `lv_tick_inc`
        lv_timer_handler();
    }

```

The device boots up but the display is not able to display anything. For some reason, I believe this could very much be due to incorrect implementation on disp_flush function.

```
static void disp_flush(lv_disp_t *disp_drv, const lv_area_t *area, lv_color_t *px_map) {
  printf("display flush \n");
  int32_t x;
  int32_t y;
  for (y = area->y1; y <= area->y2; y++) {
    for (x = area->x1; x <= area->x2; x++) {
      /*Put a pixel to the display. For example:*/
      /*put_px(x, y, *px_map)*/
      px_map++;
    }
  }
  /*IMPORTANT!!!
   *Inform the graphics library that you are ready with the flushing*/

  lv_disp_flush_ready(disp_drv);

}
```

I dont think the above is correct.

with the LVGL version 8.3.6 I have been using:
```
static void example_lvgl_flush_cb(lv_disp_drv_t *drv, const lv_area_t *area, lv_color_t *color_map)
{
    esp_lcd_panel_handle_t panel_handle = (esp_lcd_panel_handle_t)drv->user_data;
    int offsetx1 = area->x1;
    int offsetx2 = area->x2;
    int offsety1 = area->y1;
    int offsety2 = area->y2;
    // pass the draw buffer to the driver
    esp_lcd_panel_draw_bitmap(panel_handle, offsetx1, offsety1, offsetx2 + 1, offsety2 + 1, color_map);
    lv_disp_flush_ready(drv);
}
```

Could you take a look at my flush callback and help me figure out how to implement it correctly?

krupis, 2023-08-02:
@kisvegabor 

Unfortunately it is still not fully clear to me regarding ```disp_flush_cb ``` I cannot use the same disp_flush_cb because you no longer have ```lv_disp_drv_t```. 

The original disp_flush_cb that I use in version 8.3.6:
```
static void disp_flush(lv_disp_drv_t *drv, const lv_area_t *area, lv_color_t *color_map)
{
    esp_lcd_panel_handle_t panel_handle = (esp_lcd_panel_handle_t)drv->user_data;
    int offsetx1 = area->x1;
    int offsetx2 = area->x2;
    int offsety1 = area->y1;
    int offsety2 = area->y2;

    esp_lcd_panel_draw_bitmap(panel_handle, offsetx1, offsety1, offsetx2 + 1, offsety2 + 1, color_map);
    lv_disp_flush_ready(drv);
}
```

as you can see it relies on ```drv->user_data;``` 

In the version 9 of lvgl, I do not know how to retrieve ```esp_lcd_panel_handle_t``` 
```
static void disp_flush(lv_disp_t *disp_drv, const lv_area_t *area, lv_color_t *px_map) {
  printf("display flush \n");
    esp_lcd_panel_handle_t panel_handle = (esp_lcd_panel_handle_t)disp_drv->user_data;
    int offsetx1 = area->x1;
    int offsetx2 = area->x2;
    int offsety1 = area->y1;
    int offsety2 = area->y2;

    // pass the draw buffer to the driver
    esp_lcd_panel_draw_bitmap(panel_handle, offsetx1, offsety1, offsetx2 + 1, offsety2 + 1, px_map);
    lv_disp_flush_ready(disp_drv);

}
```

I hope you can give me some advice regarding this.

Additionally, thank you for your insights regarding ```lv_disp_set_draw_buffers```. I have replaced it with:

```
  void *buf1 = NULL;
  void *buf2 = NULL;
  buf1 = malloc((EXAMPLE_LCD_H_RES * EXAMPLE_LCD_V_RES) * sizeof(lv_color_t));
  assert(buf1);
  buf2 = malloc((EXAMPLE_LCD_H_RES * EXAMPLE_LCD_V_RES) * sizeof(lv_color_t));
  assert(buf2);

  lv_disp_set_draw_buffers(disp, buf1, buf2,  (EXAMPLE_LCD_H_RES * EXAMPLE_LCD_V_RES) * sizeof(lv_color_t), LV_DISP_RENDER_MODE_FULL );
```

krupis, 2023-08-03:
No I completely missed that.. :( Is that also part of the LVGL v9 changes?

Can you confirm if below is correct:

1. In LVGL 8.3.6 we used to be able to do:
```disp_drv.user_data = panel_handle;```

now we have to do:
```
lv_disp_set_user_data(disp, panel_handle);
```

2. In LVGL 8.3.6 we used to be able to do:
```disp_drv.draw_buf = &disp_buf;```

now we have to do:
```
  lv_disp_set_draw_buffers(disp, buf1, buf2,  (EXAMPLE_LCD_H_RES * EXAMPLE_LCD_V_RES) * sizeof(lv_color_t), LV_DISP_RENDER_MODE_FULL );

```

kisvegabor, 2023-08-03:
Looks good! Does it work? :slightly_smiling_face:

krupis, 2023-08-03:
@kisvegabor 
Yes. I have managed to get the display to run and even tested touch. Everything seems to be ok :) 
Now the most important questions: whether the 24 bit color depth actually works:

Steps I did:

**1. I have downloaded an image:**
[image]

**2. Run the image through the converter (https://lvgl.io/tools/imageconverter). I have selected CF_RGB565A8.**

**3. In the converted image I have replaced LV_IMG_CF_RGB565A8  with LV_COLOR_FORMAT_RGB565A8**

**4. In the lvgl.h I have set the following:**

```
#define LV_COLOR_DEPTH 24
```

and also in the menuconfig of my project:
[image]

**5.Then I run the code:**
```
    LV_IMG_DECLARE(gradient_24)
    lv_obj_t * img1 = lv_img_create(lv_scr_act());
    lv_img_set_src(img1, &gradient_24)
```

I am able to see some flaws in the color where it transitions. Please see the images below. I have marked the places that are most noticeable with red arrows. I hope you can notice it on the images, in person it is more noticeable:

[image]

[image]

I initially had this issue with the RGB565 display and I thought that these color transitioning stripes happen due to color depth not being high enough, but we can see that even in this case I have this problem.

My questions:

1. How can I ensure that this is actually 24 bit color depth? The converted images are 4KB size. If I choose CF_TRUE_COLOR option (I used to choose this for LVGL version 8.3.6), the size will be around 13KB. How can I be sure that converted image really is 24bit color depth?

2. How can I properly test whether its 24 bit color depth or 16 bit color depth? Perhaps you know a good way? Is there a special image that I can display that would surely tell me?

3. I was hoping to use ```lv_img_buf_get_px_color``` to read the color of an image to check if it is capable of returning 24 bit color information. For example I can use paint to create RGB(0,56,12) color image. And when I put this image on my display, I expect to be able to read back the color values. Is this function deprecated in lvgl v9? I used to be able to use this in v9 but it seems that it no longer exists in ```lv_img_buf.c```

kisvegabor, 2023-08-04:
For converting the image, we have [this new image converter tool](https://github.com/lvgl/lvgl/pull/4414) for v9. It should support rgb888 output as well. Could you try it out?

>  I have it set to RGB565 in the menuconfig

Exactly which parameter's value did you set to RGB565 in menuconfig?  `LV_COLOR_DEPTH`? if so `lv_conf.h` should be stronger than menuconfig. Can you check it by adding this in `main.c`:
```c
#if LV_COLOR_DEPTH == 24
printf("24 bit color depth\n");
#endif
```

kisvegabor, 2023-08-06:
I just wanted to convert your image and script failed for me too. However, I've just realized that you need an ARGB8888 image as the background has zero opacity.  So the `LV_COLOR_DEPTH == 32` part of online image converter can be used for 24 bit + alpha as well. I've just prepared the image for and also tried it out. I just removed the unused color depth parts and change the color format to `LV_COLOR_FORMAT_ARGB8888`.

[img_color_circle.c.zip](https://github.com/lvgl/lvgl/files/12270458/img_color_circle.c.zip)

> Also, can you answer one of my previous questions regarding lv_img_buf_get_px_color 

We can add it again later, but getting a color from and ARGB88888 image is very simple: 
```
const lv_color32_t * c32 = (lv_color32_t *) img_color_circle.data;
c32 += y * img_color_circle.header.w + x;
//c32->red, c32->green, c32->blue
```

krupis, 2023-08-07:
@kisvegabor 
I have tried to display the image that you have sent

***TEST1

I have set the color depth in the menuconfig to 24:
[image]

Also, I have defined ```#define LV_COLOR_DEPTH 24``` in the ```lvgl.h```

The result:
[image]

***TEST2

I have set the color depth in the menuconfig to 32:
[image]

Also, I have defined ```#define LV_COLOR_DEPTH 32``` in the ```lvgl.h```

The result:
[image]

So my display either fails to display 32 bit color depth image or there seems to be an issue with the image converter?

krupis, 2023-08-07:
@kisvegabor 
Ahh.. I am having the same issue that I have mentioned earlier and its very silly... I am again trying to display the image that you have provided:

If I enable 24 color depth in the menuconfig:
[image]

and ```#define LV_COLOR_DEPTH 24``` in lvgl__lvgl/src/lvgl.h

and build the project, the image is not displayed correctly:
[image]

However, if I change the menuconfig color depth to 16 bit:
[image]

while keeping the define in lvgl.h. It will start to work all of a sudden:
[image]

I have added the following in my main:
```
  #if LV_COLOR_DEPTH == 24
  printf("24 bit color depth\n");
  #endif
```

and both ways, it prints "24 bit color depth".

I cannot wrap my head around why would setting ```LV_COLOR_DEPTH``` via menuconfig cause troubles? I dont see any difference setting the color depth via menuconfig or setting in in the ```lvgl.h```. Both should be the same!
Perhaps you have any clues why would that happen?

krupis, 2023-08-07:
Anyways, with the ```lvgl.h``` #define LV_COLOR_DEPTH 24 and menuconfig set to 16bit color, I have tried the following test to check if my display is 24 bit color depth:

```
  lv_color_t test_color[20];
  lv_meter_indicator_t *indic[20];
  for (int i = 0; i < 20; i++) {
    test_color[i] = lv_color_make(95, 124 + i, 95);
    indic[i] = lv_meter_add_arc(meter, indic_w, test_color[i], 0);
    lv_meter_set_indicator_start_value(meter, indic[i], 0+(5*i));
    lv_meter_set_indicator_end_value(meter, indic[i], 5+(5*i));
  }
```

I am trying to display a pie chart and every indicator I append a green color by 1. So it goes like this:
(95, 124, 95)
(95, 125, 95)
(95, 126, 95)
(95, 127, 95)...

If display is 24 bit color depth, I expect smooth transitioning between colors.

The result:
[image]

I am not sure if it is properly visible from the image attached, but this to me proves that it is not 24 bit color depth. The color repeats itself every 4-5 indicators so I think the colors are 16 bit depth.

kisvegabor, 2023-08-08:
> I am now digging into the driver for st7701 that I use. Perhaps there is a special command that I need to use to put the display into rgb888 mode?

Yes, it's required. You need to tell the controller that now you are in 24 bit mode. Probably this:
[image]

krupis, 2023-08-08:
@kisvegabor 
Yep. I have tried to set this register to 0b0111000 to set it to 24 bit/pixel mode but havent managed to get anything dispalyed in 24 bit color depth format yet.
I am hoping to hear back from the LilyGo. Since they wrote the driver for this display, perhaps they know more about this...

closed 2023-08-30 by lvgl-bot