lvgl/lvgl #4429: Missing colorwheel documentation in master branch

hayschan, 2023-08-03:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

In the latest master of LVGL, the colorwheel folder under `examples/widgets` gone missing. It should be at https://github.com/lvgl/lvgl/tree/master/examples/widgets/colorwheel but it does not exist there.

This causes no Examples and API in the latest documentation of colorwheel in LVGL.

### To Reproduce

This is the screenshot of the latest documentation on the web:

[image]

### Expected behavior

This is the screenshot of the `v8` documentation on the web:

[image]

PeterBee97, 2023-08-03:
Hi @hayschan ,
The colorwheel widgets was removed in v9.

kisvegabor, 2023-08-03:
FYI, it was removed because it was non intuitive and slow.  I'm planning to replace it with something more universal later.

hayschan, 2023-08-04:
> Hi @hayschan , The colorwheel widgets was removed in v9.

omg, just started using colorwheel in my project.

hayschan, 2023-08-04:
> FYI, it was removed because it was non intuitive and slow. I'm planning to replace it with something more universal later.

I see. 

The color picker introduced in iOS 14 is quite intuitive and easy to use. We may try something similar. 

There are three modes for the color picker:

1. Grid
2. Spectrum
3. Sliders

[image]

[image]

[image]

hayschan, 2023-08-04:
I re-created the sliders mode of iOS color picker using Squareline Studio.

[image]

kisvegabor, 2023-08-04:
Thank you for sharing. My idea is to allow using an image and specifying the knob movement, such as:
- Any: Rectangle image pick any color
- Grid: As in your example, allow selecting only fixed positions. If the image is not rectangle we don't select cells where the opacity is 0.
- Circle: Set radius on which the knob can move
- Slider: Move vertically or horizontally.  
- Custom: Somehow allow using custom movement

Once you set the knob somewhere LVGL just ready the color of that image.

What do you think?

hayschan, 2023-08-04:
I also came across Google's implementation on their [web color picker](https://g.co/kgs/tkQZ91), which is quite intuitive.

- On the left box, it shows the selected color appearance. 
- On the bottom, the slider controls the Hue.
- On the right, there is rectangle plane to choose the color in two dimensional space, which is Saturation and Value/Lightness (depends on HSV/HSL model used).

[image]

kisvegabor, 2023-08-06:
I meant implementing all five but I'm not sure about it yet, just thinking aloud. I think what we need is:
- allow using an image from where colors can be picked (can be grid, small hexagons, ring, rectangle, bar) because generating the colors run time are slow an limited 
- limit the selection area (e.g. horizontal only for bars, discrete points for grid, circle for ring).

closed 2023-08-28 by lvgl-bot

dchithinh, 2024-04-09:
Hi everyone,
Does lvgl support this kind of color wheel & picker?

[image]

kisvegabor, 2024-04-12:
Not directly,but there is a workaround: Use any image you want as a color picker, and its `CLICK`event, get the clicked coordinates with  `lv_indev_get_point(lv_indev_active(),&point);` and read the color on that pixel from the image.

haigithub, 2024-04-26:
> Not directly,but there is a workaround: Use any image you want as a color picker, and its `CLICK`event, get the clicked coordinates with `lv_indev_get_point(lv_indev_active(),&point);` and read the color on that pixel from the image.

lv_img_buf_get_px_color was not found in LVGL v9 version, is there an alternative resolution?

kisvegabor, 2024-04-29:
Auch, I think there is no replacement for it, however if you are using a normal RGB565 or ARGB8888 image, it should be simple to seek to a pixel and read its color.

leelx78, 2024-09-15:
Hi can you pls give an example of how to read the color of a given pixel on the screen using LVGL9?

kisvegabor, 2024-09-16:
LVGL doesn't provide an interface to read back colors from the screen. If you are using a display controller you can read its memory or i na frame buffer you can also find the required pixel.

leelx78, 2024-09-23:
Hi I tried to implement a colorwheel based on CLICK and PRESS events but I can't manage to have it track the wheel as user might expect (i.e. you move the pinger across the wheel and the color changes continuously). It only seems to work by clicking each time.

Can you bring back the old color wheel? it wasn't perfect and indeed a bit slow but it's still better than not having it at all

Thanks!

kisvegabor, 2024-09-25:
Could the `PRESSING` event help?

> Can you bring back the old color wheel? it wasn't perfect and indeed a bit slow but it's still better than not having it at all

Let's see what we can do with the custom things before making a decision on it. :slightly_smiling_face:

smcgann99, 2024-09-30:
Just to add my interest in the return of a color wheel ;-)
I use openHASP and Tasmota projects, which both use LVGL for home automation where a color picker is an important feature.

kisvegabor, 2024-09-30:
Noted, thank you! I'll keep it mind!

kisvegabor, 2024-10-11:
Using my memory :smile: