lvgl/lvgl #4431: Line object in benchmark app does not have width or height

nicusorcitu, 2023-08-03:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
The line benchmark is composed by 6 objects, each of them having 16 points.
At the end of this composition, everything looks good - e.g. [here](https://github.com/lvgl/lvgl/blob/5d056ed2b909aaf8e17c66d4a8042e49c7672d90/demos/benchmark/lv_demo_benchmark.c#L1161) the lines are correct.
But the line object does not have any width/height set at this point.

We tried to set it manually by calling lv_obj_set_width(obj, max_x2-max_x1) and lv_obj_set_height(obj, max_y2-max_y1) but that does not work.

Because the width and height can not be set, while animation is running, the coordinates are adjusted [on x](https://github.com/lvgl/lvgl/blob/5d056ed2b909aaf8e17c66d4a8042e49c7672d90/demos/benchmark/lv_demo_benchmark.c#L1211) and [on y](https://github.com/lvgl/lvgl/blob/5d056ed2b909aaf8e17c66d4a8042e49c7672d90/demos/benchmark/lv_demo_benchmark.c#L1219) and the line points will get out of screen. The CPU somehow can handle line points out of bound, but the GPU will hang.

@kisvegabor Can you please advise?

### Screenshots or video
[image]

kisvegabor, 2023-08-03:
We can fix it for sure, but clipping the lines might be a needed in other cases too.
[image]

nicusorcitu, 2023-08-03:
We do clip the lines, we use the clip area while drawing the lines, but not sure what value is being given on the draw_ctx->clip_area. Will have to check that.

nicusorcitu, 2023-08-03:
Just reproduced the issue with a single line (two points).
The debugs are:
```
(gdb) p *point1                                                                                     
$2 = {x = 357, y = 1280}
(gdb) p *point2
$3 = {x = 518, y = 1435}
(gdb) p *draw_ctx->clip_area
$5 = {x1 = 355, y1 = 1278, x2 = 521, y2 = 1279}
(gdb)
```
Clip area does not look good either.

kisvegabor, 2023-08-04:
Ouch! I'm a little bit puzzled. How can we handle this sliding chart issue then? Do you have any idea?

nicusorcitu, 2023-08-07:
I do need to investigate how GPU behaves in these scenarios and see if the problem is only on lines (as vglite use software stroke implementation) or this is a general issue on all the other draws (rectangles, gradients...).

closed 2023-08-29 by lvgl-bot

nicusorcitu, 2023-09-01:
We have found the issue in LVGL acceleration code and fix the problem. The clipping area was wrong. Once the clip area is set correctly to not exceed the screen size, everything is good, so there is no GPU issue.

kisvegabor, 2023-09-01:
Great! :tada: 
Do we need to fix something in LVGL or was it purely in GPU code?

nicusorcitu, 2023-09-01:
In LVGL benchmark code, we still need to fix the line coordinates because they will exceed the screen and instead of 8 lines will be visible only part of them

But the crash itself is fixed, it was an issue in the acceleration code (not GPU driver), easy fix.