lvgl/lvgl #4432: LVGL8.3's switch widget emits two LV_EVENT_VALUE_CHANGED times when switching

cnjhb, 2023-08-03:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version **just8.3**
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

switch widget emits two LV_EVENT_VALUE_CHANGED times when switching
### To Reproduce
```c
#include "lvgl/lvgl.h"
static void event_handler_jhb(lv_event_t* e){
  auto code=lv_event_get_code(e);
  auto target=lv_event_get_target(e);
  if(lv_obj_has_state(target,LV_STATE_CHECKED)){
/*This is called twice when the switch component is opened*/
    LV_LOG_USER("jhbzs\n");
  }
}
void lv_example_switch_jhb(){
  auto btn=lv_switch_create(lv_scr_act());
  lv_obj_add_event_cb(btn,event_handler_jhb,LV_EVENT_VALUE_CHANGED,NULL);
}

```

### Expected behavior

The LV_EVENT_VALUE_CHANGED event is sent only once

kisvegabor, 2023-08-03:
Hi,

With the latest version from `release/v8.3` branch it works well for me.

cnjhb, 2023-08-04:
> Hi,
> 
> With the latest version from `release/v8.3` branch it works well for me.

Yes Yes. I thought the lvgl that came with the simulator was up to date.

closed 2023-08-04 by cnjhb