lvgl/lvgl #4445: Tab and Chinese

DarkLigther, 2023-08-11:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When tab and Chinese appear together, there is a probability of garbled code.

### Screenshots or video
[image]

DarkLigther, 2023-08-11:
This is text:"龟虽寿\n\t[两汉]曹操\n\t神龟虽寿，犹有竟时。腾蛇乘雾，终为土灰。老骥伏枥，志在千里。烈士暮年，壮心不已。盈缩之期，不但在天；养怡之福，可得永年。幸甚至哉，歌以咏志。\n\t译文:\n\t神龟虽然拥有长寿的能力，但它终究也会有死亡的时刻。腾蛇虽然可以在雾中翱翔，但它最终也会化为尘埃。即便年迈的千里马已经退役，它的雄心壮志仍然能够驰骋千里。有"

kisvegabor, 2023-08-11:
Plase send a code snippet to reproduce the issue

closed 2023-09-02 by lvgl-bot