lvgl/lvgl #4448: Simple new feature: load font from buffer

kisvegabor, 2023-08-11:
I think it's a local and quite straightforward feature, so if you would like contribute to LVGL, this one is a perfect candidate. If you are interested in it, just drop a line below, and we can discuss the details.

------

Continue https://github.com/lvgl/lv_binding_js/issues/15#

### Introduce the problem

Now we have `lv_font_load()` to load a font from a binary file but in some cases it would be great to have an option to load the font from a buffer as well.

### Examples and cases
- Would be useful for the [JavaScript binding](https://github.com/lvgl/lv_binding_js/issues/15#)
- It's faster than loading from file

### Suggested solution

Add an lv_font_load_from_buffer() function.  We could use some function pointers like `read_cb` which could point to `lv_fs_read` or a `buf_read` function. This way we don't need to add a bunch of duplication.

cc @derekstavis

rik43, 2023-08-12:
"Javascript binding" - bad link

kisvegabor, 2023-08-14:
> https://github.com/lvgl/lv_binding_cpp/issues/8#issuecomment-1675827276

Fixed, thank you :slightly_smiling_face:

zjanosy, 2023-08-15:
@derekstavis
After discussing with Gabor I am going to implement this feature.
We have identified two possible use cases:
1. Read the whole structure into RAM. This is how the current lv_load_font() function works. Adding support for this is simple.
2. Read only the small structures, but use pointers to the larger structures in the buffer. This may make sense if the font is in (not too slow) external flash, and RAM is a scarce resource. This is also possible, though it would require more substantial changes.
Which one is your use case?

derekstavis, 2023-08-16:
Hmm, I can see two main use-cases for loading from a buffer:

1. The font is compiled into code (like with `XPM` files), and passed to `lv_load_font_from_buffer` as a pointer
2. The font has been loaded by a binding into RAM and passed to `lv_load_font_from_buffer` as a pointer

The use-case (2) is the one I'm planning to use on the JS binding

kisvegabor, 2023-08-16:
FYI, a PR is sent: #4462

kdschlosser, 2023-08-17:
Being able to load a font from a buffer object would be a great idea if it allows loading the font without needing to use the fsdriver. For the Python binding it would be a lot easier to open the file and dump the contents of the file into a single function call.

kdschlosser, 2023-08-17:
@kisvegabor 
The load font from buffer would be a good idea if I could directly pass a buffer to the load function for a font. With an image I am able to pass a filename, `lv_img_dsc_t` structure or an array of bytes that contains the image data. The only thing that can be passed to the load function for a font is a file name.

When dealing with the Python binding there is no real need to deal with the built in file system driver in LVGL if all of the parts of LVGL that read data using the file system driver also have the ability to directly pass the data instead. 

Having to add this code.

```python
class __fs_driver(object):
    import errno

    _err_mapping = {
        errno.EPERM: FS_RES_NOT_IMP,  # NOQA
        errno.ENOENT: FS_RES_NOT_EX,  # NOQA
        errno.EINTR: FS_RES_HW_ERR,  # NOQA
        errno.EIO: FS_RES_HW_ERR,  # NOQA
        errno.ENXIO: FS_RES_HW_ERR,  # NOQA
        errno.E2BIG: FS_RES_INV_PARAM,  # NOQA
        errno.EBADF: FS_RES_FS_ERR,  # NOQA
        errno.EAGAIN: FS_RES_BUSY,  # NOQA
        errno.ENOMEM: FS_RES_OUT_OF_MEM,  # NOQA
        errno.EACCES: FS_RES_DENIED,  # NOQA
        errno.EFAULT: FS_RES_HW_ERR,  # NOQA
        errno.EBUSY: FS_RES_BUSY,  # NOQA
        errno.EEXIST: FS_RES_FS_ERR,  # NOQA
        errno.ENODEV: FS_RES_HW_ERR,  # NOQA
        errno.ENOTDIR: FS_RES_FS_ERR,  # NOQA
        errno.EISDIR: FS_RES_FS_ERR,  # NOQA
        errno.EINVAL: FS_RES_INV_PARAM,  # NOQA
        errno.EMFILE: FS_RES_BUSY,  # NOQA
        errno.ENOSPC: FS_RES_FULL,  # NOQA
        errno.ESPIPE: FS_RES_FS_ERR,  # NOQA
        errno.EROFS: FS_RES_NOT_IMP,  # NOQA
        errno.ENOSYS: FS_RES_NOT_IMP  # NOQA
    }

    del errno

    def __init__(self):
        import sys

        sys.modules['fs_driver'] = self
        self.__open_files = {}

    def __fs_open_cb(self, _, path, mode):
        if mode == FS_MODE_WR:  # NOQA
            p_mode = 'wb'
        elif mode == FS_MODE_RD:  # NOQA
            p_mode = 'rb'
        elif mode == FS_MODE_WR | FS_MODE_RD:  # NOQA
            p_mode = 'rb+'
        else:
            print(
                f"fs_open_cb - open mode error, {mode} is invalid mode"
            )
            return FS_RES_NOT_IMP  # NOQA

        count = 0
        chars = ''

        while True:
            char = path[count].value

            if char == b'\x00':
                break

            chars += char.decode('utf-8')
            count += 1

        path = chars
[... 145 lines cut ...]

        if cache_size >= 0:
            fs_drv.cache_size = cache_size

        fs_drv_register(fs_drv)  # NOQA

__fs_driver()
```

To accomplish the same task as this code.

```python

with open('some_font.bin', 'rb') as f:
    data = f.read()

font = lv.load_font(data)

```

The fist code block is how it has to be done currently when dealing with fonts.  

In the above example data is an array of bytes.

I am sure that there is a reason why and it is probably because of the size of the font data and an MCU not being able to load it all into memory. So calls are made to the file system driver to collect only the data that is needed at that specific point in time. When dealing with any port that is able to run on a PC there is not much of a concern regarding memory consumption.

kdschlosser, 2023-08-17:
When dealing with images there is a routine that is done to see what the data is that has been passed. Keep the API the same. use a marker at the beginning of a structure to be able to identify if the passed in data is a file name or a structure that is holding the font data.

```c

typedef struct {
    /*  
     * IDK what the font file format is this here here only to 
     * ensure the first character is not in the ordinal range of 
     * an ascii character.
     \* 
    uint8_t reserved; 
    uint8_t * data;
} lv_font_buffer_t;

enum _lv_font_src_t {
    LV_FONT_SRC_UNKNOWN = 0,
    LV_FONT_SRC_FILE,
    LV_FONT_SRC_BUFFER,
};

typedef uint8_t lv_font_src_t;
lv_font_src_t lv_font_src_get_type(const void * src)
{
    lv_font_src_t font_src_type = LV_FONT_SRC_UNKNOWN;

    if(src == NULL) return font_src_type;
    const uint8_t * u8_p = src;

    /*The first byte shows the type of the image source*/
    if(u8_p[0] >= 0x20 && u8_p[0] <= 0x7F) {
        font_src_type = LV_FONT_SRC_FILE; /*If it's an ASCII character then it's file name*/
    }
    else {
        font_src_type = LV_FONT_SRC_BUFFER;
    }

    return font_src_type;
}

lv_font_src_t lv_font_src_get_type(const void * src);

lv_font_t * lv_font_load(const void * src) {
    lv_font_src_t src_type = lv_font_src_get_type(src);
    /*
    code here to handle the 2 different font data types
    */
}

```

Keeps the API consistent with what is already in place when dealing with image data. or as close as possible.

zjanosy, 2023-08-18:
There is no need to get into the details of the file system. The mechanism is completely transparent to the user. The trick is to "generalize" the path parameter (which is normally a const char *), and pass a pointer to a structure to lv_fs_open() instead of a string. This struct, lv_fs_path_ex, contains the parameters of the memory-mapped file (buffer address and size). The trick is that the struct begins with a char, which is initialized to the the appropriate drive letter, so the file system knows which driver it has to assign to the file. There is a helper function lv_fs_make_path_ex() to fill out the struct.

There is also a helper function called lv_load_font_from_buffer(), which creates and initializes the struct, so you don't even need to be concerned with that part.

kdschlosser, 2023-08-19:
will this same thing work for images as well?

kisvegabor, 2023-08-19:
Nope. I see that images and fonts look similar but actually they are not so similar in the practice. 

The main difference is that images can be used directly as an array. That is if you have an RGB bitmap, LVGL can render it. A decoder (e.g. PNG) can read the image data directly as well. On the other hand fonts are complex, built from multiple linked structures and therefore cannot be used from a binary array directly.

kdschlosser, 2023-08-21:
Any can't they be read from a binary array? It's the same exact thing as reading it from a binary file... a byte number marks the location in a file just like byte number would be a location in an array. That sounds kind of silly that it cannot be read from an array. All an array is is a location in memory exactly the same thing as what this specific idea is doing just in a round about way.

kisvegabor, 2023-08-21:
So the fonts are much more complex than just their bitmaps. See this for example: https://github.com/lvgl/lvgl/blob/master/src/font/lv_font_montserrat_10.c

All these are coded in into a binary file, requiring a lot of bit reads, creating sever array and descriptor. So it could be read, but really complicated. 

Reading an image is just the pixel values next to each other. Super simple.

zjanosy, 2023-08-23:
After discussing why loading bitmaps from memory-mapped file would not work with @kisvegabor, we concluded that it does work, even though it makes little sense to load a bitmap from memory on an embedded system. But maybe for JS or Python bindings it would be useful.

kdschlosser, 2023-08-24:
There aren't the memory constraints and using the file system adds a lot of complexity to it from a user standpoint, Having to create all of the callbacks to handle storing an open file and assigning it some kind of an identifier and dealing with the type conversions going from C code to python and vice versa makes it exceedingly complex. 

A lot of it has to do with creating a buffer that a point to can be attached to one of the pointer that is passed to the read callback. That buffer needs to stay in scope and it would need to be a global. The problem is when doing that how do I know when that buffer is no longer in use so it can be deleted. The answer is when the file gets closed. which is the solution but the entire file is now loaded into memory at that point anyhow so why not have the ability to just pass all the data as one shot and remove having to use the file system driver all together.

kisvegabor, 2023-08-25:
It might be true in Python on desktop, but on embedded loading the entire file/image to RAM by default is no working.

kdschlosser, 2023-08-26:
Ya lost me on that last statement. I am able to load an image into memory and pass it to an image object without an issue on an ESP32.

kisvegabor, 2023-08-27:
On ESP32 larger images can be loaded into the external PSRAM which is much slower. Anyway, with the new MEMFS feature, you have to opportunity to easily  treat buffer in RAM or flash as files if you wish.

kisvegabor, 2023-09-11:
https://github.com/lvgl/lvgl/pull/4462 is merged, so I close this issue.

Thank you all for the contribution! :blush:

closed 2023-09-26 by kisvegabor