lvgl/lvgl #4453: label not properly aligned after text change

hyansuper, 2023-08-13:
```python
import lvgl as lv
import display_driver

label = lv.label(lv.scr_act())
label.set_text("AAAAAAA")
label.center()

label2 = lv.label(lv.scr_act())
label2.align_to(label, lv.ALIGN.OUT_BOTTOM_MID, 0, 10)
label2.set_text("BBBBBBBBBBBBBB")    # not centered ?
```
[image]

same result in C if I use `label_set_text()` **after** `lv_obj_align_to()`

kisvegabor, 2023-08-16:
Ah, yes! I thought it's `lv_obj_align()`. `lv_obj_align_to()` really can't update alignment when the size of the reference or the target object changes. 

Unfortunately, it'd be quite difficult to fix. Instead I suggest using the [flex](https://docs.lvgl.io/master/layouts/flex.html) layout if possible.

hyansuper, 2023-08-17:
ok, I can simply call `lv_obj_align_to()` again after size change

closed 2023-08-17 by hyansuper