lvgl/lvgl #4456: Access Violation Writing Error

kdschlosser, 2023-08-14:
I am not sure what is causing this error. It is internal to LVGL as it happens when I move a slider.

This is using SDL so all of the display and indev related bits are internal to LVGL.

Here is the debugging stack trace from Visual Studio

```shell
_lib_lvgl.dll!lv_event_send(lv_event_list_t * list=0x0000021d173df588, _lv_event_t * e=0x000000f7f4fee708, bool preprocess=false) Line 72
	at *****\lv_cpython\src\lvgl\src\misc\lv_event.c(72)
_lib_lvgl.dll!event_send_core(_lv_event_t * e=0x000000f7f4fee708) Line 310
	at *****\lv_cpython\src\lvgl\src\core\lv_obj_event.c(310)
_lib_lvgl.dll!lv_obj_send_event(_lv_obj_t * obj=0x0000021d1789dc90, lv_event_code_t event_code=LV_EVENT_VALUE_CHANGED, void * param=0x0000000000000000) Line 64
	at *****\lv_cpython\src\lvgl\src\core\lv_obj_event.c(64)
_lib_lvgl.dll!update_knob_pos(_lv_obj_t * obj=0x0000021d1789dc90, bool check_drag=true) Line 476
	at *****\lv_cpython\src\lvgl\src\widgets\slider\lv_slider.c(476)
_lib_lvgl.dll!lv_slider_event(const _lv_obj_class_t * class_p=0x00007ffbc9933560, _lv_event_t * e=0x000000f7f4fee9a8) Line 140
	at *****\lv_cpython\src\lvgl\src\widgets\slider\lv_slider.c(140)
_lib_lvgl.dll!lv_obj_event_base(const _lv_obj_class_t * class_p=0x0000000000000000, _lv_event_t * e=0x000000f7f4fee9a8) Line 89
	at *****\lv_cpython\src\lvgl\src\core\lv_obj_event.c(89)
_lib_lvgl.dll!event_send_core(_lv_event_t * e=0x000000f7f4fee9a8) Line 307
	at *****\lv_cpython\src\lvgl\src\core\lv_obj_event.c(307)
_lib_lvgl.dll!lv_obj_send_event(_lv_obj_t * obj=0x0000021d1789dc90, lv_event_code_t event_code=LV_EVENT_PRESSING, void * param=0x0000021d16aa2430) Line 64
	at *****\lv_cpython\src\lvgl\src\core\lv_obj_event.c(64)
_lib_lvgl.dll!indev_proc_press(_lv_indev_t * indev=0x0000021d16aa2430) Line 1083
	at *****\lv_cpython\src\lvgl\src\indev\lv_indev.c(1083)
_lib_lvgl.dll!indev_pointer_proc(_lv_indev_t * i=0x0000021d16aa2430, lv_indev_data_t * data=0x000000f7f4feeab8) Line 557
	at *****\lv_cpython\src\lvgl\src\indev\lv_indev.c(557)
_lib_lvgl.dll!lv_indev_read_timer_cb(_lv_timer_t * timer=0x0000021d17398eb0) Line 207
	at *****\lv_cpython\src\lvgl\src\indev\lv_indev.c(207)
_lib_lvgl.dll!lv_timer_exec(_lv_timer_t * timer=0x0000021d17398eb0) Line 348
	at *****\lv_cpython\src\lvgl\src\misc\lv_timer.c(348)
_lib_lvgl.dll!lv_timer_handler() Line 114
	at *****\lv_cpython\src\lvgl\src\misc\lv_timer.c(114)
_lib_lvgl.dll!lv_task_handler() Line 47
	at *****\lv_cpython\src\lvgl\src\lv_api_map.h(47)
_lib_lvgl.dll!py_lv_task_handler() Line 7169
	at *****\lv_cpython\build\temp.win-amd64-cpython-311\dllmain.c(7169)
libffi-8.dll!00007ffc6c6e4771()
libffi-8.dll!00007ffc6c6e4493()
libffi-8.dll!00007ffc6c6e42c2()
_ctypes.pyd!_call_function_pointer(int flags=0x00001101, int(*)() pProc=0x00007ffbc94f4920, void * * avalues=0x000000f7f4feee60, _ffi_type * * atypes=0x0000021d17366fd0, _ffi_type * restype, void * resmem=0x000000f7f4feee60, int argcount=0x00000000, int) Line 926
	at D:\a\1\s\Modules\_ctypes\callproc.c(926)
_ctypes.pyd!_ctypes_callproc(int(*)() pProc=0x00007ffbc94f4920, _object * argtuple, IUnknown * pIunk=0x0000000000000000, _GUID * iid=0x0000000000000000, int flags=0x00001101, _object * argtypes=0x00007ffbca01c5d8, _object * restype=0x0000021d17366fd0, _object * checker=0x0000000000000000) Line 1262
	at D:\a\1\s\Modules\_ctypes\callproc.c(1262)
_ctypes.pyd!PyCFuncPtr_call(PyCFuncPtrObject * self=0x0000021d1a1a29c0, _object * inargs=0x0000000000000000, _object * kwds=0x0000000000000000) Line 4201
	at D:\a\1\s\Modules\_ctypes\_ctypes.c(4201)
python311.dll!_PyObject_MakeTpCall(_ts * tstate=0x00007ffbca036960, _object * callable=0x0000021d1a1a29c0, _object * const * args=0x0000021d16d30110, __int64 nargs=0x0000000000000000, _object * keywords=0x0000000000000000) Line 216
	at D:\a\1\s\Objects\call.c(216)
[Inline Frame] python311.dll!_PyObject_VectorcallTstate(_ts *) Line 90
	at D:\a\1\s\Include\internal\pycore_call.h(90)
python311.dll!PyObject_Vectorcall(_object * callable=0x0000021d1a1a29c0, _object * const * args=0x0000021d16d30110, unsigned __int64 nargsf, _object * kwnames=0x0000000000000000) Line 299
	at D:\a\1\s\Objects\call.c(299)
python311.dll!_PyEval_EvalFrameDefault(_ts * tstate=0x00007ffbca036960, _PyInterpreterFrame * frame=0x0000021d16fb5900, int throwflag=0xca036960) Line 4778
	at D:\a\1\s\Python\ceval.c(4778)
[Inline Frame] python311.dll!_PyEval_EvalFrame(_ts *) Line 73
	at D:\a\1\s\Include\internal\pycore_ceval.h(73)
python311.dll!_PyEval_Vector(_ts * tstate=0x00007ffbca036960, PyFunctionObject * func, _object * locals, _object * const * args, unsigned __int64 argcount=0x0000000000000000, _object * kwnames=0x0000000000000000) Line 6443
	at D:\a\1\s\Python\ceval.c(6443)
python311.dll!PyEval_EvalCode(_object * co, _object * globals=0x0000021d16fb5900, _object * locals=0x0000021d16fb5900) Line 1155
	at D:\a\1\s\Python\ceval.c(1155)
python311.dll!run_eval_code_obj(_ts * tstate=0x00007ffbca036960, PyCodeObject * co=0x0000021d17322eb0, _object * globals=0x0000021d16fb5900, _object * locals=0x0000021d16fb5900) Line 1715
	at D:\a\1\s\Python\pythonrun.c(1715)
python311.dll!run_mod(_mod * mod, _object * filename, _object * globals=0x0000021d16fb5900, _object * locals=0x0000021d16fb5900, PyCompilerFlags * flags=0x000000f7f4fef610, _arena * arena=0x0000021d16edbb70) Line 1736
	at D:\a\1\s\Python\pythonrun.c(1736)
python311.dll!pyrun_file(_iobuf * fp=0x0000021d16b3afa0, _object * filename=0x0000021d16eecf10, int start, _object * globals=0x0000021d16fb5900, _object * locals=0x0000021d16fb5900, int closeit=0x00000001, PyCompilerFlags * flags=0x000000f7f4fef610) Line 1630
	at D:\a\1\s\Python\pythonrun.c(1630)
python311.dll!_PyRun_SimpleFileObject(_iobuf * fp=0x0000021d16b3afa0, _object * filename=0x0000021d16eecf10, int closeit=0x00000001, PyCompilerFlags * flags=0x000000f7f4fef610) Line 443
	at D:\a\1\s\Python\pythonrun.c(443)
python311.dll!_PyRun_AnyFileObject(_iobuf * fp=0x0000021d16b3afa0, _object * filename=0x0000021d16eecf10, int closeit=0x00000001, PyCompilerFlags * flags=0x000000f7f4fef610) Line 79
	at D:\a\1\s\Python\pythonrun.c(79)
[... 152 lines cut ...]
	cnt	0x00000001	unsigned int
	preprocess	false	bool
```

and finally the exact error

```
Exception thrown at 0x000001D8B7E00FC0: 0xC0000005: Access violation writing location 0x00007FFBC96C4920.
```

This one has me baffled, I am not sure what is causing it.

The latest development version of the Python binding can be gotten from here.

https://github.com/kdschlosser/lv_cpython/tree/develop

This error occurs when running the slider_1 example. Make sure if you already have LVGL installed in Python to remove it before installing the development version. You can compile a debugging version if you use the following command
```
python -m setup.py build -g
```

Locate the built components under the build directory. 
To run the example in it's current location you will need to copy the lvgl folder from the build directory into the root of the cloned repo overwriting any files and folders if it asks. 

Any insight will be really helpful..

kdschlosser, 2023-08-14:
I did want to mention the same error also happens for the btn_3 example but it does not happen in the arc_2 example. 

If I manually trigger the event using 

```python
lv.obj_send_event(slider, lv.EVENT_VALUE_CHANGED, None)
```

It works like it should. The problem only seems to come from an event generated by a mouse click

kisvegabor, 2023-08-16:
It seems LVGL's memory is messed up somehow. 

To find the issue please do the followings:
- Use the latest LVGL version from `master`
- Enable `LV_ASSERT_...`s in `lv_conf.h`
- Enable `LV_USE_LOG` and `LV_LOG_PRINTF`
- Be sure to use `LV_USE_STDLIB_MALLOC    LV_STDLIB_CLIB`
- Enable ASAN. (add `-fsanitize=address`, see [this](https://github.com/lvgl/lvgl/blob/master/tests/CMakeLists.txt) as an example)

kdschlosser, 2023-08-16:
I have tried to enable the logging. for some reason the callback to output the log information doesn't get called. I will try it again to see what happens.

I am using MSVC to compile with and I don't think there is the -fsanitize=address option. 

I will check and see what is enabled for asserts and make sure the stdlib is being used for malloc.

kdschlosser, 2023-08-16:
``c
#define LV_USE_STDLIB_MALLOC    LV_STDLIB_CLIB
#define LV_USE_STDLIB_STRING    LV_STDLIB_CLIB
#define LV_USE_STDLIB_SPRINTF   LV_STDLIB_CLIB
```

```c
#define LV_USE_ASSERT_NULL          1   /*Check if the parameter is NULL. (Very fast, recommended)*/
#define LV_USE_ASSERT_MALLOC        1   /*Checks is the memory is successfully allocated or no. (Very fast, recommended)*/
#define LV_USE_ASSERT_STYLE         0   /*Check if the styles are properly initialized. (Very fast, recommended)*/
#define LV_USE_ASSERT_MEM_INTEGRITY 0   /*Check the integrity of `lv_mem` after critical operations. (Slow)*/
#define LV_USE_ASSERT_OBJ           0   /*Check the object's type and existence (e.g. not deleted). (Slow)*/
``` 

The MALLOC is already enabled. I will do the rest and enable logging. and see what comes of it.

kdschlosser, 2023-08-16:
I have logging turned on and I do not get a callback to print the logging data

kdschlosser, 2023-08-16:
```c
/*Enable the log module*/
#define LV_USE_LOG 1
#if LV_USE_LOG

    /*How important log should be added:
    *LV_LOG_LEVEL_TRACE       A lot of logs to give detailed information
    *LV_LOG_LEVEL_INFO        Log important events
    *LV_LOG_LEVEL_WARN        Log if something unwanted happened but didn't cause a problem
    *LV_LOG_LEVEL_ERROR       Only critical issue, when the system may fail
    *LV_LOG_LEVEL_USER        Only logs added by the user
    *LV_LOG_LEVEL_NONE        Do not log anything*/
    #define LV_LOG_LEVEL LV_LOG_LEVEL_WARN

    /*1: Print the log with 'printf';
    *0: User need to register a callback with `lv_log_register_print_cb()`*/
    #define LV_LOG_PRINTF 0

    /*1: Enable print timestamp;
     *0: Disable print timestamp*/
    #define LV_LOG_USE_TIMESTAMP 0

    /*Enable/disable LV_LOG_TRACE in modules that produces a huge number of logs*/
    #define LV_LOG_TRACE_MEM        1
    #define LV_LOG_TRACE_TIMER      1
    #define LV_LOG_TRACE_INDEV      1
    #define LV_LOG_TRACE_DISP_REFR  1
    #define LV_LOG_TRACE_EVENT      1
    #define LV_LOG_TRACE_OBJ_CREATE 1
    #define LV_LOG_TRACE_LAYOUT     1
    #define LV_LOG_TRACE_ANIM       1
    #define LV_LOG_TRACE_MSG        1

#endif  /*LV_USE_LOG*/
```

kisvegabor, 2023-08-16:
It might get work if you just enable LV_LOG_PRINTF

kdschlosser, 2023-08-16:
I got it. Here is the log.

<details>
  <summary>Click me</summary>
  
```shell

0[Trace] lv_malloc: allocating 48 bytes 	(in lv_mem.c line #70)
0[Trace] lv_malloc: allocated at 000002582DA1A580 	(in lv_mem.c line #94)
0[Trace] lv_malloc: allocating 40 bytes 	(in lv_mem.c line #70)
0[Trace] lv_malloc: allocated at 000002582EB4BFF0 	(in lv_mem.c line #94)
0[Trace] lv_malloc: allocating 832 bytes 	(in lv_mem.c line #70)
0[Trace] lv_malloc: allocated at 000002582E6A7950 	(in lv_mem.c line #94)
0[Trace] lv_malloc: allocating 112 bytes 	(in lv_mem.c line #70)
0[Trace] lv_malloc: allocated at 000002582E6A3CE0 	(in lv_mem.c line #94)
0[Trace] lv_malloc: allocating 48 bytes 	(in lv_mem.c line #70)
0[Trace] lv_malloc: allocated at 000002582DA1AE40 	(in lv_mem.c line #94)
0[Trace] lv_malloc: allocating 1728 bytes 	(in lv_mem.c line #70)
0[Trace] lv_malloc: allocated at 000002582E2385A0 	(in lv_mem.c line #94)
0[Trace] lv_malloc: allocating 20 bytes 	(in lv_mem.c line #70)
0[Trace] lv_malloc: allocated at 000002582D989B60 	(in lv_mem.c line #94)
0[Trace] lv_realloc: reallocating 000002582D989B60 with 30 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582EB4C020 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582EB4C020 with 40 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582EB4C3B0 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582EB4C3B0 with 50 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582DA1B100 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582DA1B100 with 60 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582E9F50D0 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582E9F50D0 with 70 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582E9F5CB0 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582E9F5CB0 with 80 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582DA18BF0 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582DA18BF0 with 90 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582DA19C80 	(in lv_mem.c line #138)
0[Trace] lv_malloc: allocating 20 bytes 	(in lv_mem.c line #70)
0[Trace] lv_malloc: allocated at 000002582D989CC0 	(in lv_mem.c line #94)
0[Trace] lv_realloc: reallocating 000002582D989CC0 with 30 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582EB4BF90 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582EB4BF90 with 40 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582EB4C3E0 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582EB4C3E0 with 50 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582DA1AD00 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582DA1AD00 with 60 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582E9F5350 	(in lv_mem.c line #138)
0[Trace] lv_malloc: allocating 20 bytes 	(in lv_mem.c line #70)
0[Trace] lv_malloc: allocated at 000002582D989C40 	(in lv_mem.c line #94)
0[Trace] lv_realloc: reallocating 000002582D989C40 with 30 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582EB4C410 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582EB4C410 with 40 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582EB4C3B0 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582EB4C3B0 with 50 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582DA1A5C0 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582DA1A5C0 with 60 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582E9F54E0 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582E9F54E0 with 70 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582E9F5170 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582E9F5170 with 80 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582DA191F0 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582DA191F0 with 90 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582DA19EB0 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582DA19EB0 with 100 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582DA19F20 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582DA19F20 with 110 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582E6BBDD0 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582E6BBDD0 with 120 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582E67EF50 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582E67EF50 with 130 size 	(in lv_mem.c line #122)
0[Trace] lv_realloc: reallocated at 000002582E9738E0 	(in lv_mem.c line #138)
0[Trace] lv_realloc: reallocating 000002582E9738E0 with 140 size 	(in lv_mem.c line #122)
[... 1030 lines cut ...]
0[Trace] lv_malloc: allocated at 000002582E24C900 	(in lv_mem.c line #94)
0[Trace] lv_anim_start: begin 	(in lv_anim.c line #80)
0[Trace] lv_malloc: allocating 136 bytes 	(in lv_mem.c line #70)
0[Trace] lv_malloc: allocated at 000002582E23DE40 	(in lv_mem.c line #94)
0[Trace] lv_anim_start: finished 	(in lv_anim.c line #111)
[... cut ...]
0[Trace] event_send_core: Sending event 28 to 000002582E211770 with 0000000000000000 param 	(in lv_obj_event.c line #290)

Traceback (most recent call last):
  File "C:\Users\drsch\PycharmProjects\lv_cpython\examples\slider_1.py", line 95, in <module>
    lv.task_handler()
  File "C:\Users\drsch\PycharmProjects\lv_cpython\lvgl\__init__.py", line 81845, in task_handler
    res = _lib_lvgl.py_lv_task_handler()
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
OSError: [WinError -1073741795] Windows Error 0xc000001d

```
</details>

kisvegabor, 2023-08-17:
Can I compile the binding on Linux? If so I can enable ASAN which should help too find the problem.

kdschlosser, 2023-08-17:
idk I haven't tested it yet on Linux. it's in the develop branch of the repo.

kisvegabor, 2023-08-18:
> idk I haven't tested it yet on Linux. it's in the develop branch of the repo.

Could you provide some guide about compiling the binding on Linux?

kdschlosser, 2023-08-19:
Before you compile LVGL change the config in the `src` directory.

uninstall any previously installed versions of the library
'python -m pip uninstall lvgl'

Make sure you are in the root directory of the cloned repo.

Then run the following command to compile it but not install it. 

'python -m setup.py build'

That will build it into the build directory. Traverse the build directory until you locate the `lvgl` directory and copy the contents of that directory into the `lvgl` directory in the root of the repo. You should be asked if you want to overwrite, say yes. 

Once that is done you will be able to run the examples in place

You will need to add the following code to the slider_1 example in order to get the logging output from LVGL.

add this at the top just after the imports. but before the first try/except block

```python
class STDOut(object):

    def __init__(self):

        self._stdout = sys.stdout
        sys.stdout = self

    def flush(self):
        self._stdout.flush()

    def write(self, d):
        if ' object ' not in d and ('<' not in d and '>' not in d) and 'lvgl.' not in d:
            if d.strip():
                self._stdout.write(d)
                self.flush()

stdout = STDOut()
```

and add this right before `sdl_window_create` is called

```python
def logging_callback(level, buf):

    data = b''

    for i in range(512):
        char = bytes(buf.__obj__[i])
        if char == b'\x00':
            break
        data += char
        
    print(data.decode('utf-8'))

lv.log_register_print_cb(logging_callback)
```

The first block of code filters out any debugging I have going on in the binding so only the log from LVGL will get output.

kdschlosser, 2023-08-19:
OK so I have come to an issue I have no way of knowing how to solve without some pretty heavy modification to LVGL.

When accessing any kind of a pointer I am able to get the object the pointer is pointing to but here is the catch. ctypes in python creates a new python object each time the object is gotten. What that means is this.

so say I do the following.

```python
obj1 = lv.object_create(None)
obj2 = lv.object_create(obj1)

for i in range(lv.obj_get_child_cnt(obj1):
    child = lv.obj_get_child(obj1, i)
    if child == obj2:
        print('This is never going to happen')
    
    child2 = lv.obj_get_child(obj1, i)

    if child1 == child2:
        print('this is not going to ever happen either')
    
```

There is no way to perform equality testing. Not with anything passed as a pointer. and by definition if it is not passed as a pointer a copy is made so that wouldn't work either.

ctypes in doing something screwy because even memory addresses don't point to the same locations. 

This is a huge issue because of reference counting and garbage collection. If the original object goes out of scope for some reason because the user believes that they have a reference to the object the program is going to crash. 

What a fantastic design I will say.... The object has to be passed by reference so why the original python object isn't tied to that reference is beyond me. if that specific memory location has a python object attached to it then the original object is what should be returned, not a new instance of it that has a completely different memory location.

The only way I can see handling this kind of thing is if  a unique id field is added to structures and unions and a new id gets assigned to each one that gets created

kisvegabor, 2023-08-19:
Uhh, that's really odd...

 However, I'm a little bit optimistic, because I can't believe that others didn't have the same issue and there is no solution for that. 

I have zero experience with ctypes, so I can help only with keeping up your faith. :slightly_smiling_face: 

Actually I tried something. I copy pasted your comment as it is to ChatGPT-4. This is the answer:

----------

The issue you are encountering is due to the fact that when you use ctypes in Python to interface with C code, every time you retrieve a pointer to an object in C, a new Python object wrapping that pointer is created. Since these Python objects are distinct, even if they wrap the same pointer in C, equality checks will fail in Python.

One solution to this issue is to maintain a mapping of the C pointers to their corresponding Python objects. Here's an example of how you can achieve this using Python's `weakref` module:

```python
import weakref

# Create a dictionary to store the mapping of C pointers to Python objects
pointer_mapping = weakref.WeakValueDictionary()

# Create a function to wrap the object creation process and store the mapping
def create_object(parent):
    obj = lv.obj_create(parent)
    # Use the memory address of the C pointer as the key
    pointer_mapping[id(obj)] = obj
    return obj

# Create a function to get the object by its C pointer
def get_object_by_pointer(ptr):
    # Use the memory address of the C pointer as the key
    return pointer_mapping.get(id(ptr))

# Your code example
obj1 = create_object(None)
obj2 = create_object(obj1)

for i in range(lv.obj_get_child_cnt(obj1)):
    child = lv.obj_get_child(obj1, i)
    if child == obj2:
        print('This is now going to happen')
    
    child2 = lv.obj_get_child(obj1, i)
    if child == child2:
        print('This is going to happen too')
```

In this code, the `create_object` function is a wrapper around `lv.obj_create`. When an object is created, it is stored in the `pointer_mapping` dictionary. The `get_object_by_pointer` function retrieves the Python object by its C pointer using the memory address of the pointer as the key. The `weakref.WeakValueDictionary` ensures that the objects are removed from the dictionary once they are no longer referenced in Python, helping with garbage collection.

This solution allows you to preserve the identity of Python objects wrapping the same C pointer, enabling equality testing and proper reference counting.

kdschlosser, 2023-08-20:
Most people that use ctypes don't need for a ctype object to stay persistant throught the running of a program. It is usually used for quick and simple function calls and things of that nature. Typically when an entire binding like this gets made it's not done in ctypes. It gets done in Cython, swig or some other middleware that generates the code needed to compile a python c extension. In the extension is the code that handles creating proper types that represent the objects in C so the user can access them in Python.

https://docs.python.org/3/library/ctypes.html

Another example that may behave differently from what one would expect is this:

```python
>>>

s = c_char_p()

s.value = b"abc def ghi"

s.value
b'abc def ghi'

s.value is s.value
False
>>>
```
Note: Objects instantiated from c_char_p can only have their value set to bytes or integers.

Why is it printing False? ctypes instances are objects containing a memory block plus some descriptors accessing the contents of the memory. Storing a Python object in the memory block does not store the object itself, instead the contents of the object is stored. Accessing the contents again constructs a new Python object each time!

And that is where the problem is. No way to identify the object because in the case of a structure the structure itself is a Python object and the data contained within is what gets added to a newly created structure each time the structure is accessed from say a pointer. Or when a structure is returned from a function call.  If I use the example in my previous post with the children and I get the address of the child and then enumerate the children using the function call and I get the memory addresses for each child none of the addresses will match the one that was initially created. After all a structure is simply a container that has a set of rules on how to access the information inside of that container.

It's really frustrating to have this serious glitch in how ctypes works. I am not able to track references properly because the attachment of a python object to a c object simply doesn't exist in ctypes. I am not able to track references of the c objects form inside of python without a layer in the middle being written to do so.

Now here does the question of the day. am I able to extend the structures for the purposes of the binding to add an id field to the structure in order to track it?

Example...

```c
typedef struct _lv_obj_t {
    const lv_obj_class_t * class_p;
    struct _lv_obj_t * parent;
    _lv_obj_spec_attr_t * spec_attr;
    _lv_obj_style_t * styles;
    void * user_data;
    lv_area_t coords;
    lv_obj_flag_t flags;
    lv_state_t state;
    uint16_t layout_inv : 1;
    uint16_t scr_layout_inv : 1;
    uint16_t skip_trans : 1;
    uint16_t style_cnt  : 6;
    uint16_t h_layout   : 1;
    uint16_t w_layout   : 1;
} lv_obj_t;

typedef struct {
    lv_object_t * object;
    uint32_t id;
} py_obj_t
```

Is there a way to get functions that have a parameter with the type of lv_obj_t to accept py_obj_t??? It is a simple cast from py_obj_t to lv_obj_t. I guess the question is this. If I cast the py_obj_t to lv_obj_t and then pass that lv_obj_t to a function. Then I call another function to retrieve that object if I cast the returned value from lv_obj_t to py_obj_t would the id field contain the same number as the one that was initially created?

I know I am able to use the user_data field that some of the structures have for this kind of a purpose but not all structures have this field available. Anything that can be passed to a function or returned from a function or is attached to a field in a structure or union and is a structure or union would need to have a way to identify it by using some form of an id. Arrays I am tinkering about with arrays. const arrays don't matter much because the data is not able to be altered. But with an array that is not a const I believe I can attach a python array.array instance as a buffer and that buffer is mutable and will change if the data gets changed in that array from anywhere inside of the c code.

kdschlosser, 2023-08-20:
It's strange because I am seeing the use of `'id` instead of `ctypes.addressof`. `id` is the memory location of the python object where as `ctypes.addresssof` is the memory location of the underlying c object. I am thinking it would pretty safe to say that `id` is not going to work because the python object is not persistent between calls to collect the object. It states the very fact in the Python documentation for ctypes. The Python object not being persistent is what the problem stems from.

kdschlosser, 2023-08-20:
Another option is to create a series of functions that handle the identification of objects and doing equality comparisons. This can be done using void* types I believe.

```c

void * _id_array[];
void* _obj_array[];
uint32_t _num_id_array_elements;
uint32_t _next_id = 1;

uint32_t lv_get_c_obj_id(void * obj) {
    for (uint32_t i=0;i<_num_id_array_elements;i++) {
        if (_obj_array[i] == obj) {
            return _id_array[i];
        }
    }
    return 0;
}

uint32_t lv_obj_new_id(void * obj) {
    for (uint32_t i=0;i<_num_id_array_elements;i++) {
        if (_obj_array[i] == obj) {
            return 0;
        }
    }
    _id_array[_num_id_array_elements] = obj;
    _obj_array[_num_id_array_elements] = _next_id;
    _next_id++;
    _num_id_array_elements++;
    return _next_id - 1;
}

bool lv_remove_obj_id(void * obj) {
    uint32_t id = 0;
    uint32_t i = 0;
    for (i=0;i<_num_id_array_elements;i++) {
        if (_obj_array[i] == obj) {
            id  = _id_array[i];
            break;
        }
    }
    if (id != 0) {
        uint32_t j;
        for(j = i;j < _num_id_array_elements - 1; j++) {
            _id_array[j] = _id_array[j + 1];
            _obj_array[j] = _obj_array[j + 1];
        }
        _num_id_array_elements--;
        return true;
    }
    return false;
}

bool lv_is_same_obj(void * obj1, void * obj2) {
    if (obj1 == obj2) {
        return true;
    }
    return false;
```

Would that work? Remember I am not proficient at c so be easy one me. LOL. The code is only for idea purposes and it more than likely can be tweaked to get better performance from it. 

The code wouldn't have to be included with lvgl at all and it could be added at the binding level and compiled when LVGL is compiled. I can cast anything to a void * and seeing as how this is specifically looking at only the memory pointers to underlying c objects I can store the id along with a weak reference to a python object. if the python object is about to get deleted a callback gets called and then I could remove the stored id and weak reference and also remove the c_obj from the id storage in c code.

kisvegabor, 2023-08-23:
I really can't comment on this part. :slightly_frowning_face:  It might be worth asking it on Stack Overflow.

kdschlosser, 2023-08-24:
I have been looking through the code that gets generated for MicroPython. The way it is set up is it doesn't care if there are references to objects that are made on the Python side of things. As an example. If I create an object and then create a child to that object but I keep a reference to that child for later use. Say I want to delete the object from being used the child objects become orphaned so to speak. The underlying c object gets deleted but the reference to it in MicroPython still exists. If that child object then gets used an error will get thrown, If the child object never gets used again that would constitute as a memory leak.

The MicroPython binding is adding a pointer to an LV type into a structure and that structure is what is getting returned. Hence the reason why I am not able to access anything in the returned type when calling `lv.obj()` It's because the returned type is not `lv_obj_t`, instead it's type is `mp_lv_obj_t`

```c
typedef struct mp_lv_obj_t {
    mp_obj_base_t base;
    LV_OBJ_T *lv_obj;
    LV_OBJ_T *callbacks;
} mp_lv_obj_t;
```

closed 2023-09-15 by lvgl-bot