lvgl/lvgl #4463: Support SIMD instructions for rendering acceleration

terry0012, 2023-08-17:
### Introduce the problem

Some chip architectures support SIMD acceleration instructions, such as MVE on Cortex-M55 and NEON on Cortex-A7.

To use SIMD acceleration, a common usage pattern is to embed SIMD code into the software draw unit using some macros.

Another way is to follow the usage of GPU and implement LVGL draw unit with SIMD, but when GPU and SIMD acceleration instructions exist at the same time, the current draw unit does not have a fallback mechanism, and a custom fallback mechanism  and another draw unit with SIMD need to be implemented in the GPU draw unit.

Generally speaking, when a single chip supports SIMD acceleration instructions, SIMD can completely replace the implementation function of software draw unit. Therefore, I think it is more appropriate to embed SIMD code into the software draw unit. 

### Examples and cases

this is the example in sw blend if we emed SIMD code into it. [refer to the libyuv repository](https://github.com/lemenkov/libyuv)

```

void argb8888_fill_row_c(const uint32_t color32,  uint32_t * dest_buf, int width) {
            int x;
            for(x = 0; x < w - 16; x += 16) {
                dest_buf[x + 0] = color32;
                dest_buf[x + 1] = color32;
                dest_buf[x + 2] = color32;
                dest_buf[x + 3] = color32;

                dest_buf[x + 4] = color32;
                dest_buf[x + 5] = color32;
                dest_buf[x + 6] = color32;
                dest_buf[x + 7] = color32;

                dest_buf[x + 8] = color32;
                dest_buf[x + 9] = color32;
                dest_buf[x + 10] = color32;
                dest_buf[x + 11] = color32;

                dest_buf[x + 12] = color32;
                dest_buf[x + 13] = color32;
                dest_buf[x + 14] = color32;
                dest_buf[x + 15] = color32;
            }
            for(; x < w; x ++) {
                dest_buf[x] = color32;
            }
}

void argb8888_fill_row_neon(const uint32_t color32,  uint32_t * dest_buf, int width) {
    asm volatile(
    ...
    ...
   );
}

void argb8888_fill_row_mve(const uint32_t color32,  uint32_t * dest_buf, int width) {
    asm volatile(
    ...
    ...
   );
}

LV_ATTRIBUTE_FAST_MEM void lv_draw_sw_blend_color_to_argb8888(_lv_draw_sw_blend_fill_dsc_t * dsc)
{
    int32_t w = dsc->dest_w;
    int32_t h = dsc->dest_h;
    lv_opa_t opa = dsc->opa;
    const lv_opa_t * mask = dsc->mask_buf;
    lv_coord_t mask_stride = dsc->mask_stride;
    lv_coord_t dest_stride = dsc->dest_stride;

    lv_color_mix_alpha_cache_t cache;
    lv_color_mix_with_alpha_cache_init(&cache);

    int32_t x;
    int32_t y;

    /*Simple fill*/
    void (*argb8888_fill_row)(const uint32_t color32,  uint32_t * dest_buf, int width) = argb8888_fill_row_c;

    if(mask == NULL && opa >= LV_OPA_MAX) {
        uint32_t color32 = lv_color_to_u32(dsc->color);
        uint32_t * dest_buf = dsc->dest_buf;
        for(y = 0; y < h; y++) {
#ifdef ARCH_HAVE_NEON
            argb8888_fill_row = argb8888_fill_row_neon;
#endif

#ifdef ARCH_HAVE_MVE
            argb8888_fill_row = argb8888_fill_row_mve;
#endif

            argb8888_fill_row(dest_buf, color32, w);
            dest_buf += dest_stride;
        }
    }
    /*Opacity only*/
   ...
   ...
}
```

### Suggested solution

kisvegabor, 2023-08-17:
When I read the title of the issue I had same argument in my head about creating a new draw unit or embedding to the SW draw unit. And I concluded the same. Happy that we are on the same page and other also have a good understanding about the new architecture already. :slightly_smiling_face: 

I guess here 
```c
for(y = 0; y < h; y++) {
#ifdef ARCH_HAVE_NEON
            argb8888_fill_row = argb8888_fill_row_neon;
#endif

#ifdef ARCH_HAVE_MVE
            argb8888_fill_row = argb8888_fill_row_mve;
#endif

            argb8888_fill_row(dest_buf, color32, w);
            dest_buf += dest_stride;
        }
```

you meant:
```c
//Init outside of the loop
#ifdef ARCH_HAVE_NEON
    argb8888_fill_row = argb8888_fill_row_neon;
#endif

#ifdef ARCH_HAVE_MVE
    argb8888_fill_row = argb8888_fill_row_mve;
#endif

    for(y = 0; y < h; y++) {
         argb8888_fill_row(dest_buf, color32, w);
         dest_buf += dest_stride;
    }
```

@igrr, it could be useful for ESP as well, right?

terry0012, 2023-08-18:
> When I read the title of the issue I had same argument in my head about creating a new draw unit or embedding to the SW draw unit. And I concluded the same. Happy that we are on the same page and other also have a good understanding about the new architecture already. 🙂
> 
> I guess here
> 
> ```c
> for(y = 0; y < h; y++) {
> #ifdef ARCH_HAVE_NEON
>             argb8888_fill_row = argb8888_fill_row_neon;
> #endif
> 
> #ifdef ARCH_HAVE_MVE
>             argb8888_fill_row = argb8888_fill_row_mve;
> #endif
> 
>             argb8888_fill_row(dest_buf, color32, w);
>             dest_buf += dest_stride;
>         }
> ```
> 
> you meant:
> 
> ```c
> //Init outside of the loop
> #ifdef ARCH_HAVE_NEON
>     argb8888_fill_row = argb8888_fill_row_neon;
> #endif
> 
> #ifdef ARCH_HAVE_MVE
>     argb8888_fill_row = argb8888_fill_row_mve;
> #endif
> 
>     for(y = 0; y < h; y++) {
>          argb8888_fill_row(dest_buf, color32, w);
>          dest_buf += dest_stride;
>     }
> ```
> 

Yes, you are right, init outside is better.

> @igrr, it could be useful for ESP as well, right?

igrr, 2023-08-18:
> @igrr, it could be useful for ESP as well, right?

> Generally speaking, when a single chip supports SIMD acceleration instructions, SIMD can completely replace the implementation function of software draw unit.

I agree in principle, but for me the question would be related to maintenance. If I were the maintainer of the SW draw unit code, would I want to have to understand and know about all the different architecture specific extensions which could be used there? I'd rather keep different architecture specific implementations as different draw units, so that folks who understand each extension could develop and maintain those.

As for the fallback from GPU to SW draw unit, as long as I can get the pointer to a function of a software draw unit, implementing the fallback doesn't seem to be difficult? We check if GPU can be used, if not — call the software draw unit function, forwarding all the arguments.
(I could be not seeing some important aspect here, please correct me if I'm wrong.)

terry0012, 2023-08-18:
> I agree in principle, but for me the question would be related to maintenance. If I were the maintainer of the SW draw unit code, would I want to have to understand and know about all the different [...]

In fact, there is a lot of redundant code between the draw unit with SIMD and the SW draw unit. The more types of SIMD instructions there are, the more redundant code there will be.

Moreover, when the content of the SW draw unit changes, the draw unit with SIMD also needs to change accordingly, which will bring more trouble to the maintenance of the code.

For example, if we implement a NEON draw unit,  we should execute drawing as below:

```
static void neon_execute_drawing(lv_draw_sw_unit_t * u)
{
    /*Render the draw task*/
    lv_draw_task_t * t = u->task_act;
    switch(t->type) {
        case LV_DRAW_TASK_TYPE_FILL:
            lv_draw_neon_fill((lv_draw_unit_t *)u, t->draw_dsc, &t->area);
            break;
        ...
        ...
        default:
            break;
}
```

 Most of the code in lv_draw_neon_fill is the same as lv_draw_sw_fill , The only difference is in the parallel processing of data during the final blending.

kisvegabor, 2023-08-18:
Hi Ivan,

Thank you for joining! 

Just to see the whole picture, e.g. to render a rectangle:
1. the widget  creates a `draw_task` which will be added to a list
2. each draw unit (SW, GPU, etc) can pick `draw_task`s from the  list during a dispatching process
3. finally the `draw_unit`s needs to draw the rectangle somehow (handle gradient, radius, shadow, border, etc). 
4. In case of the SW render there is are "blend" functions to fill areas with colors or blend images with various color formats.

The ASM code needs to added only in the 4th step. So adding a dispatcher, shadow, gradient, radius etc renderer would be a huge duplication. 

We could duplicate only the blending functions, but still there are many option there, e.g. blending RGB565 to ARGB888 and all the other combinations. Besides only the very core of the loops needs to be in ASM. See the SW blend functions [here](https://github.com/lvgl/lvgl/tree/master/src/draw/sw/blend).

So I agree with @terry0012. From maintenance point of view, it seems simpler to inject the ASM code right into the SW render.

igrr, 2023-08-18:
Okay, would it be possible to inject the architecture-specific implementation via the config file, or at link time?

E.g, instead of
```c++
#ifdef ARCH_HAVE_NEON
    argb8888_fill_row = argb8888_fill_row_neon;
#endif

#ifdef ARCH_HAVE_MVE
    argb8888_fill_row = argb8888_fill_row_mve;
#endif
```

have a config file (provided externally to LVGL) with
```c++
#define argb8888_fill_row argb8888_fill_row_custom_arch
```

Or, simply allow argb8888_fill_row to be defined as an exported function in some other library (again, external to LVGL).

Essentially, I'd like to be able to add a custom SIMD based implementation without having to patch LVGL, and without being dependent on an LVGL release, should we have to fix a bug in the said implementation.

kisvegabor, 2023-08-18:
I understand your motivation and it seems doable, but note that there will be a lot of functions. These are only for RGB565 destination:
```c
rgb565_fill(dest_buf, px_cnt, opa, mask); //mask can be NULL
rgb565_blend_rgb565(dest_buf, src_buf, px_cnt, opa, mask);
rgb565_blend_rgb888(dest_buf, src_buf, px_cnt, opa, mask);
rgb565_blend_rgb888(dest_buf, src_buf, px_cnt, opa, mask, src_px_size); //src_px_size can be 3 or 4 to support RGB888 and XRGB888
rgb565_blend_argb8888(dest_buf, src_buf, px_cnt, opa, mask);
```
The other option is to have 
```c
rgb565_fill_full_cover
rgb565_fill_opa
rgb565_fill_full_cover_mask
rgb565_fill_opa_mask
...
```
But it would be really a lot of functions.

And the same for:
```c
rgb888_*
argb8888_*

//Future plans/ideas
l8_*   //Useful to render masks (Lightness 8 bit)
l1_*  //For monochrome, packed into columns or rows
i8_*  //Very simple, disable all color mixing and just use e.g. the red channel or brightness of the colors 
rgb565a8_* //rgb565 followed by a8
argb4444_* //was requested by customer as it much faster then argb8888
```

I also wonder how will we maintain it when we add support for a new destination format:
1. Have a fallback from ASM to C? That is somehow detect if the the ASM version is missing for a case?
2. Make it configurable per color format? That is you can support e.g. all `rgb565_*` functions and non of the `argb8888_*` ones, but you cannot support only a portion of `rgb888_*` It would be much less config. E.g. 
```c
#define LV_DRAW_SW_RGB565_ASM  NEON/MVE/CUSTOM   //NEON and MVE can be built in, and ESP can use CUSTOM 
```

2) seems cleaner to me.

xiaoxiang781216, 2023-08-21:
how about:
````
#ifdef CONFIG_NEON
#include "neon.h"
#elif defined(xxx)
#include "xxx.h"
#endif

#ifndef rgb565_blend_rgb565
#define rgb565_blend_rgb565 rgb565_blend_rgb565_sw
#endif
....
```
so , all caller don't need change, just sw version need apend _sw suffix, SIMD acceleration could overwrite the drawing macro step by step.

kisvegabor, 2023-08-23:
Good idea! Just one thing: for performance reasons I wouldn't move the C part into functions.  So we can do this:
```c
    for(y = 0; y < h; y++) {
#ifdef rgb565_blend_rgb565
         rgb565_blend_rgb565(...)
#else  
         C implementation as now
#endif
         dest_buf += dest_stride;
    }
```

xiaoxiang781216, 2023-08-23:
But inline function could migrate this issue? it's more flexible to has a dedicated SW version since NEON/MVE version my need fallback to sw.

kisvegabor, 2023-08-25:
If we want to make these static inline functions available for other modules as well, we need to implement them in the header files. And it's not only those `rgb565_blend_rgb565`-like function but several helpers as well to mix colors (e.g.  `lv_color_32_32_mix`).

Why do we need a fallback? Isn't it possible do implement all these in ASM? In the end all these are "just"  multiplications, additions and shifting on arrays.

kisvegabor, 2023-08-28:
To give you a more detailed example, this how I imagine it:

**lv_conf.h**
```c
#define  LV_DRAW_SW_ASM_NEON    0          //Built into LVGL
#define   LV_DRAW_SW_ASM_CUSTOM  1
#define  LV_DRAW_SW_ASM_CUSTOM_INCLUDE    "esp_lvgl_asm.h"
```

**esp_lvgl_asm.h**
```c
void esp_rgb565_belnd_to_rgb565_function(...); //or it can be a static inline function too

#define rgb565_belnd_to_rgb565        esp_rgb565_belnd_to_rgb565_function
```

**esp_lvgl_asm.c**
```c
void rgb565_belnd_to_rgb565_function(...)
{
   ... 
}
```

**lv_draw_sw_belnd_to_rgb565.c**
```c
#if LV_DRAW_SW_ASM_NEON
    #inlcude "lv_neon.h"
#elif LV_DRAW_SW_ASM_CUSTOM
    #include LV_DRAW_SW_ASM_CUSTOM_INCLUDE
#endif

...

//In a draw function
    for(y = 0; y < h; y++) {
#if rgb565_belnd_to_rgb565      //If not defined fall back to SW implementation
     rgb565_belnd_to_rgb565(...)
#else
    SW implementation
#endif   
}
```

@igrr @terry0012 @xiaoxiang781216 please check if it can work for you.

C47D, 2023-08-28:
I would do change this

```c
#define  LV_DRAW_SW_ASM_NEON    0          //Built into LVGL
#define  LV_DRAW_SW_ASM_CUSTOM  1
#define  LV_DRAW_SW_ASM_CUSTOM_INCLUDE    "esp_lvgl_asm.h"
```

To this
```c
#define  LV_DRAW_SW_ASM_NEON    0          //Built into LVGL

#define  LV_DRAW_SW_ASM_CUSTOM  1
#if LV_DRAW_SW_ASM_CUSTOM
#define  LV_DRAW_SW_ASM_CUSTOM_INCLUDE    "esp_lvgl_asm.h"
#endif
```

What do you think?

kisvegabor, 2023-08-28:
> #define  LV_DRAW_SW_ASM_CUSTOM  1
#if LV_DRAW_SW_ASM_CUSTOM
#define  LV_DRAW_SW_ASM_CUSTOM_INCLUDE    "esp_lvgl_asm.h"

I agree!

GorgonMeducer, 2023-08-29:
Sorry guys, but arm-2d has already provided the Helium acceleration for LVGL.
In the future, arm-2d will add NEON support and hence LVGL can benefit from the NEON version automatically without any modification.

kisvegabor, 2023-08-29:
@GorgonMeducer It's interesting. Let's think about what happens if we have 1 CPU core with SW render and arm-2d enabled. As arm-2d is also a software implementation (i.e. there is no real GPU hardware), the SW render and arm-2d can't work in parallel. We can draw only one task at a time. Let's say arm-2d wants to draw a simple fill, it can take that task and draw it. If it can't draw something SW will draw it.

It seems like a good extension for the SW render.  I just wonder how complicated it is to integrate arm-2d to any Cortex-M55 and Cortex-M85 MCU. Let's use the most simple case: I would like to build a project with Make or CMake. How can I integrate arm-2d into the project?

GorgonMeducer, 2023-08-30:
@kisvegabor 
> As arm-2d is also a software implementation (i.e. there is no real GPU hardware), the SW render and arm-2d can't work in parallel. 

That's not true. Arm-2D is a hardware abstraction layer. It is the driver to cover both Helium, DMAC-350, Arm Custom Instruction (ACI) and other 3rd-party GPU. 

Of course, for DMAC-350, ACI and other 3rd-party GPUs, someone can create dedicated acceleration for LVGL, but if people want to use arm-2d to cover all those parts for saving time, I should make sure arm-2d can cover this.

So Arm-2D is not only a software implementation. It depends on the actual hardware layer under arm-2d. 
For example, if there is a DMAC350, then as the official driver of DMAC-350, people might wants to use arm-2d together with Helium to accelerate. In other words, Parallel acceleration is possible. 

NOTE: The previous version of lv_gpu_arm2d.c hasn't implemented parallel processing simply because the previous architecture of LVGL isn't friendly to parallel processing. That's why you are introducing the LVGL parallel render.

>  just wonder how complicated it is to integrate arm-2d to any Cortex-M55 and Cortex-M85 MCU. 

It is actually simple, leave this part to arm-2d:

- Arm-2D will provide the Helium acceleration as the extension of the SW render. 
- Arm-2D will provide the parallel acceleration for DMAC-350 if it is available (and user choose to use arm-2d to manage the DMAC350)
- Arm-2D will provide the synchronous acceleration for ACI as the extension of the SW render if it is available (and user choose to use arm-2d to manage the ACI)
- Arm-2D can provide the parallel acceleration for 3rd-party GPU if it is available (i.e. the hardware and the arm-2d driver is ready) and the user chooses to use arm-2d to manage the acceleration.

For practice, when you enabled arm-2d support for M55 and M85, this will happen: Arm-2D acceleration consists of two parts: the helium (and ACI) extension for SW render and a PE:
- The Helium (and ACI) SW render extension part can use the method discussed in this issue to enhance/embed the Helium acceleration. And when necessary, arm-2d adaption layer might hack some part of the SW render to take advantage of the arm-2d library features. 

- PE: Arm-2D will implement a PE to support the DMAC-350 or other 3rd-party GPUs that choose to use arm-2d as their default driver. 

So in the view of LVGL, arm-2d will enhance the SW render and add a PE for parallel processing. This means, there is no extra complexity left for LVGL to handle, i.e. we are using the standard model of acceleration proposed by LVGL9 new architecture. 

I hope this helps.

GorgonMeducer, 2023-08-30:
> I would like to build a project with Make or CMake. How can I integrate arm-2d into the project?

It should be the same or similar to what LVGL8 does, e.g. including a header file `lv_gpu_arm2d.h` and adding a C source file called `lv_gpu_arm2d.c`. 

The `lv_gpu_arm2d.h` can provide information on inline acceleration for the SW render.

kisvegabor, 2023-08-30:
> So Arm-2D is not only a software implementation. It depends on the actual hardware layer under arm-2d.
For example, if there is a DMAC350, then as the official driver of DMAC-350, people might wants to use arm-2d together with Helium to accelerate. In other words, Parallel acceleration is possible.

I see, in this case we can really consider it as GPU hardware.

@xiaoxiang781216 @terry0012, do you still think that we need to add built in ASM support Helium? 
I think "custom asm" part it's still important to allow using ASM for any vendor, like Espressif. 

> By the way, arm-2d has built-in support for cmake:

That's great! So just copy paste ARM-software/Arm-2D, and compile it (just like LVGL). I was afraid that it works only with CMSIS-PACK.

GorgonMeducer, 2023-08-30:
> That's great! So just copy paste ARM-software/Arm-2D, and compile it (just like LVGL). I was afraid that it works only with CMSIS-PACK.

Sorry about creating such misunderstanding. If you have any issue during cmake compilation, please let me know or sending PR to fix the CmakeList.txt.

GorgonMeducer, 2023-08-30:
> do you still think that we need to add built in ASM support Helium?
I think "custom asm" part it's still important to allow using ASM for any vendor, like Espressif.

This is also important about how arm-2d enhance the SW renderer with inline intrinsics 

I think intriniscs is a better name for the inline asm in this case, this is because : 
- intrinsics is function-like
- intrinsics is usually implemented as inline assembly 
- intrinsics can also be implemented as functions

So I propose to call it the LVGL Low Level intrinsics rather than ASM.

xiaoxiang781216, 2023-08-30:
> I see, in this case we can really consider it as GPU hardware.

But arm-2d focus on arm owned IP and is controlled by ARM not driven by an open-source community.

> @xiaoxiang781216 @terry0012, do you still think that we need to add built in ASM support Helium? I think "custom asm" part it's still important to allow using ASM for any vendor, like Espressif.
> 

Yes, for example, I don't see that arm-2d has any plan to support RISCV vector extension(https://github.com/riscv/riscv-v-spec/blob/master/v-spec.adoc) which is very popular on the recent RISCV chip. Espressif ship many SoC with Xtensa DSP, which also has the dedicated SIMD instruction.

BTW, we found that GPU optimization is fast in some case, but ASM optimization is fast in other case. Since LVGL only support GPU render fallback to SW render, but not support GPU(Hardware)->GPU(Software) fallback, GPU software can't cover the case I just mention.

So, I think both approaches are useful.

GorgonMeducer, 2023-08-30:
> But arm-2d focus on arm owned IP and is controlled by ARM not driven by an open-source community.

I don't understand why this stops Arm-2d from being a PE.  Almost all GPUs belong to some 3rd-party but not open-source community. Am I right? 

> Yes, for example, I don't see that arm-2d has any plan to support RISCV vector extension

Did I mention anything that Arm-2D providing support for Helium stops LVGL from supporting other SIMD support?

In fact, I have mentioned that

>> do you still think that we need to add built in ASM support Helium?
>> I think "custom asm" part it's still important to allow using ASM for any vendor, like Espressif.

> This is also important about how arm-2d enhance the SW renderer with inline intrinsic

That means, the method you discussed here can be used by arm-2d to accelerate Helium for LVGL.

Arm-2D doesn't want to control LVGL or stop other architecture from accelerating LVGL. 
Why you have such misunderstanding? Please point out that which parts of my saying giving you this feeling? If there are any, I will do my apology. If there is non, please allow me to make the following statement:

- Arm-2D isn't something so special that wants to control how LVGL implementing its SIMD acceleration
- Arm-2D is here to provide acceleration just for Arm processors. Arm-2D has no plan or capability to stop LVGL from using other SIMD architecture
- Arm-2D will use the method you guys discussed here to enhance SW render.

I will quit this discussion to avoid any further understanding. Once the solution is done, Arm-2D will follow the method used by LVGL community.

GorgonMeducer, 2023-09-14:
not stale.

kisvegabor, 2023-10-02:
The discussion continues in #4463

closed 2023-10-02 by kisvegabor

GorgonMeducer, 2023-10-02:
Hi @kisvegabor 
This is 4463.... and you closed it.

kisvegabor, 2023-10-02:
Thanks, updated.