lvgl/lvgl #4465: Can't touch LV_EVENT_DEFOCUSED.

qq8512852, 2023-08-18:
### Perform all steps below and tick them with [x]
- [ yes] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [yes ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ x] Update lvgl to the latest version
- [yes ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

LVGL version: 8.1.0
When I create a textarea in an img object（lv_img_create()）, I can't get LV_EVENT_DEFOCUSED when clicked out of the textarea. But it works when I create it with obj (lv_obj_create()).
### To Reproduce

    scr = lv_obj_create(parent);
    MainScrRoot = lv_img_create(scr);
    lv_img_set_src(MainScrRoot, &iot);
    lv_obj_align(MainScrRoot, LV_ALIGN_TOP_LEFT, 0, 0);
   textarea_account = lv_textarea_create(MainScrRoot); 
   lv_obj_add_event_cb(textarea_account, textarea_event_cb, LV_EVENT_FOCUSED, NULL);
  lv_obj_add_event_cb(textarea_account, textarea_event_cb, LV_EVENT_DEFOCUSED, NULL);
### Expected behavior

get LV_EVENT_DEFOCUSED

### Screenshots or video

huenrong, 2023-08-18:
I have a similar problem, but my textarea is a child object of tabview, i use lvgl 8.3.7

kisvegabor, 2023-08-18:
Does using `lv_obj_add_flag(MainScrRoot, LV_OBJ_FLAG_CLICKABLE)` solve the issue?

huenrong, 2023-08-19:
@kisvegabor 
I think the problem is that the parent object of the textarea does not have the `LV_OBJ_FLAG_CLICK_FOCUSABLE` flag, example code:
```
lv_obj_t *tabview = lv_tabview_create(page, LV_DIR_TOP, 40);
lv_obj_t *set_param_tab = lv_tabview_add_tab(tabview, "set_param");
lv_obj_add_flag(set_param_tab, LV_OBJ_FLAG_CLICK_FOCUSABLE); // Without this line local_ip_textarea cannot receive LV_EVENT_DEFOCUSED
lv_obj_t *local_ip_textarea = lv_textarea_create(set_param_tab);
```

kisvegabor, 2023-08-19:
True, both `LV_OBJ_FLAG_CLICK_FOCUSABLE` and `LV_OBJ_FLAG_CLICKABLE` needs to be enabled.

qq8512852, 2023-08-20:
> Does using `lv_obj_add_flag(MainScrRoot, LV_OBJ_FLAG_CLICKABLE)` solve the issue?
Much appreciated. After tested, LV_OBJ_FLAG_CLICKABLE is enough.

qq8512852, 2023-08-21:
Done

closed 2023-08-21 by qq8512852