lvgl/lvgl #4467: LVGL makes me confused now!!!

incity, 2023-08-18:
Why lv_obj_get_x return 0?

```
int main(int, char**) {
    /*Initialize LVGL*/
    lv_init();

    /*Initialize the HAL (display, input devices, tick) for LVGL*/
    hal_init();
    lv_obj_t* screen = lv_scr_act();
    printf("%d %d %d %d\n", lv_obj_get_x(screen), lv_obj_get_y(screen), lv_obj_get_width(screen), lv_obj_get_height(screen));
    lv_obj_t* title = lv_btn_create(lv_scr_act());
    lv_obj_set_x(title, 100);

    printf("lv_obj_get_x %d\n", lv_obj_get_x(title));

    printf("%d %d %d %d\n", lv_obj_get_x(screen), lv_obj_get_y(screen), lv_obj_get_width(screen), lv_obj_get_height(screen));

    //app_main();

    while(1) {
        /* Periodically call the lv_task handler.
        * It could be done in a timer interrupt or an OS task too.*/
        lv_timer_handler();
        usleep(5 * 1000);
    }
}
```

[image]

[image]

kisvegabor, 2023-08-18:
Hi,

See here: https://docs.lvgl.io/8.3/overview/coords.html#postponed-coordinate-calculation

We should figure out something for this in v9.

closed 2023-08-18 by kisvegabor

incity, 2023-08-21:
v9 has fixed?