lvgl/lvgl #4474: How to optimize the transition of image backgrounds to make it smoother?

qq8512852, 2023-08-21:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [no] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

LVGL version:8.1.0 
When the page changes, it is not smooth with the simulator on the Windows platform.
Please refer to the following video.

### To Reproduce

Images info: (3 background images are 800*480) 
[image]
lv_config.c   refresh period changed from 30 to 10 ms
[image]

### Expected behavior

Achieve smoother transition of images without resizing the image.

### Screenshots or video

https://github.com/lvgl/lvgl/assets/15918325/1dd5a9f7-3cc4-4a1f-bb60-c1b69b4fdecf

kisvegabor, 2023-08-23:
If these are backgrouud images  you should use `LV_IMG_CD_TRUE_COLOR` format. It should be much faster this way.

qq8512852, 2023-08-23:
> If these are backgrouud images you should use `LV_IMG_CD_TRUE_COLOR` format. It should be much faster this way.

Thanks , let me try.

qq8512852, 2023-08-24:
> If these are backgrouud images you should use `LV_IMG_CD_TRUE_COLOR` format. It should be much faster this way.

Even the size reduce  to  ".data_size = 384000 * LV_COLOR_SIZE / 8," with LV_IMG_CF_TRUE_COLOR.  However, the display effect is still no different from the video.

kisvegabor, 2023-08-24:
It seems like  a large display. How fast your MCU is? And what is the actual resulution and color depth of the display?

qq8512852, 2023-08-24:
> It seems like a large display. How fast your MCU is? And what is the actual resolution and color depth of the display?

Actually, I just want to make a demo in simulator. I though to switch background image was using a computer GPU, and there shouldn't be such a stuck feeling. And I also want to learn and see if there are any ways to optimize it.

It seems that there are no intentions to implement this on a real MCU. Clearly, the current approach is not very practical. For example, a background image alone consumes around 27 MB.

kisvegabor, 2023-08-25:
> Actually, I just want to make a demo in simulator. I though to switch background image was using a computer GPU, and there shouldn't be such a stuck feeling. And I also want to learn and see if there are any ways to optimize it.

In simulator it's shouldn't be that slow. Can you share code snippet to reproduce it?

qq8512852, 2023-08-26:
> In simulator it's shouldn't be that slow. Can you share code snippet to reproduce it?

Sure.   
Execution order:
Create->Enter->Layout
change page:
last screen Exit-> last screen Destroy -> target Create-> target Enter -> target Layout

[image]

[image]

[image]

[image]

kisvegabor, 2023-08-27:
Sorry, but I can't see what should be smooth. I thought there will be an image sequence animation or so, but it seems you just set a few images and that's what I see in the video too.

qq8512852, 2023-08-28:
> Sorry, but I can't see what should be smooth. I thought there will be an image sequence animation or so, but it seems you just set a few images and that's what I see in the video too.

Thanks for your support.
I understand that screen transitions are implemented using the lvgl library, and when changing the background image, the lv_btn_create() function is called. As the caller, you may not see any specific optimizations to be made, so you're unsure about the potential areas for optimization. You noticed some lag during page transitions, which shouldn't ideally occur on a powerful PC with the simulator. 
I speculate whether it is necessary to consider optimizing the underlying LVGL interface for displaying animations with large images？

C47D, 2023-08-28:
That might be the reason, the third parameter is the time of the animation, see [lv_scr_load_anim docs here](https://docs.lvgl.io/8.3/overview/display.html?highlight=lv_scr_load_anim#_CPPv416lv_scr_load_animP8lv_obj_t18lv_scr_load_anim_t8uint32_t8uint32_tb)

qq8512852, 2023-08-29:
> 

Yeah, much appreciate.

closed 2023-08-29 by qq8512852