lvgl/lvgl #4475: NXP VG-Lite reports problem with vg-lite.h

mbrazdil, 2023-08-21:
### Introduce the problem

I am working on a project that uses the LVGL v8.3.8 graphics library. I am using an NXP iMX RT 1060 processor with PXP as the GPU. This processor also supports acceleration using VG-Lite. I tried to set lv_conf.h to use VG-Lite acceleration, but I get an error with the vg_lite.h header file when compiling.  
When I made PXP work this way, all I had to do was set LV_USE_GPU_NXP to 1 and everything worked. But when I set LV_USE_GPU_NXP_VG_LITE, it gives me compilation errors. So does this mean that the VG-LITE implementation is not fully complete? Also, I wonder if it is better to use PXP or VG-LITE? Is there a significant difference?

kisvegabor, 2023-08-23:
@nicusorcitu could you take a look at this issue?

nicusorcitu, 2023-08-23:
iMX RT1060 has PXP only.

mbrazdil, 2023-08-25:
Thank you, that settles it :)

closed 2023-08-25 by mbrazdil