lvgl/lvgl #4479: how to button color change when i clicked

yusufbender, 2023-08-22:
### Introduce the problem

### Examples and cases

### Suggested solution

i look at the doc but i cant find anything. just code func . no example

yusufbender, 2023-08-22:
What MCU/Processor/Board and compiler are you using?
2.8inch_ESP32-2432S028R

What LVGL version are you using?
8.3.9

C47D, 2023-08-23:
I'm using LVGL master but it is almost the same, you need to make `btn` a global variable.

[Screencast from 22-08-23 18:30:53.webm](https://github.com/lvgl/lvgl/assets/11037705/c18e59a8-61a7-4135-be8d-4e0cac51a1de)

yusufbender, 2023-08-23:
how to use in your code `lv_color_t lv_color_make(uint8_t r, uint8_t g, uint8_t b)` pls share me full code

C47D, 2023-08-24:
This code changes the button color from green to red when the button is pressed.

```c
lv_obj_t * btn;

static void btn_event_cb(lv_event_t* e) {
  lv_event_code_t code = lv_event_get_code(e);

  if (code == LV_EVENT_CLICKED) {
    client.subscribe(mqtt_topic);
    client.setCallback(callback);
    if (buttonState) {
      client.publish(mqtt_topic, "0");
      static lv_style_t style_btn_red2;
      lv_style_init(&style_btn_red2);
      lv_style_set_bg_color(&style_btn_red2, lv_color_make(255, 0, 0));
      lv_style_set_bg_opa(&style_btn_red2, LV_OPA_COVER);
      lv_obj_add_style(btn, &style_btn_red2, 0);
    } else {
      client.publish(mqtt_topic, "1");
    }
  }
}

void lv_example_get_started_1(void)
{
    btn = lv_btn_create(lv_scr_act());  
    static lv_style_t style_btn_red;
    lv_style_init(&style_btn_red);
    lv_style_set_bg_color(&style_btn_red, lv_color_make(0, 255, 0)); // BLUE RED GREEN 
    lv_style_set_bg_opa(&style_btn_red, LV_OPA_COVER);
    lv_obj_add_style(btn, &style_btn_red, 0);
    lv_obj_set_size(btn, 90, 50);                   
    lv_obj_align(btn, LV_ALIGN_TOP_LEFT, 0, 0);  // Üst sol köşede hizalama
    lv_obj_add_event_cb(btn, btn_event_cb, LV_EVENT_ALL, NULL);           
}
``

yusufbender, 2023-08-25:
thank you so much . you saved my life o7

closed 2023-08-25 by yusufbender