lvgl/lvgl #4482: Screen load animation corrupts display buffer

faxe1008, 2023-08-23:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Excessive interaction with the display causing screen load animations (MOVE_LEFT/MOVE_RIGHT) causes a corruption of the entire display buffer (It just turns solid white). I reproduced the issue on the lv_port_pc_eclipse project.

I tested this on v8.3.9 and on the current master HEAD. When changing to LV_SCR_LOAD_ANIM_NONE the issue does not appear.

### To Reproduce

```c
lv_obj_t* screen_1;
lv_obj_t* screen_2;

static void btn1_handler(lv_event_t * e)
{
    lv_scr_load_anim(screen_2, LV_SCR_LOAD_ANIM_MOVE_LEFT, 500, 0, false);
}

static void btn2_handler(lv_event_t * e)
{
    lv_scr_load_anim(screen_1, LV_SCR_LOAD_ANIM_MOVE_RIGHT, 500, 0, false);
}

// in main

screen_1 = lv_obj_create(NULL);
lv_obj_t *btn_1 = lv_btn_create(screen_1);
lv_obj_set_size(btn_1, 250, 250);
lv_obj_align(btn_1, LV_ALIGN_CENTER, 0, 0);
lv_obj_add_event_cb(btn_1, btn1_handler, LV_EVENT_CLICKED, NULL);

screen_2 = lv_obj_create(NULL);
lv_obj_t *btn_2 = lv_btn_create(screen_2);
lv_obj_set_size(btn_2, 250, 250);
lv_obj_align(btn_2, LV_ALIGN_CENTER, 0, 0);
lv_obj_add_event_cb(btn_2, btn2_handler, LV_EVENT_CLICKED, NULL);

lv_scr_load(screen_1);

lv_monkey_config_t config;
lv_monkey_config_init(&config);
config.type = LV_INDEV_TYPE_POINTER;
lv_monkey_t * monkey1 = lv_monkey_create(&config);
lv_monkey_t * monkey2 = lv_monkey_create(&config);

lv_monkey_set_enable(monkey1, true);
lv_monkey_set_enable(monkey2, true);
```

### Expected behavior
The display buffer is not corrupted regardless of the chosen animation.

### Screenshots or video

https://github.com/lvgl/lvgl/assets/5813055/6a44c058-ba2d-4fa0-aec3-ba5e4297ba84

faxe1008, 2023-08-23:
I think I found (part) of the issue. My suspicion was that somehow the "wrong" animation was triggered.
I logged the pointer of `screen_1` and `screen_2` and added logging in the `lv_scr_load_anim` function:

```
diff --git a/src/disp/lv_disp.c b/src/disp/lv_disp.c
index 10f9c3df7..f1560ca6e 100644
--- a/src/disp/lv_disp.c
+++ b/src/disp/lv_disp.c
@@ -507,9 +507,16 @@ void lv_scr_load_anim(lv_obj_t * new_scr, lv_scr_load_anim_t anim_type, uint32_t
     lv_disp_t * d = lv_obj_get_disp(new_scr);
     lv_obj_t * act_scr = lv_scr_act();

+    if (anim_type == LV_SCR_LOAD_ANIM_MOVE_LEFT){
+        LV_LOG_WARN("act_scr: %p, new_scr: %p, anim: LEFT", act_scr, d);
+    } else (anim_type == LV_SCR_LOAD_ANIM_MOVE_RIGHT)
+        LV_LOG_WARN("act_scr: %p, new_scr: %p, anim: RIGHT", act_scr, d);
+    }
```
Running the example above yields:
```
[Warn]	(0.261, +261)	 main: screen_1: 0x560b2cc17140, screen_2: 0x560b2cfe4e90 	(in main.c line #101)
[Warn]	(0.261, +0)	 lv_scr_load_anim: act_scr: 0x560b2ba75850, new_scr: 0x560b2ba75cd0, anim: RIGHT 	(in lv_disp.c line #513)
[Error]	(0.261, +0)	 lv_scr_load_anim: lv_scr_load_anim: Shortcut immediate load 	    (in lv_disp.c line #556)
[Warn]	(1.071, +810)	 lv_scr_load_anim: act_scr: 0x560b2cc17140, new_scr: 0x560b2ba75cd0, anim: LEFT 	(in lv_disp.c line #511)
[Warn]	(4.575, +3504)	 lv_scr_load_anim: act_scr: 0x560b2cfe4e90, new_scr: 0x560b2ba75cd0, anim: RIGHT 	(in lv_disp.c line #513)
[Warn]	(8.643, +4068)	 lv_scr_load_anim: act_scr: 0x560b2cc17140, new_scr: 0x560b2ba75cd0, anim: LEFT 	(in lv_disp.c line #511)
[Warn]	(12.463, +3820)	 lv_scr_load_anim: act_scr: 0x560b2cfe4e90, new_scr: 0x560b2ba75cd0, anim: RIGHT 	(in lv_disp.c line #513)
[Warn]	(13.910, +1447)	 lv_scr_load_anim: act_scr: 0x560b2cc17140, new_scr: 0x560b2ba75cd0, anim: LEFT 	(in lv_disp.c line #511)
[Warn]	(16.162, +2252)	 lv_scr_load_anim: act_scr: 0x560b2cfe4e90, new_scr: 0x560b2ba75cd0, anim: RIGHT 	(in lv_disp.c line #513)
[Warn]	(17.868, +1706)	 lv_scr_load_anim: act_scr: 0x560b2cc17140, new_scr: 0x560b2ba75cd0, anim: LEFT 	(in lv_disp.c line #511)
[Warn]	(20.744, +2876)	 lv_scr_load_anim: act_scr: 0x560b2cfe4e90, new_scr: 0x560b2ba75cd0, anim: RIGHT 	(in lv_disp.c line #513)
[Warn]	(22.543, +1799)	 lv_scr_load_anim: act_scr: 0x560b2cc17140, new_scr: 0x560b2ba75cd0, anim: LEFT 	(in lv_disp.c line #511)
[Warn]	(27.277, +4734)	 lv_scr_load_anim: act_scr: 0x560b2cfe4e90, new_scr: 0x560b2ba75cd0, anim: RIGHT 	(in lv_disp.c line #513)
[Warn]	(28.463, +1186)	 lv_scr_load_anim: act_scr: 0x560b2cc17140, new_scr: 0x560b2ba75cd0, anim: RIGHT 	(in lv_disp.c line #513)
```

On the last call where the issue occured you can see that the current and new screen are different from what they should be.
Maybe this has to do with the fact that the button was hit once and during the animation was hit again?

kisvegabor, 2023-08-25:
The PR is merged, closing this issue.

closed 2023-08-25 by kisvegabor