lvgl/lvgl #4484: lv_obj can only scroll in one direction and cannot scroll in both directions simultaneously.

incity, 2023-08-24:
`
lv_obj_set_scroll_dir(obj, LV_DIR_ALL);
`
I has enabled LV_DIR_ALL. But when scrolling horizontally, it is not possible to scroll vertically, or when scrolling vertically, it is not possible to scroll horizontally

kisvegabor, 2023-08-24:
Yes, it's the intended behavior to keep things simpler. 

What is the case when you need 2 axis scrolling?

incity, 2023-08-24:
for example, an image viewer. I need to scale it and scroll it. I also have other use cases. in the LVGL 7.0 the lv_page can scroll horizontally and vertically simultaneously

kisvegabor, 2023-08-25:
In v8 I added the one direction scroll because it hugely simplified a lot of things. So I wonder if a simple workaround like [this](https://github.com/lvgl/lvgl/blob/master/examples/widgets/obj/lv_example_obj_2.c) can be applied in your case.

incity, 2023-08-28:
No, this method change the position of an object. But what I want is change the scroll(x and y) of an object but keep the position of his child objects.

kisvegabor, 2023-08-28:
You can use `lv_obj_scroll_by_bounded` like this:
```c
    lv_point_t vect;
    lv_indev_get_vect(indev, &vect);
    lv_obj_scroll_by_bounded(obj, vect.x, vect.y, LV_ANIM_ON);
```

incity, 2023-08-29:
Will this conflict with scrolling? and how about a scroll throw?

kisvegabor, 2023-08-29:
> Will this conflict with scrolling?

You should disable scrolling in this case with `lv_obj_clear_flag(obj, LV_OBJ_FLAG_SCROLLABLE);`.

>  and how about a scroll throw?

Well, it's really not working out of the box here. 

---

2 axis scrolling is not on the roadmap now, and that's what I can recommend as a workaround. :slightly_frowning_face:

closed 2023-09-20 by lvgl-bot