lvgl/lvgl #4489: Cannot set the object's size in percentage of the parent that has the LV_SIZE_CONTENT setting.

lzj85, 2023-08-25:
[obj1 is the parent window of obj2 and obj3. The height of obj1 is set to LV_SIZE_CONTENT, obj2's height is set to 100px, and obj3's height is set to 100%. However, in actual testing, the height of obj3 is not 100%.

This is my test code:

```
lv_obj_t *obj1 = lv_obj_create(lv_scr_act());
lv_obj_set_width(obj1, LV_SIZE_CONTENT);
lv_obj_set_height(obj1, LV_SIZE_CONTENT);	// obj1 height: content
lv_obj_set_layout(obj1, LV_LAYOUT_FLEX);
lv_obj_set_flex_flow(obj1, LV_FLEX_FLOW_ROW);
lv_obj_set_flex_align(obj1, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);
lv_obj_center(obj1);

lv_obj_t *obj2 = lv_obj_create(obj1);
lv_obj_set_height(obj2, 100);		// obj2 height: 100 pixels

lv_obj_t *obj3 = lv_obj_create(obj1);
lv_obj_set_height(obj3, LV_PCT(100));	// obj3 height 100%
```

**Logically, the height of obj2 determines the height of obj1, so the height of obj3 should be equal to that of obj2. However, in reality, the actual test result is not like that.**
[image]

C47D, 2023-08-25:
Hi, while trying to replicate the issue I'm getting the width and height of all three objects and I get 0's for both?
```c
LV_LOG_USER("Obj1 size: %d, %d", lv_obj_get_height(obj1), lv_obj_get_width(obj1));
LV_LOG_USER("Obj2 size: %d, %d", lv_obj_get_height(obj2), lv_obj_get_width(obj2));
LV_LOG_USER("Obj3 size: %d, %d", lv_obj_get_height(obj3), lv_obj_get_width(obj3));
```

```console
[User]	(0.065, +65)	 main: Obj1 size: 0, 0 	(in main.c line #139)
[User]	(0.065, +65)	 main: Obj2 size: 0, 0 	(in main.c line #140)
[User]	(0.065, +65)	 main: Obj3 size: 0, 0 	(in main.c line #141)
```

Am I doing something wrong or you get the same?

lzj85, 2023-08-25:
> Hi, while trying to replicate the issue I'm getting the width and height of all three objects and I get 0's for both?
> 
> ```c
> LV_LOG_USER("Obj1 size: %d, %d", lv_obj_get_height(obj1), lv_obj_get_width(obj1));
> LV_LOG_USER("Obj2 size: %d, %d", lv_obj_get_height(obj2), lv_obj_get_width(obj2));
> LV_LOG_USER("Obj3 size: %d, %d", lv_obj_get_height(obj3), lv_obj_get_width(obj3));
> ```
> 
> ```
> [User]	(0.065, +65)	 main: Obj1 size: 0, 0 	(in main.c line #139)
> [User]	(0.065, +65)	 main: Obj2 size: 0, 0 	(in main.c line #140)
> [User]	(0.065, +65)	 main: Obj3 size: 0, 0 	(in main.c line #141)
> ```
> 
> Am I doing something wrong or you get the same?

Hi, Thank you for your test.
Usually, after we set the size of an object, we cannot immediately obtain its correct size. Because when we set the object size, the size value is stored in the object's style property. But, when we obtain the object size, it is calculated from the object's coordinates. However, when we set the object size, the lvgl core does not immediately update the object's coordinates. Instead, it updates the layout of the relevant objects in the next call to lv_task_handler(). And the object's coordinates will be updated. After that, we can use lv_obj_get_width, lv_obj_get_height to obtain the object's actual size. This is the reason why you got 0. Of course, we can also forcibly update the layout of the child objects within the specified object by calling lv_obj_update_layout(), such as lv_obj_update_layout(obj1). After that, we can also obtain the correct object size. However, this is the normal functioning mechanism of lvgl, so I think this is not the cause of my question.

C47D, 2023-08-25:
> Usually, after we set the size of an object, we cannot immediately obtain its correct size. Because when we set the object size, the size value is stored in the object's style property. But, when we [...]

Thanks for the detailed explanation, I was trying to visualize the object sizes to know more or less where the problem is, I'll take a look again over the weekend.

lzj85, 2023-08-26:
> > Usually, after we set the size of an object, we cannot immediately obtain its correct size. Because when we set the object size, the size value is stored in the object's style property. But, when we obtain the object size, it is calculated from the object's coordinates. However, when we set the object size, the lvgl core does not immediately update the object's coordinates. Instead, it updates the layout of the relevant objects in the next call to lv_task_handler(). And the object's coordinates will be updated. After that, we can use lv_obj_get_width, lv_obj_get_height to obtain the object's actual size. This is the reason why you got 0. Of course, we can also forcibly update the layout of the child objects within the specified object by calling lv_obj_update_layout(), such as lv_obj_update_layout(obj1). After that, we can also obtain the correct object size. However, this is the normal functioning mechanism of lvgl, so I think this is not the cause of my question.
> 
> Thanks for the detailed explanation, I was trying to visualize the object sizes to know more or less where the problem is, I'll take a look again over the weekend.
> 
> I forgot to ask you which version of LVGL are you using, I'm currently using the top of the master branch.

I'm currently using LVGL v8.3.3. However, I have tested the top of the master branch and the same issue still exists.

C47D, 2023-08-26:
@kisvegabor should obj3 has the same height of obj2? Obj2 is the only one setting the content size, I guess this is what you expected to see @lzj85

kisvegabor, 2023-08-27:
Yes, what happens is  exactly what @C47D has been described. For example:
1. Let's say the parent initially has 500 px height and it has a child with 50% height. So the child height is 250 px
2. Now you set the parent's height to `CONTENT`
3. Parent height will be 250 px (assuming zero padding)
4. Now child should have 125 px height
5. Now parent should have 125 px height too
6. But then the  child should have 62 px height
7. and so on until child height is 0 px

I admit that there are some special cases when a CONTENT parent and a % child works and these are not considered by the algorithm now:
- 100% child size won't cause this loop but only 1-2 extra iterations
- if the parent has a fixed sized child (like in your example), not 100 % children will stop at one point, but it's less obvious if the there x or y offset on the child, especially if the x or y offset is also in percent.

lzj85, 2023-08-29:
> @kisvegabor should `obj3` has the same height of `obj2`? `obj2` is the only one setting the content size, I guess this is what you expected to see @lzj85
> 
> See this note in `lv_obj_refr_size`.
> 
> > If parent has content size and the child has pct size a circular dependency will occur. To solve it keep child size at zero
> 
> https://github.com/lvgl/lvgl/blob/b6c5ab395435e945f51e242481eb40c1d57270b4/src/core/lv_obj_pos.c#L124-L154
> 
> The result of the `obj3` height calculation is `44`, which matches with the `obj3` coordinates `(y2 - y1) + 1` [image]

Thanks! That is indeed the cause of the issue I mentioned.

lzj85, 2023-08-29:
> Yes, what happens is exactly what @C47D has been described. For example:
>
>     1. Let's say the parent initially has 500 px height and it has a child with 50% height. So the child height is 250 [...]

Thanks. In some special cases, it seems that handling the events such as LV_EVENT_SIZE_CHANGED and LV_EVENT_LAYOUT_CHANGED is also an option. I guess it could work.

closed 2023-09-20 by lvgl-bot