lvgl/lvgl #4499: rethink about the weights in the LVGL benchmark

GorgonMeducer, 2023-08-29:
### Introduce the problem

Rotation and scaling (zoom) have become more and more important recently, especially in budget watches. In benchmark, I notice that the score related to rotations and scaling contributes less than their actual value. We should re-think and adjust those weights.

For example, increase the `Image ARGB rotate anti aliased` to `20` etc.

https://github.com/lvgl/lvgl/blob/5b60d1de726aa65069410a57c8c7fdf59cecd8d0/demos/benchmark/lv_demo_benchmark.c#L595-L602

### Examples and cases

### Suggested solution

kisvegabor, 2023-08-30:
Actually I'm thinking about creating a new much simpler benchmark. When I was working with different companies with different target platforms using the current benchmark it was hard to tell what performance can we achieve.

This is what I have in mind for the steps in benchmark:
1. Empty screen, change background color
2. Screen fully covered by tiled images
3. Screen with a lot of ARGB images
4. Some ARGB image rotation 
5. Several containers with small radius and texts moving
9. Same as 4) but with a semi transparent overlay changing color
6. Same as 4) but the widget has animated `opa`
7. Same as 4) but the widget has animated `opa_layer`
8. Same as 4) but the widget are rotated
10. A balanced UI with some widgets on tabview and changing tabs

This is 10 only the parameters but for a user it tells almost everything. I think the current benchmark should be used for development purposes to compare GPUs, and related configurations. 

What do you think?

GorgonMeducer, 2023-08-31:
That's what I called "scene (scenario) based benchmark ". The key is 
- create a set of scenes for typical scenarios
- measure the performance of each scenes and generate reports for each scenes

Our current benchmark can be rename and reused as a tool for function correctness checking. That's how I use it when porting GPU drivers. 

So, We can raise an issue and collect the ideas about the useful scenarios.

GorgonMeducer, 2023-08-31:
For short term, can we update the weights of the existing benchmark topics? i.e rotation etc?

kisvegabor, 2023-08-31:
> Our current benchmark can be rename and reused as a tool for function correctness checking. That's how I use it when porting GPU drivers.

I do the same, and I believe @nicusorcitu too.

> For short term, can we update the weights of the existing benchmark topics? i.e rotation etc?

Yes, can you send a Pull request with the suggested changes?

GorgonMeducer, 2023-08-31:
> Yes, can you send a Pull request with the suggested changes?

I will do this in the next v8.3.10 release. Is this OK?

GorgonMeducer, 2023-09-01:
OK, I will send a PR for master.

GorgonMeducer, 2023-09-15:
It is not stale.

GorgonMeducer, 2023-09-30:
not stale. I will generate a PR next week.

closed 2023-10-03 by GorgonMeducer