lvgl/lvgl #4510: [Demo widgets] Selecting a point in the line chart shows info of point before the selected one

C47D, 2023-09-01:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I'm seeing two issues with the demo widgets (I'm using the latest lv_port_pc_eclipse simulator)

1. When selecting a point in the line chart the information displayed is about the previous point.
2. Is not possible to select the first point, when selecting it the part received is MAIN instead of the expected ITEMS.

### To Reproduce
Clone `lv_port_pc_eclipse` and go to commit `86df5c7`. Compile and run the demo widgets, and replay the usage from the video below.

### Expected behavior

1. Information displayed is from the selected point
2. Information for first point is also available

### Screenshots or video
[Screencast from 31-08-23 23:05:59.webm](https://github.com/lvgl/lvgl/assets/11037705/fa33f4ac-6ded-4337-8390-be790c00eb74)

kisvegabor, 2023-09-01:
It should be fixed already. Please try with the latest master.

C47D, 2023-09-03:
It is working now!

closed 2023-09-03 by C47D