lvgl/lvgl #4516: Cannot change outline style in v8.3

wuzhongjie1997, 2023-09-04:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Updating from 8.1 to 8.3 broke the ability to change a widget's outline style

### To Reproduce

Paste the following block into [v8.1 simulator](https://sim.lvgl.io/v8.1/micropython/ports/javascript/index.html), and hit 'Restart':
```python
# Initialize 
import display_driver
import lvgl as lv

# Create a button with a label 
scr = lv.obj()
lv.scr_load(scr)

btn = lv.btn(scr)
btn.align(lv.ALIGN.CENTER, 0, 0)
btn.set_style_outline_width(30, lv.STATE.DEFAULT)
btn.set_style_outline_color(lv.color_hex(0xFF0000), lv.STATE.DEFAULT)

# Set focused outline color
btn.set_style_outline_color(lv.color_black(), lv.STATE.FOCUSED)

label = lv.label(btn)
label.set_text('Hello World!')
```
Notice the thick red outline.

Now do the same in [v8.3](https://sim.lvgl.io/v8.3/micropython/ports/javascript/index.html), the outline style remains unchanged (thin, blue outline as though the set_style_outline_* calls weren't there). 

Test code modified from [this forum post](https://forum.lvgl.io/t/how-to-change-the-button-outline-color-when-it-is-focused/8344/2)

### Expected behavior

Updating the version in simulator should not produce different results. 

### Screenshots or video

[image]
[image]

wuzhongjie1997, 2023-09-04:
Might be relevant: this issue was observed when porting to STM32F411CEU6, with a 240x320 LCD driven by ST7789V, followed porting guide in docs. Will try with v8.1 in same environment, BRB

wuzhongjie1997, 2023-09-04:
Confirmed with STM32 that the set_style_outline_* calls work as expected in v8.1. Any response much appreciated.

kisvegabor, 2023-09-04:
It's because the blue ting is in the `FOCUS_KEY` state: `
btn.set_style_outline_color(lv.color_black(), lv.STATE.FOCUS_KEY)`

wuzhongjie1997, 2023-09-05:
Thanks for the clarification, does this mean the behavior in v8.1 is wrong, and v8.3 actually works as expected?

wuzhongjie1997, 2023-09-05:
```python
# Initialize 
import display_driver
import lvgl as lv

# Create a button with a label 
scr = lv.obj()
lv.scr_load(scr)

btn = lv.btn(scr)
btn.align(lv.ALIGN.CENTER, 0, 0)
btn.set_style_outline_width(30, lv.STATE.DEFAULT)
btn.set_style_outline_color(lv.color_hex(0xFF0000), lv.STATE.DEFAULT)

# Set focused outline color
btn.set_style_outline_color(lv.color_black(), lv.STATE.FOCUS_KEY)
btn.clear_state(lv.STATE.FOCUS_KEY) # adding this was the key
label = lv.label(btn)
label.set_text('Hello World!')
```
Apparently I was not changing the style which the button used. The original issue was probably due to different 'default state' policy of the LVGL simulator v8.1 vs v8.3. Closing as completed.

closed 2023-09-05 by wuzhongjie1997