lvgl/lvgl #4529: Release v8.3.10

kisvegabor, 2023-09-07:
@GorgonMeducer 
Could you prepare for releasing v8.3.10?

GorgonMeducer, 2023-09-18:
Sorry, just saw this today. OK, I will do.

GorgonMeducer, 2023-09-19:
Done #4572

kisvegabor, 2023-09-20:
[v8.3.10 is released](https://github.com/lvgl/lvgl/releases/tag/v8.3.10), thank you!

kisvegabor, 2023-09-20:
@tore-espressif 
Uploading LVGL v8.3.10 to Espressif's component registry failed. See [here](https://github.com/lvgl/lvgl/actions/runs/6246940142/job/16958555398). 
Is it an issue on our side?

closed 2023-09-20 by kisvegabor

tore-espressif, 2023-09-25:
> Is it an issue on our side?

No, it does not seem so. Could you please re-run the job?

kisvegabor, 2023-09-25:
Re running helped, thank you! :slightly_smiling_face: 
https://github.com/lvgl/lvgl/actions/runs/6246940142/job/17106585460
[image]