lvgl/lvgl #4531: Discussion on the draw label process when drawing tick in chart

W-Mai, 2023-09-07:
### Introduce the problem
1. In chart, when drawing tick, a piece of buf will be prepared first, then the formatted buf will be filled in, and then the actual text width of the buf will be measured, and the text will be drawn according to the text width.
2. The difference is that in v8, there is `LV_EVENT_DRAW_PART_BEGIN`, so if I want to hook the process of drawing tick, I just need to refill the content of the buf in my callback, because the actual width of the text will be calculated later in the chart.
3. But v9 is not like this. In v9, after going through step 1, it calls `lv_draw_label`, in which not only allocate new memory and copy it (for multithreading), and then trigger the `LV_EVENT_DRAW_TASK_ADDED` event, refills a new buf in my hook's callback, and free the old buf. In this process, it will allocate new memory and copy it (`lv_demo_widgets.c:1344`), not only that, though the measurement of the text has already occurred, I still need to calculate the actual width of the text again.
4. In short, compared with v8, v9 adds two memory allocation and copies and one measurement of text width. I think this process has caused some waste. So, how should we optimize this process?

Welcome to discuss it, thank you😮

### Examples and cases

N/A
### Suggested solution

N/A

kisvegabor, 2023-09-08:
Hey,

Thank for bringing this up. 

In v9 I'm planning to replace all scale-like  things (mainly chart and meter) with `lv_scale`. I still don't know all the details, but my idea is the following:
1. Create a chart and set the ranges
2. Create a vertical `lv_scale`. We will use it for the left side
3. Set the texts and other parameters of the scale
4. Simply align the scale next to the chart
5. Do the same with a horizontal scale and align it to bottom of the chart

This approach has several advantages:
- First, we need only one good scale implementation, and we can reuse it for any widgets
- It makes the chart less bulky.  
- It's more flexible as you can place and customize the scale anyhow you want. (position, range, scroll/not scroll with the chart, etc). 
- Now we have the `draw_size` hack in `lv_chart_set_axis_tick` to tell how many pixels you need for the ticks and texts. If the scale would be a real widget it wouldn't be required.
- If the scale is a real widget you can use layouts and `LV_SIZE_CONTENT` on chart+tick combinations.

There is only one thing which is not clear to me: how to make the scale scroll with the chart? I'm thinking about removing `lv_chart_set_zoom_x/y` and instead the user can simply set the width/height of the chart to larger value. If the chart is larger than the parent the parent will be scrollable (not the chart).  This way it would be easy to implement both the scrolling scale and non-scrolling scale:
- non scrolling scale: create the scale on a parent of the chart, set the scale's width to 100% (of parent size) and set the `LV_OBJ_FLAG_FLOATING` to prevent the scale to be scrolled
- scrolling scale: create the scale and the chart on a wrapper container, set 100 % width for both the chart and the scale, and adjust the size of the wrapper to make it scroll on its parent.

These all were not directly related to your question, but I needed to tell it to someone.  Thank you if you were reading it :joy: 

To really answer your question: `lv_scale` has a `lv_scale_set_text_src()` function, so you don't need to use a hook to set the texts. With this there is no allocation at all,  and the text is measured/positioned only once in `lv_scale`.

What do you think?

W-Mai, 2023-09-11:
Thank you for your long reply.

The above reply does solve the text display problem of tick, and there is no need for the process of hook drawing.

But if I want to draw a line chart and so on, how should I align what I draw with tick? Do we need a method like `lv_pos_t lv_scale_get_pos(scale, marjor, minor);` and so on? If it is a rotating arc scale, there will be a method similar to `lv_pos_t lv_scale_get_pos_arc(scale, marjor, minor, radius);` to get the appropriate coordinates through the radius?

What we need to solve now is the custom chart content and custom tick, and the chart content and tick should be perfectly aligned.

What do you think?

kisvegabor, 2023-09-11:
Just to summarize what we concluded in our meeting:
- If you would like to set the tick evenly for each points of a chart, just set the same number of major ticks for the scale you do for the chart
- If need an uneven scale (e.g. logarithmic) we can add a "mapper" function which maps the value into the size of scale

W-Mai, 2023-09-21:
so how can we push forward this issue

kisvegabor, 2023-09-22:
I'll work on it next week.

kisvegabor, 2023-10-08:
Sorry for the delay, I was finalizing [this feature](https://github.com/lvgl/lvgl/pull/4541/) and couldn't finish updating LVGL charts and meters to lv_scale on all places. I still need a few days :slightly_frowning_face:

closed 2023-10-30 by lvgl-bot