lvgl/lvgl #4532: Data-widget binding, and life cycle management

kisvegabor, 2023-09-07:
### Introduce the problem

Now there is a few possible ways to handle the data and the widgets together, and deal with creating and deleting them dynamically, but none of them is really clean and officially recommended. It would be great to have a recommended LVGL-ish way of handling all these.

### Examples and cases

1. How/Where to store the data related to widgets?
1. How/When to create and delete these data?
1. How to connect the value of the widgets to the values stored in the data? 
1. When a screen or widget is created how to set their values to the value stored in a "database"/ "controller" automatically or at least simply? 

### Suggested solution

Now we have [`lv_fragment`](https://docs.lvgl.io/8.3/others/fragment.html) but to be honest I'm not big fan of it as it 's quite complicated, requires a lot of boiler plate, and forces you to do things in a specific way.

What I have in mind is a variable binding module. It would be based on [`lv_msg`](https://docs.lvgl.io/8.3/others/msg.html) and we would have API functions like these:
```c
lv_label_bind_to_text(label, some_string);
lv_dropdown_bind_selected(dropdown, &some_int);
lv_slider_bind_value(slider, &some_int);
lv_obj_bind_state(obj, LV_STATE_CHECKED, &some_bool);
lv_obj_bind_flag(obj, LV_OBJ_FLAG_HIDDEN, &some_bool);
``` 

When the widget's value changes it sets the bound variable too. When the bound variable is changed by the application an `lv_var_update(&var)` function needs to be called to notify the interested widgets. `lv_var_update` is called in the widgets too e.g. when the slider's value is changed by the user,

To get notification in the app about the change of a variable the  app needs to subscribe to that related variable:
```c
lv_msg_subscribe(&var, some_callback);
```

It would solve how to connect the data to the widgets and keep the data and widgets in sync. We  can store the values in global variables, and "communicate" through them, so the "where to store the data" is also clean. Or if you know that there is smaller scope, of course you can store it there too.

An additional layer of magic could be something like that:
```c
//-------------------------------
//Example password authentication
//-------------------------------

//Store the current password in a string
char password_string[64];
lv_textarea_bind_text(password_textarea, password_string);

//If the password is ok, set a bool variable
bool password_ok;
lv_bind_string_eq_to_bool(password_string, "1234", &password_ok);   //If password_string is  "1234" set password_ok=true, else false

//If the password is ok enable a next button
lv_obj_bind_state(next_button, LV_STATE_DISABLED, &password_ok);

//Set the status message based according to password_ok
char state_message[64];
lv_string_bind_bool(state_message, &password_ok, "Password ok", "Wrong password"); //state_message = password_ok ? "Password ok" : "Wrong password"

//Show the status on a label 
lv_label_bind_text(status_label, state_message);
```

Of course the authentication can be more complicated but I think your get the idea. To building up connection even between  distinct parts of the UI these data bindings can be easily used with global variable variables,  which also act as a single source of truth. 
  
cc @XuNeo @bjsylvia

bjsylvia, 2023-09-09:
Data binding typically consists of a data source and some observers. Is it more appropriate to add related definitions to the data source in order to unify and constrain its add, delete, and modify operations?

For example:

```
/* we need a data source structure, which contains type, value and observers information: lv_ds_t */

lv_ds_t* lv_ds_create();

lv_ds_add_observer(lv_ds_t*, lv_ds_update_cb_t update_cb, void* data);
lv_ds_remove_observer(lv_ds_t*,  lv_ds_update_cb_t update_cb, void* data);

// when operation is 'set', it will notify all observers
void lv_ds_set_int_data(lv_ds_t*, int value);
int lv_ds_get_int_data(lv_ds_t*);
...

void lv_ds_destroy(lv_ds_t*);
```
Other higher-level bind APIs can be built based on lv_ds_xxx.

kisvegabor, 2023-09-11:
We started to work on it: https://github.com/lvgl/lvgl/pull/4541

kisvegabor, 2023-09-26:
Let's continue in the PR.

closed 2023-09-26 by kisvegabor