lvgl/lvgl #4534: Hello, I have a question for you.

Smallyellowcroaker, 2023-09-08:
Hello! What is the reason that the function "lv_obj_set_style_transform_angle" can change the angle of the image, but not the angle of the label. My version is 8.1.0

kisvegabor, 2023-09-08:
Hi,

Transforming any widgets (not only images) was added in v8.3.

Smallyellowcroaker, 2023-09-11:
> 你好
> 
> 在 v8.3 中添加了转换任何小部件（不仅是图像）。

Hi, is it not possible with v8.1?

kisvegabor, 2023-09-11:
It's not possible in a reasonable. It was one of the big updates of v8.3 and we needed to to do lot of changes in LVGL to support it. 

However v8.3 should be a drop in replacement of v8.1 so you should be able to update it without any issues.

Smallyellowcroaker, 2023-09-12:
> 这在合理的情况下是不可能的。这是 v8.3 的重大更新之一，我们需要在 LVGL 中进行大量更改以支持它。
> 
> 但是，v8.3 应该是 v8.1 的替代品，因此您应该能够毫无问题地更新它。

Thank you.

closed 2023-10-03 by lvgl-bot