lvgl/lvgl #4540: Memory violation in roller widget when event comes from overlay

stwirth, 2023-09-08:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

When a roller widget has another object on top and that object has bubble events enabled, the internal call to `get_label` accesses an invalid pointer.

### To Reproduce

```diff
diff --git a/tests/src/test_cases/test_roller.c b/tests/src/test_cases/test_roller.c
index a84afc08e..3fa6130eb 100644
--- a/tests/src/test_cases/test_roller.c
+++ b/tests/src/test_cases/test_roller.c
@@ -161,6 +161,14 @@ void test_roller_keypad_events(void)
     TEST_ASSERT_EQUAL(expected_index, actual_index);
 }
 
+void test_roller_with_overlay_and_bubble_events_enabled(void)
+{
+    lv_obj_t * overlay = lv_obj_create(roller);
+    lv_obj_add_flag(overlay, LV_OBJ_FLAG_EVENT_BUBBLE);
+
+    lv_obj_send_event(overlay, LV_EVENT_PRESSED, NULL);
+}
+
 //void test_roller_infinite_mode_first_option_gets_selected_after_last_option(void)
 //{
 //    char actual_str[OPTION_BUFFER_SZ] = {0x00};

```

### Expected behavior

No crash.

### Screenshots or video

kisvegabor, 2023-09-08:
I think the proper solution would be to explicitly save the label in `lv_roller_t`. If you agree, could you send a pull request?

stwirth, 2023-09-11:
I'm not sure if the access to the label was the actual problem. After some experiments I found that just using `lv_event_get_current_target` instead of `lv_event_get_target` in `lv_roller_event` makes the test pass:
```diff
diff --git a/src/widgets/roller/lv_roller.c b/src/widgets/roller/lv_roller.c
index f9aa7d519..ce818f167 100644
--- a/src/widgets/roller/lv_roller.c
+++ b/src/widgets/roller/lv_roller.c
@@ -334,7 +334,7 @@ static void lv_roller_event(const lv_obj_class_t * class_p, lv_event_t * e)
     if(res != LV_RES_OK) return;
 
     const lv_event_code_t code = lv_event_get_code(e);
-    lv_obj_t * obj = lv_event_get_target(e);
+    lv_obj_t * obj = lv_event_get_current_target(e);
     lv_roller_t * roller = (lv_roller_t *)obj;
 
     if(code == LV_EVENT_GET_SELF_SIZE) {
```

Is that a valid fix? I'm seeing `lv_event_get_target` being used in many event handlers in widgets.
When bubble events is enabled for the object that overlays the roller, then the event handler is called with that object in `lv_event_get_target` and the cast to `lv_roller_t *` is invalid.

kisvegabor, 2023-09-11:
Ah,that's absolutely a proper fix. There were some changes back and forth about how to handle event bubbling and what kind of API to have. Seemingly the changes weren't correctly followed up for all widget. 

Please send a pull request with the suggested changes for the roller. If you see other places where the seems fix should be applied, please include it into the PR as well.

stwirth, 2023-09-19:
@kisvegabor I added the fix to this PR: https://github.com/lvgl/lvgl/pull/4542

kisvegabor, 2023-09-20:
The PR is merged, losing this issue.

Thank you!

closed 2023-09-20 by kisvegabor