lvgl/lvgl #4543: bug in lvgl v8.3.9 - does no longer works properly with lv_port_pc_visual_studio release/v8.3

ataweg, 2023-09-08:
after updating lvgl to the latest version in lv_port_pc_visual_studio the keyboard application no longer works properly. When changing case, the keyboard layout only changes after clicking a character key. This is different to the original version in lv_port_pc_visual_studio.
It seems to me that with the introduction of lv_obj_get_style_opa_recursive() events are lost. If LV_USE_PERF_MONITOR  or LV_USE_MEM_MONITOR is set to 1, it works. 
The behavior is not limited to lv_port_pc_visual_studio. I have this behavoir in my project too.
I think this is not a problem of v_port_pc_visual_studio.

How to reproduce:
git clone --recurse-submodules https://github.com/lvgl/lv_port_pc_visual_studio.git -b release/v8.3

update lvgl in the LVGL.Simulator folder to the latest version  (commit 9787d47d0b6a000e662a65c8a9dc456e47e9e331)
modify LVGL.Simulator.cpp
    disable build of lv_demo_widgets();    (line 316)
    enable build of lv_example_keyboard_1(); (line 419)
modify lv_conf.h
   #define LV_USE_PERF_MONITOR     0
   #define LV_USE_MEM_MONITOR      0

kisvegabor, 2023-09-08:
Hi,

Thanks for reporting it.

I've found the bug and pushed a fix. See 28094404a.
Does it work on your end too? :crossed_fingers:

ataweg, 2023-09-23:
now it works fine. Thanks.

closed 2023-09-23 by ataweg