lvgl/lvgl #4548: Usage of class_p in constructor

XuNeo, 2023-09-12:
### Introduce the problem

`class_p` is always marked as `UNUSED` in widget constructor callback. Is `class_p` meant to construct an `obj` with type `class_p`?

Originally, `obj->class_p` is changed temprarily before calling constructor, with this patch [feat(obj): construct obj with specified class_p](https://github.com/lvgl/lvgl/pull/4536/commits/9c2cce6f9be361e48ff5f8740f5a99fdda932611) `obj->class_p` is kept untouched, widget constructor should use parameter `class_p` instead, if really needed.

Does this patch make sense? 

```
static void lv_obj_constructor(const lv_obj_class_t * class_p, lv_obj_t * obj)
{
    LV_UNUSED(class_p);
}
```

### Examples and cases

### Suggested solution

kisvegabor, 2023-09-12:
In C it wouldn't be required to pass the the `class_p` to the constructor, as we know that e.g. `lv_btn_constructor` is always related to `lv_btn`. So we can read the required data from `lv_btn_class` directly.  

It was related to the MicroPython binding's conventions. That is, if we store a callback in a `struct`, we need to pass that `struct`  to the callback as well. In this case we store the constructor in the `class_p` so we need to pass `class_p` to the callback. That's because MicroPython stores some data about the callbacks in the  `struct`'s `user_data`.

So in case of button we can say that always `lv_btn_class` should be passed to the button's constructor. 

However, we start to construct from the widgets from the oldest ancestor. E.g.:
1. obj
2. bar
3. slider
 
class_p` has an `event_cb` field. It cannot happen that while we are creating the `obj` part, the `slider` event is being called, as the slider part is not initialized yet. That's why we change `obj->class_p` to the current class. It ensures that only the being constructed class' events can run.

XuNeo, 2023-09-21:
Added obj's final class type in https://github.com/lvgl/lvgl/pull/4582

closed 2023-09-21 by XuNeo