lvgl/lvgl #4549: Lvgl V8.2 lv_fs read *.png show “no data”, but *.bin it work

acer1204, 2023-09-12:
hi
does anyone has this problem?
my core is stm32h7, LV_MEM_SIZE is 80 * 1024U,
my test image is 120*120-32bit, if i use lv_fs to read it.

`
lv_imgbtn_set_src(ui->screen_home_imgbtn_state, LV_IMGBTN_STATE_RELEASED, NULL, "S:status.png", NULL);
`
it always show “no data”, but if i use 40 * 40-32bit, it working fine.
i find some article, but always say is memory too small.

but if i convert it to bin format. and use lv_fs to read the 120*120-32bit bin. it working.

`
lv_imgbtn_set_src(ui->screen_home_imgbtn_state, LV_IMGBTN_STATE_RELEASED, NULL, "S:status.bin", NULL);
`
it just draw slow but work.(can see it draw from up to down).
why the bin is work but png can’t?
does png format can like bin, just slow draw but show?

kisvegabor, 2023-09-13:
Hi,

Could you attache the PNG image to reproduce it? 

A few questions:
- What is `LV_COLOR_DEPTH`?
- Which LVGL version do you use?

> does png format can like bin, just slow draw but show?

Bin images are read line-by-line, while PNG images are fully decoded into RAM. So bin images are slower but uses less memory. To keep the the decoded image cached, you should set `LV_IMG_CACHE_DEF_SIZE` e.g. to `4` (4 is the number of images to be cached)

acer1204, 2023-09-14:
my image depth is 32 bit and lv_config LV_COLOR_DEPTH is 16.
LVGL version is 8.2, and image is this. 120*120*32bit
[image]

kisvegabor, 2023-09-14:
Thanks, I've tried it out and it's working fro me, but there was an out of memory issue initially. I've realized that the PNG decoder temporarily reads the whole image data into RAM and decodes it from there. So during the decoding you will need +10.6kB RAM. 

Does it work if you increase `LV_MEM_SIZE`? Let's double it, if it's possible.

closed 2023-10-06 by lvgl-bot