lvgl/lvgl #4551: lv_scale updates

C47D, 2023-09-12:
### Introduce the problem

Updates based on observations from @kisvegabor made during `lv_scale` integration.
- Using `float` should be avoided. Instead some up scaling or smart ordering of operation should be used.
- The horizontal and vertical scale should consider the horizontal and vertical padding respectively. That is the ticks should start after the left/top padding and stop before the right/bottom padding.

### Examples and cases
N/A

### Suggested solution
Fix both observations

C47D, 2023-09-12:
For the float issue, when using lv_coords there was a rouding issue with the positioning of the ticks, I'll check if it's possible to remove that rounding issue without using the float type.

For the padding issue, do you mean the ticks should start X pixels after the start of the main line and stop Y pixels before the ending of the line?

C47D, 2023-09-13:
For reference, this is the current renders of the examples (using float) commit 140369c5c:

Example 1
[image]

Example 2
[image]

Example 3
[image]

Example 4
[image]

Example 5
[image]

kisvegabor, 2023-09-13:
> For the float issue, when using lv_coords there was a rouding issue with the positioning of the ticks, I'll check if it's possible to remove that rounding issue without using the float type.

I mean the whole scale (not only the ticks) should start after/stop before the padding.

PR #4552 (2023-09-22): lv_scale: Update with feedback
### Description of the feature or fix

Fixes #4551 

### Checkpoints
- [x] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed
- [ ] Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant. 
- [ ] Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- [ ] If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https://github.com/l [... cut]
closed 2023-09-22 by kisvegabor