lvgl/lvgl #4554: Header file installation problem in custom. cmake

597244721, 2023-09-13:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
When I build the lvgl library using cmake, I import lvgl as a subdirectory. After compiling, execute “make install” and find that there is a problem in the include directory. The directory becomes recursive, causing execution failure

### Expected behavior
Imported as a subdirectory, can be compiled and installed normally

### Screenshots or video
If applicable, add screenshots to help explain your problem.

597244721, 2023-09-13:
[image]
I tried to modify it like this and it works as expected

kisvegabor, 2023-09-13:
Thank you for pointing it out. Could you open Pull request with the proposed fix?

zephyr-zhu, 2023-09-14:
ok, I'll try to submit it

PR #4557 (2023-09-14): fix(cmake): fix issue of cmake generating header files
### Description of the feature or fix

#4554  Fixing the issue with lvgl being referenced as a subdirectory and generating header files with cmake install

### Checkpoints
- [ ] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed
- [ ] Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant. 
- [ ] Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/READ [... cut]
closed 2023-09-14 by kisvegabor