lvgl/lvgl #4556: Function that can determine whether a text is rotating or not

Bartaaak, 2023-09-13:
### Introduce the problem

#### Function that are called on an object and can determine whether the text can fit into its parent or whether text will rotate because it does not fit. 

### Examples and cases

#### Recently I tried to create "dynamic" popup that can determine whether a text inside a label is too long and after that create a popup or just let it be. Because of that i had to know the size of text in pixels.

### Suggested solution

#### I think that a simple functions that returns true if the text doesnt fit into its parent would be really handy.

kisvegabor, 2023-09-13:
Hi,

I don't understand perfectly what the popup does if text fits or doesn't fit. Could you describe it again?

Bartaaak, 2023-09-14:
Well if the text fits perfectly, the popup won't be created, but if the text is too big-only a part of the text is shown and pop up with full message is created, that means you are able to click and popup will show you whole message.

kisvegabor, 2023-09-14:
I see. Can you use `lv_txt_get_size()` for this purpose?

Bartaaak, 2023-09-14:
Yes. In the end I used this function.

closed 2023-10-01 by Bartaaak