lvgl/lvgl #4562: lv_bar trigger value changed event

krupis, 2023-09-15:
### Introduce the problem
Hello. I have a button and bar. When I click a button, I want the bar value to go from 0 to 100 indicating update status. When the value reaches 100, I want to generate a signal to notify me and then I want to do something else. I cannot figure a way to generate a signal or event when the value reaches 100.

### Examples and cases
Some pseudo code:

```
   ui_CalibrateStart = lv_btn_create(ui_ConfirmationScreen);
    lv_obj_set_width(ui_CalibrateStart, 100);
    lv_obj_set_height(ui_CalibrateStart, 50);
    lv_obj_set_align(ui_CalibrateStart, LV_ALIGN_CENTER);
    lv_obj_add_flag(ui_CalibrateStart, LV_OBJ_FLAG_SCROLL_ON_FOCUS); /// Flags
    lv_obj_clear_flag(ui_CalibrateStart, LV_OBJ_FLAG_SCROLLABLE);    /// Flags
    lv_obj_set_style_bg_color(ui_CalibrateStart, lv_color_hex(0xFFFFFF), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(ui_CalibrateStart, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_text_font(ui_CalibrateStart, &lv_font_montserrat_14, LV_PART_MAIN | LV_STATE_DEFAULT);

    ui_StartLabel = lv_label_create(ui_ConfirmationScreen);
    lv_obj_set_width(ui_StartLabel, LV_SIZE_CONTENT);  /// 1
    lv_obj_set_height(ui_StartLabel, LV_SIZE_CONTENT); /// 1
    lv_obj_set_align(ui_StartLabel, LV_ALIGN_CENTER);
    lv_label_set_text(ui_StartLabel, "Calibrate");

    ui_UpdateProgress = lv_bar_create(ui_ConfirmationScreen);
    lv_obj_set_width(ui_UpdateProgress, 150);
    lv_obj_set_height(ui_UpdateProgress, 20);
    lv_obj_set_x(ui_UpdateProgress, 0);
    lv_obj_set_y(ui_UpdateProgress, 70);
    lv_obj_set_align(ui_UpdateProgress, LV_ALIGN_CENTER);
    lv_obj_set_style_bg_color(ui_UpdateProgress, lv_color_hex(0xFFFFFF), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(ui_UpdateProgress, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_anim_time(ui_UpdateProgress, 5000, LV_PART_MAIN | LV_STATE_DEFAULT);

    lv_obj_set_style_bg_color(ui_UpdateProgress, lv_color_hex(0x409E3B), LV_PART_INDICATOR | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(ui_UpdateProgress, 255, LV_PART_INDICATOR | LV_STATE_DEFAULT);

    lv_obj_add_event_cb(ui_CalibrateStart, ui_event_CalibrateStart, LV_EVENT_ALL, NULL);
    lv_obj_add_event_cb(ui_UpdateProgress, ui_event_UpdateProgress, LV_EVENT_ALL, NULL);
```

```
void ui_event_CalibrateStart(lv_event_t *e)
{
    lv_event_code_t event_code = lv_event_get_code(e);
    lv_obj_t *target = lv_event_get_target(e);
    if (event_code == LV_EVENT_CLICKED)
    {
        _ui_bar_set_property(ui_UpdateProgress, _UI_BAR_PROPERTY_VALUE_WITH_ANIM, 100);
    }
}

void ui_event_UpdateProgress(lv_event_t *e)
{
    lv_event_code_t event_code = lv_event_get_code(e);
    lv_obj_t *target = lv_event_get_target(e);
    if (event_code == LV_EVENT_VALUE_CHANGED )
    {
        int value = lv_bar_get_value(ui_UpdateProgress);
        printf("value = %u \n", value);
    }
}
```

As you can see from the code above, when I click button, ```ui_event_CalibrateStart``` is triggered that calls the function:
```
_ui_bar_set_property(ui_UpdateProgress, _UI_BAR_PROPERTY_VALUE_WITH_ANIM, 100);
```

I was hoping that everytime the value is updated, I will get the event triggered:
```
void ui_event_UpdateProgress(lv_event_t *e)
{
    lv_event_code_t event_code = lv_event_get_code(e);
    lv_obj_t *target = lv_event_get_target(e);
    if (event_code == LV_EVENT_VALUE_CHANGED )
    {
        int value = lv_bar_get_value(ui_UpdateProgress);
        printf("value = %u \n", value);
        if(value == 100){
            //do something
        }
    }
}
```

But this never gets triggered. I would appreciate the suggestion to trigger the value changed event

kisvegabor, 2023-09-17:
Hi,

If the `VALUE_CHANGED` event were sent on manual value change it wasn't possible to **not** send the event. 

If the `VALUE_CHANGED` event is not sent automatically, you have to opportunity to decide wheter you want to send it manually after setting the value or not.

---

However, I understand your issue in SquareLine Studio and I suggest approaching it differently. I think you should add a `Call function` action to the button click to execute a custom function. In this function you can start a normal [LVGL animation](https://docs.lvgl.io/8.3/overview/animation.html), and if the the value is 100, do something extra.

krupis, 2023-09-19:
@kisvegabor 
Thanks for a reply but its not fully clear to me.

My bar LVGL element is initialized as following:
```
    ui_UpdateProgress = lv_bar_create(ui_ConfirmationScreen);
    lv_obj_set_width(ui_UpdateProgress, 150);
    lv_obj_set_height(ui_UpdateProgress, 20);
    lv_obj_set_x(ui_UpdateProgress, 0);
    lv_obj_set_y(ui_UpdateProgress, 70);
    lv_obj_set_align(ui_UpdateProgress, LV_ALIGN_CENTER);
    lv_obj_set_style_bg_color(ui_UpdateProgress, lv_color_hex(0xFFFFFF), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(ui_UpdateProgress, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_anim_time(ui_UpdateProgress, 5000, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_color(ui_UpdateProgress, lv_color_hex(0x409E3B), LV_PART_INDICATOR | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(ui_UpdateProgress, 255, LV_PART_INDICATOR | LV_STATE_DEFAULT);
```

I have replaced my function that is called when the Calibrate button is clicked
```
void ui_event_CalibrateStart(lv_event_t *e)
{
    lv_event_code_t event_code = lv_event_get_code(e);
    lv_obj_t *target = lv_event_get_target(e);
    if (event_code == LV_EVENT_CLICKED)
    {
        _ui_bar_set_property(ui_UpdateProgress, _UI_BAR_PROPERTY_VALUE_WITH_ANIM, 100);
    }
}
```

to your suggested method:

```
void ui_event_CalibrateStart(lv_event_t *e)
{
    lv_event_code_t event_code = lv_event_get_code(e);
    lv_obj_t *target = lv_event_get_target(e);
    if (event_code == LV_EVENT_CLICKED)
    {
        printf("start animation \n");
        //start animation

        lv_anim_t a;
        lv_anim_init(&a);
        lv_anim_set_exec_cb(&a, set_update_progress_value);
        lv_anim_set_values(&a, 0, 100);
        lv_anim_set_time(&a, 5000);
        lv_anim_set_var(&a, ui_UpdateProgress);
        lv_anim_start(&a);
    }
}

static void set_update_progress_value(void *indic, int32_t v)
{
    lv_bar_set_value(ui_UpdateProgress, indic, v);
}
```

The issue with this method is that the value is set to 100 instantly even though I have set time to 5000 (it should take 5 seconds to complete the animation).

Also, after the value has reached 100, the even still has not been triggered:
```
void ui_event_UpdateProgress(lv_event_t *e)
{
    lv_event_code_t event_code = lv_event_get_code(e);
    lv_obj_t *target = lv_event_get_target(e);
    if (event_code == LV_EVENT_VALUE_CHANGED )
    {
        int value = lv_bar_get_value(ui_UpdateProgress);
        printf("value = %u \n", value);
    }
}
```

kisvegabor, 2023-09-20:
The problem should be here:
```c
static void set_update_progress_value(void *indic, int32_t v)
{
    //lv_bar_set_value(ui_UpdateProgress, indic, v);
   lv_bar_set_value(ui_UpdateProgress, v, LV_ANIM_ON);
}
```

krupis, 2023-10-05:
@kisvegabor 
Sorry for the delayed response.

Yes you were right, updating my ```set_update_progress_value function as following:
```
static void set_update_progress_value(void *indic, int32_t v)
{
    //lv_bar_set_value(ui_UpdateProgress, indic, v);
   lv_bar_set_value(ui_UpdateProgress, v, LV_ANIM_ON);
}
```
solved one of my issues. Now the progress bar updates and reaches max value in 5 seconds (which is exactly what I need).

However, I still do not fully understand how can I add a callback to execute specific function when the value reaches 100.

I use this progress bar to indicate calibration progress. After the calibration is complete (5 seconds), the device should perform a reset or reboot. Hence I want to do something like that:

Can you help me understand why this:
```
    ui_UpdateProgress = lv_bar_create(ui_ConfirmationScreen);
    lv_obj_set_width(ui_UpdateProgress, 150);
    lv_obj_set_height(ui_UpdateProgress, 20);
    lv_obj_set_x(ui_UpdateProgress, 0);
    lv_obj_set_y(ui_UpdateProgress, 70);
    lv_obj_set_align(ui_UpdateProgress, LV_ALIGN_CENTER);
    lv_obj_set_style_bg_color(ui_UpdateProgress, lv_color_hex(0xFFFFFF), LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(ui_UpdateProgress, 255, LV_PART_MAIN | LV_STATE_DEFAULT);
    lv_obj_set_style_anim_time(ui_UpdateProgress, 5000, LV_PART_MAIN | LV_STATE_DEFAULT);

    lv_obj_set_style_bg_color(ui_UpdateProgress, lv_color_hex(0x409E3B), LV_PART_INDICATOR | LV_STATE_DEFAULT);
    lv_obj_set_style_bg_opa(ui_UpdateProgress, 255, LV_PART_INDICATOR | LV_STATE_DEFAULT);

    lv_obj_add_event_cb(ui_UpdateProgress, ui_event_UpdateProgress, LV_EVENT_ALL, NULL);

```

```
void ui_event_UpdateProgress(lv_event_t *e)
{
    lv_event_code_t event_code = lv_event_get_code(e);
    lv_obj_t *target = lv_event_get_target(e);
    if (event_code == LV_EVENT_VALUE_CHANGED)
    {
        int value = lv_bar_get_value(ui_UpdateProgress);
        printf("value = %u \n", value);
    }
}
```

Is not triggering ```ui_event_UpdateProgress``` everytime the value is changed? What is the alternative method?

kisvegabor, 2023-10-06:
> However, I still do not fully understand how can I add a callback to execute specific function when the value reaches 100.

The simplest would be to do this:
```c
static void set_update_progress_value(void *indic, int32_t v)
{
   lv_bar_set_value(ui_UpdateProgress, v, LV_ANIM_ON);
   if(v == 100) do_something();
}
```

krupis, 2023-10-09:
@kisvegabor 
That works. Thanks for your help as usual :)

closed 2023-10-09 by krupis

kisvegabor, 2023-10-09:
Happy to help :)