lvgl/lvgl #4563: An obj has been HIDDEN, but it will be affected by LV_EVENT_CLICKED

fuxiujun, 2023-09-17:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

After an obj is pressed, it or its parent is hidden. The obj is no longer displayed, but it will still receive the click event.

### To Reproduce

simple example
```c
static void btn_click_cb(lv_event_t* e)
{
  lv_event_code_t code = e->code;
  lv_obj_t* btn = e->target;
  if (code == LV_EVENT_PRESSED) {
    lv_obj_add_flag(btn, LV_OBJ_FLAG_HIDDEN);
    LV_LOG_USER("pressed, hidden btn");
  } else if (code == LV_EVENT_CLICKED) {
    LV_LOG_USER("btn clicked");
  }
}

static void test(void)
{
  lv_obj_t* btn = lv_btn_create(lv_scr_act());
  lv_obj_add_event_cb(btn, btn_click_cb, LV_EVENT_ALL, NULL);
  lv_obj_center(btn);
}

```

### Expected behavior

The hidden obj should not receive `LV_EVENT_CLICKED`. Should we lv_indev_reset?

### Screenshots or video

kisvegabor, 2023-09-17:
> The hidden obj should not receive LV_EVENT_CLICKED. Should we lv_indev_reset?

Yes, exactly. The hidden flag is considered only when LVGL searches for a new widget on press. When the object is being pressed, LVGL already has a widget so it doesn't check it anymore. 

So resetting the indev is a good idea.

fuxiujun, 2023-09-21:
Hi, I submitted a PR to try to fix this problem, do you have any suggestions?

closed 2023-09-22 by fuxiujun