lvgl/lvgl #4571: Button matrix - click on disabled button sends LV_EVENT_VALUE_CHANGED event with previous clicked button id

gbuss79, 2023-09-19:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
A 'disabled' button sends the LV_EVENT_VALUE_CHANGED and the function lv_btnmatrix_get_selected_btn()
returns the last selected button id.
But I would either expect no event at all or at least that the function lv_btnmatrix_get_selected_btn() returns LV_BTNMATRIX_BTN_NONE.

### To Reproduce
```c
/**
  * slightly modified code example from
  * https://docs.lvgl.io/8.3/widgets/core/btnmatrix.html#simple-button-matrix
  * only one line modifed (marked with MODIFIED) - ORed LV_BTNMATRIX_CTRL_DISABLED
  */

#include "../../lv_examples.h"
#if LV_USE_BTNMATRIX && LV_BUILD_EXAMPLES

static void event_handler(lv_event_t * e)
{
    lv_event_code_t code = lv_event_get_code(e);
    lv_obj_t * obj = lv_event_get_target(e);
    if(code == LV_EVENT_VALUE_CHANGED) {
        uint32_t id = lv_btnmatrix_get_selected_btn(obj);
        const char * txt = lv_btnmatrix_get_btn_text(obj, id);

        LV_LOG_USER("%s was pressed\n", txt);
    }
}

static const char * btnm_map[] = {"1", "2", "3", "4", "5", "\n",
                                  "6", "7", "8", "9", "0", "\n",
                                  "Action1", "Action2", ""
                                 };

void lv_example_btnmatrix_1(void)
{
    lv_obj_t * btnm1 = lv_btnmatrix_create(lv_scr_act());
    lv_btnmatrix_set_map(btnm1, btnm_map);
    lv_btnmatrix_set_btn_width(btnm1, 10, 2);        /*Make "Action1" twice as wide as "Action2"*/
    lv_btnmatrix_set_btn_ctrl(btnm1, 10, LV_BTNMATRIX_CTRL_CHECKABLE | LV_BTNMATRIX_CTRL_DISABLED); /*MODIFIED LINE*/
    lv_btnmatrix_set_btn_ctrl(btnm1, 11, LV_BTNMATRIX_CTRL_CHECKED);
    lv_obj_align(btnm1, LV_ALIGN_CENTER, 0, 0);
    lv_obj_add_event_cb(btnm1, event_handler, LV_EVENT_ALL, NULL);
}

#endif
```

Steps to see the problem (via LV_LOG_USER):
- Click the disabled button 'Action1' (id 10) -> nothing happens, as expected
- Click one of the 'number' buttons, e.g. 7 -> "7 was pressed" appears in the log
- Click the disabled button 'Action1' again ->"7 was pressed" appears again in the log

So the disabled button "repeats" the last selected button (but only once).

### Expected behavior

A 'click' on a disabled button should not send a LV_EVENT_VALUE_CHANGED event or at least
the method lv_btnmatrix_get_selected_btn() should return LV_BTNMATRIX_BTN_NONE (but not the last selected button id).

### Proposal for a simple fix
src/widgets/lv_btnmatrix.c in function lv_btnmatrix_event (round about line 417)
```c
    else if(code == LV_EVENT_PRESSED) {
        void * param = lv_event_get_param(e);
        invalidate_button_area(obj, btnm->btn_id_sel);

        lv_indev_type_t indev_type = lv_indev_get_type(lv_indev_get_act());
        if(indev_type == LV_INDEV_TYPE_POINTER || indev_type == LV_INDEV_TYPE_BUTTON) {
            uint16_t btn_pr;
            /*Search the pressed area*/
            lv_indev_get_point(param, &p);
            btn_pr = get_button_from_point(obj, &p);
            /*Handle the case where there is no button there*/
            if(btn_pr != LV_BTNMATRIX_BTN_NONE) {
                if(button_is_inactive(btnm->ctrl_bits[btn_pr]) == false &&
                   button_is_hidden(btnm->ctrl_bits[btn_pr]) == false) {
                    btnm->btn_id_sel = btn_pr;
                    invalidate_button_area(obj, btnm->btn_id_sel); /*Invalidate the new area*/
                }
/* BEGIN OF FIX */
                else {
                    btnm->btn_id_sel = LV_BTNMATRIX_BTN_NONE;
                }
/* END OF FIX */
            }
        }
```

kisvegabor, 2023-09-20:
Hi,

Thank you for reporting this issue. 

I've just looked in to and I agree with the proposed fix. Could you send a Pull request?

gbuss79, 2023-09-20:
Hi,

I'm very excited - my very first pull request ;-)

kisvegabor, 2023-09-21:
It was perfect, well done :blush: 

Thank you! 

I've also added the same fix to `master`: https://github.com/lvgl/lvgl/commits/master

closed 2023-09-21 by kisvegabor