lvgl/lvgl #4575: dropdown: list displays abnormal if setting different fonts

ting-liu, 2023-09-20:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
drowdown list displays abnormal if setting different fonts for SELECTED and MAIN part

### To Reproduce
Using online Simulator, both 8.3 and 9.0.
```python

# Initialize

import display_driver
import lvgl as lv

# Create a normal drop down list
dd = lv.dropdown(lv.scr_act())
dd.set_options("\n".join([
    "Apple",
    "Banana",
    "Orange",
    "Cherry",
    "Grape",
    "Raspberry",
    "Melon",
    "Orange",
    "Lemon",
    "Nuts"]))

dd.align(lv.ALIGN.TOP_MID, 0, 20)
ddlist = dd.get_list()
ddlist.set_style_text_font(lv.font_montserrat_14, lv.PART.MAIN|lv.STATE.DEFAULT)
ddlist.set_style_text_font(lv.font_dejavu_16_persian_hebrew, lv.PART.SELECTED|lv.STATE.CHECKED)

```

### Expected behavior

### Screenshots or video

[image]
[image]

C47D, 2023-09-22:
Thanks for your report @ting-liu , I'm able to reproduce the issue in master. @kisvegabor I've captured the behavior in this video, notice how when selecting `Orange` and opening the dropdown again `Melon` appears selected. This only happens when customizing the dropdown list style.

[Screencast from 22-09-23 13:08:56.webm](https://github.com/lvgl/lvgl/assets/11037705/53c250e2-4c7a-4551-b4d8-ef1ce7706cf5)

This is the C version of the code I've used:
```c
  lv_obj_t * dd = lv_dropdown_create(lv_scr_act());
  lv_dropdown_set_options(dd,
    "Apple\n"
    "Banana\n"
    "Orange\n"
    "Cherry\n"
    "Grape\n"
    "Raspberry\n"
    "Melon\n"
    "Orange\n"
    "Lemon\n"
    "Nuts");

  lv_obj_align(dd, LV_ALIGN_TOP_MID, 0, 20);

  lv_obj_t * dd_list = lv_dropdown_get_list(dd);

  static lv_style_t default_state;
  static lv_style_t checked_state;

  lv_style_init(&default_state);
  lv_style_init(&checked_state);

  lv_style_set_text_font(&default_state, &lv_font_montserrat_14);
  lv_style_set_text_font(&checked_state, &lv_font_dejavu_16_persian_hebrew);

  lv_obj_add_style(dd_list, &default_state, LV_PART_MAIN | LV_STATE_DEFAULT);
  lv_obj_add_style(dd_list, &checked_state, LV_PART_SELECTED | LV_STATE_CHECKED);
```
I've also tried not adding the `LV_PART_MAIN` style, but it has the same misbehavior

kisvegabor, 2023-09-25:
I can confirm that it's not supported to use a different font only on the selected option. However you can make to whole list use a different font like this: `lv_obj_set_style_text_font(dd_list, &lv_font_dejavu_16_persian_hebrew, 0);`

kisvegabor, 2023-10-12:
As this is really no supported now I close this issue.

closed 2023-10-12 by kisvegabor