lvgl/lvgl #4576: Arc : clickable area outside of the draw

spilz87, 2023-09-20:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

With a partial arc (120°), when I click outside of these 120° but still in the ring, it changes the value.
which can be a big issue when it controls volume for example and it put it at 100% :(

### To Reproduce

### Expected behavior

I would expect it change only when I click on the drown part of the arc, not on the full ring
### Screenshots or video

None

thanks in advance for your help

C47D, 2023-09-21:
I think this patch make it work as you expected @spilz87 

Does it look good @kisvegabor ?
```patch
diff --git a/src/widgets/arc/lv_arc.c b/src/widgets/arc/lv_arc.c
index 6156328e3..9aafec9d0 100644
--- a/src/widgets/arc/lv_arc.c
+++ b/src/widgets/arc/lv_arc.c
@@ -589,7 +589,7 @@ static void lv_arc_event(const lv_obj_class_t * class_p, lv_event_t * e)
         }
     }
     else if(code == LV_EVENT_HIT_TEST) {
-        lv_hit_test_info_t * info = lv_event_get_param(e);;
+        lv_hit_test_info_t * info = lv_event_get_param(e);
 
         lv_point_t p;
         lv_coord_t r;
@@ -609,6 +609,28 @@ static void lv_arc_event(const lv_obj_class_t * class_p, lv_event_t * e)
             return;
         }
 
+        /*Calculate the angle of the pressed point using a temporary point*/
+        lv_point_t tmp_point;
+        tmp_point.x = info->point->x;
+        tmp_point.y = info->point->y;
+        /*Make point relative to the arc's center*/
+        tmp_point.x -= p.x;
+        tmp_point.y -= p.y;
+
+        LV_ASSERT_MSG(tmp_point.x == 0 && tmp_point.y == 0, "Both coordinates are 0");
+
+        int16_t angle = lv_atan2(tmp_point.y, tmp_point.x);
+        /*Make the angle relative to the start angle*/
+        angle -= arc->rotation;
+        angle -= arc->bg_angle_start;
+        if(angle < 0) angle += 360;
+
+        /* Point is not in the arc background, do not update value */
+        if(angle > arc->bg_angle_end) {
+            info->res = false;
+            return;
+        }
+
         /*Valid if no clicked outside*/
         lv_area_increase(&a, w + ext_click_area * 2, w + ext_click_area * 2);
         info->res = _lv_area_is_point_on(&a, info->point, LV_RADIUS_CIRCLE);

```

I've attached it, change the extention from txt to patch and apply it using `git apply arc_click_on_bg.patch
[arc_click_on_bg.txt](https://github.com/lvgl/lvgl/files/12680336/arc_click_on_bg.txt)
`

spilz87, 2023-09-21:
@C47D thanks a lot for your very quick reply.
I’m using v8.3, the latest stable release on arduino (if I’m right) 

Yes it looks like the invisible part of the arc is sensible to click which is confusing for UI.

I will modify the file as you suggest and I will let you know in few hours. 

Thanks again

spilz87, 2023-09-21:
I modify the file lv_arc.c as you suggest (if I understand it well)
compile everything
but it doesn't change the behavior

C47D, 2023-09-21:
Can you send me an screenshot of the updated section of the code? I'll check it tomorrow morning, it's midnight here :smile:

spilz87, 2023-09-21:
I will do some test today and I will send you 
It’s morning here :) 
Have a good night and thanks for your help

kisvegabor, 2023-09-21:
Hey,

It's really odd as it works now and I agree with the fix proposed by @C47D. However,
1. It should be added to the `PRESSING` event, else it will work only if `LV_OBJ_FLAG_ADV_HITTEST` is enabled. 
2. We should add a little bit of extra room  before/after the start/end angle, else it would be impossible to set the knob to the very end. Maybe +/- 10 degrees wold make it. However care must be taken is the start and end angles are very close to each other. 

Regarding the +/- 10 degree, it would be better to say we add e.g. `LV_DPX(50)` (LV_DPX considers the DPI too) tolerance.  And we need convert the pixels to angle like: 
```
circumference = (2 * radius * 314) / 100; //avoid float
tolerance_deg = 360 * lv_dpx(50) / circumference;
```

spilz87, 2023-09-21:
the flag `LV_OBJ_FLAG_ADV_HITTEST` is not very clear for me about it's consequences

spilz87, 2023-09-21:
I add : 

```
/* Point is not in the arc background, do not update value */
        if(angle > arc->bg_angle_end) {
            return;
        }
```

in `PRESSING` event after `if(angle < 0) angle += 360;` 
it seems to work :)

C47D, 2023-09-21:
@kisvegabor I'll update the patch with your suggestions and send a PR :smile_cat:

spilz87, 2023-09-21:
> Hi @kisvegabor, thanks for taking a look at the proposal, I'll update it with your suggestions and send a PR 😸

if there is a way it's done in v8.3 relaese too as it's the one used by arduino users, it would be very nice :)

spilz87, 2023-09-21:
thanks a lot for your help :D

C47D, 2023-09-22:
@kisvegabor Could you expand on this? I don't follow the need to convert angle to pixels, shouldn't the arc range angles be enough?
> Regarding the +/- 10 degree, it would be better to say we add e.g. LV_DPX(50) (LV_DPX considers the DPI too) tolerance. And we need convert the pixels to angle like

@spilz87 Sure, I can send the fix on master version of lvgl and it will be also ported into v8.3

spilz87, 2023-09-22:
> if(angle > arc->bg_angle_end) {
>             return;
>         }

I think this is enought.
i can access to 0 and 100% with no problem

kisvegabor, 2023-09-22:
> i can access to 0 and 100% with no problem

Even if you just click on an end? I think it's easy to miss the end with one pixel and in this nothing will happen.

> Could you expand on this? I don't follow the need to convert angle to pixels, shouldn't the arc range angles be enough?

Because with angles on a smaller arc the same angle means smaller sensitive area than on larger arc. And you finger has the same size, so  we should make the sensitive area independent of the actual size of the arc.

kisvegabor, 2023-10-07:
Let's continue in #4586.

PR #4586 (2023-10-12): fix(arc): Handle click outside background angle range
### Description of the feature or fix

Fixes #4576 

### Checkpoints
- [ ] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed
- [ ] Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant. 
- [ ] Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- [ ] If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.p [... cut]
closed 2023-10-07 by kisvegabor