lvgl/lvgl #4588: how to stop default style transition overwriting non-default?

hyansuper, 2023-09-23:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

I want to make `translate_x` style transition on default state, but **not** on pressed state,

below is the code I tested on python online editor https://sim.lvgl.io/v9.0/micropython/ports/javascript/index.html

### To Reproduce
```python
import lvgl as lv
import display_driver

props_def = [lv.STYLE.TRANSLATE_X, 0]
trans_def = lv.style_transition_dsc_t()
trans_def.init(props_def, lv.anim_t.path_linear, 1000, 0, None)

trans_pr = lv.style_transition_dsc_t()
trans_pr.init([0], lv.anim_t.path_linear, 1000, 0, None)

style_def = lv.style_t()
style_def.init()
style_def.set_translate_x(0)
style_def.set_transition(trans_def)

style_pr = lv.style_t()
style_pr.init()
style_pr.set_translate_x(100)
style_pr.set_transition(trans_pr)

obj = lv.obj(lv.scr_act())
obj.add_style(style_def, 0)
obj.add_style(style_pr, lv.STATE.PRESSED)

obj.center()
obj.set_style_bg_color(lv.palette_main(lv.PALETTE.GREY), 0)
```

### Expected behavior
when I press the obj ( hold pressed for a sec to see the effect), I expect the obj to "jump" to x=100, and only make trasitional return when I release.
But it acually make transitional effect on both pressed and release state.

C47D, 2023-09-24:
Hi @hyansuper , is this the behavior you expect? The cursor is not recorded, but I pressed the object and it jumped to 100, then I released the clic and it moved back to 0 smoothly

[Screencast from 24-09-23 12:48:41.webm](https://github.com/lvgl/lvgl/assets/11037705/f07d2784-0d42-44f8-af75-e1363bfd9eb7)

If so, this is the code, left some comments with the changes I added
```python
import lvgl as lv
import display_driver

props_def = [lv.STYLE.TRANSLATE_X, 0]
trans_def = lv.style_transition_dsc_t()
trans_def.init(props_def, lv.anim_t.path_linear, 1000, 0, None)

props_pr = [lv.STYLE.TRANSLATE_X, 0] # Added lv.STYLE.TRANSLATE_X, def needed to work
trans_pr = lv.style_transition_dsc_t()
trans_pr.init(props_pr, lv.anim_t.path_linear, 0, 0, None) # Set delay to 0, so the transition is instant

style_def = lv.style_t()
style_def.init()
style_def.set_translate_x(0)
style_def.set_transition(trans_def)

style_pr = lv.style_t()
style_pr.init()
style_pr.set_translate_x(100)
style_pr.set_transition(trans_pr)

obj = lv.obj(lv.scr_act())
obj.add_style(style_def, lv.STATE.DEFAULT) # Added lv.STATE.DEFAULT, I don't think it's needed
obj.add_style(style_pr, lv.STATE.PRESSED)

obj.center()
obj.set_style_bg_color(lv.palette_main(lv.PALETTE.GREY), 0)

```

hyansuper, 2023-09-25:
Thank you, that will do! 
But my real application is a bit more complex.

Consider that I want to animate two properties `translate_x` and `opa` on release, but only want to animate `translate_x` on press, `opa` should "jump" to 100% on press.

```python
import lvgl as lv
import display_driver

props_def = [lv.STYLE.TRANSLATE_X, lv.STYLE.OPA, 0]  # animate 2 properties on release
trans_def = lv.style_transition_dsc_t()
trans_def.init(props_def, lv.anim_t.path_linear, 1000, 0, None)

props_pr = [lv.STYLE.TRANSLATE_X, 0] # animate only one on press, turns out both properties are animated
trans_pr = lv.style_transition_dsc_t()
trans_pr.init(props_pr, lv.anim_t.path_linear, 1000, 0, None)

style_def = lv.style_t()
style_def.init()
style_def.set_translate_x(0)
style_def.set_opa(25)
style_def.set_transition(trans_def)

style_pr = lv.style_t()
style_pr.init()
style_pr.set_translate_x(100)
style_pr.set_opa(255)
style_pr.set_transition(trans_pr)

obj = lv.obj(lv.scr_act())
obj.add_style(style_def, lv.STATE.DEFAULT)
obj.add_style(style_pr, lv.STATE.PRESSED)

obj.center()
obj.set_style_bg_color(lv.palette_main(lv.PALETTE.GREY), 0)
```

C47D, 2023-09-25:
We might need some help from @kisvegabor to know if that's possible (transforming to 100% both x and opa) when pressing the obj

kisvegabor, 2023-10-12:
As we have no better option than the suggested workaround I close this issue.

closed 2023-10-12 by kisvegabor