lvgl/lvgl #4594: Not used macros to remove

PGNetHun, 2023-09-24:
### Introduce the problem
After the multi-instance and parallel rendering architecture changes some macros are left in documentations and in other config files (for example in .devcontainer, lv_conf_cmsis.h).

**Before cleaning up the source code and documentations, let's discuss which of these SHOULD not be removed.**

The macros are collected from [MicroPython binding repo](https://github.com/lvgl/lv_binding_micropython/) (which still used the "old" architecture), from [lv_conf.h](https://github.com/lvgl/lv_binding_micropython/blob/8053b277cdad75973abb6fab93bb86d71c5897c9/lv_conf.h) file.

### Examples and cases
- LV_DRAW_SDL_CUSTOM_BLEND_MODE
- LV_DRAW_SDL_INCLUDE_PATH
- LV_DRAW_SDL_LRU_SIZE
- LV_DRAW_SW_FONT_SUBPX
- LV_DRAW_SW_FONT_SUBPX_BGR
- LV_DRAW_SW_GRADIENT_CACHE_DEF_SIZE
- LV_DRAW_SW_GRADIENT_DITHER
- LV_DRAW_SW_GRADIENT_DITHER_ERROR_DIFFUSION
- LV_DRAW_SW_LAYER_SIMPLE_FALLBACK_BUF_SIZE
- LV_ENABLE_GC
- LV_FONT_MONTSERRAT_12_SUBPX
- LV_GC_INCLUDE
- LV_GC_ROOT
- LV_GPU_DMA2D_CMSIS_INCLUDE
- LV_GPU_SWM341_DMA2D_INCLUDE
- LV_IMG_CACHE_DEF_SIZE
- LV_SDL_PARTIAL_MODE
- LV_TICK_CUSTOM
- LV_TICK_CUSTOM_INCLUDE
- LV_TICK_CUSTOM_SYS_TIME_EXPR
- LV_TXT_COLOR_CMD
- LV_USE_COLORWHEEL
- LV_USE_DRAW_MASKS
- LV_USE_DRAW_SDL
- LV_USE_GPU_ARM2D
- LV_USE_GPU_GD32_IPA
- LV_USE_GPU_NXP_PXP
- LV_USE_GPU_NXP_PXP_AUTO_INIT
- LV_USE_GPU_NXP_VG_LITE
- LV_USE_GPU_STM32_DMA2D
- LV_USE_GPU_SWM341_DMA2D
- LV_USE_PNG
- LV_USE_SJPG
- LV_USE_USER_DATA

### Suggested solution
We should remove the unused macros.

Other suggestion related to macros:
There are some GPU related macros, too. (`LV_USE_GPU_`) 
Those are HW specific, and some are used in different port repos (for example: `LV_USE_GPU_ARM2D` -> https://github.com/lvgl/lv_port_an547_cm55_sim/blob/Using-LVGL8/lv_gpu_arm2d.c#L60 ).

I think these GPU (or HW) specific configs should go to a sub-config, for example: `lv_conf_hw.h`  (`lv_conf.h` might include it).
The aim would be to keep the generic LVGL config and HW specific configs separately, so there will be no big mess when there are new changes in LVGL config (`lv_conf_template.h`), and we need to copy those changes to other bindings and HW-specific ports.

PGNetHun, 2023-09-25:
Ok, I will start to remove the above mentioned macros.

But about the config grouping changes: I don't know much about the internal architecture of LVGL, so it's up to you how to change or keep actual config file.

All I understand/think is that configs could be separated into 2 main groups:
1. **Platform independent**: feature toggles and feature settings, examples, demo, 3rd party libraries, themes, etc (basically your `lv_conf_ui.h`)
2. **Platform dependent**: HAL, rendering, OS, drivers, compiler settings, etc (same as your `lv_conf_sys.h` )

kisvegabor, 2023-09-26:
Thanks.

Separating `lv_conf.h` already came up multiply times, and I will keep thinking about it.

PGNetHun, 2023-09-26:
Cleanup PRs:
https://github.com/lvgl/lvgl/pull/4593
https://github.com/lvgl/lvgl/pull/4607

PGNetHun, 2023-09-26:
**TODO**

clean up config files of other environments:
- https://github.com/lvgl/lvgl/blob/master/.devcontainer/__lv_conf.h__
- https://github.com/lvgl/lvgl/blob/master/env_support/cmsis-pack/lv_conf_cmsis.h

PGNetHun, 2023-09-27:
@kisvegabor 
where is that backlog you mentioned earlier?
Is it this file?  https://github.com/lvgl/lvgl/blob/master/docs/ROADMAP.rst
(sorry, but I could not find any place where the TODOs are listed)

kisvegabor, 2023-09-28:
> where is that backlog you mentioned earlier?

I just meant the docs is not updated yet according to the latest changes of LVGL.

PGNetHun, 2023-10-13:
Not stale

PGNetHun, 2023-10-29:
Ticket can be closed, we will remove unused macros when they are found.

closed 2023-10-29 by PGNetHun

clouds56, 2024-03-09:
here `LV_TICK_CUSTOM` in https://docs.lvgl.io/master/integration/framework/arduino.html#configure-lvgl

kisvegabor, 2024-03-11:
Thanks, I fixed it in #5832.