lvgl/lvgl #4597: Understanding the benchmark frame time - v8.3.9

nicusorcitu, 2023-09-25:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
I have run the benchmark using 8.3.9 release.
I modified the CPU usage to be taken from the FreeRTOS idle task, so the load percentage become more accurate. I can also share the results I have so far for LVGL v8, but there is something I can not understand on benchmark results.

In order for the results to become more precise, I am using an us timer from DWT (data watchpoint and trace) on RT1170.
The FPS calculation in benchmark is now:
`scenes[scene_act].fps_normal = (1000000 * scenes[scene_act].refr_cnt_normal) / scenes[scene_act].time_sum_normal;`

But I wanted to see more, so I kept the elapsed time for each frame in a vector to figure out how that evolves.
I Could clearly see that some frames are taken 7ms while others 20ms..
Actually I see a pattern that gets repeated. E.G. on rectangle frames I see this repeating:
```
Rectangle:7171 ms, 921600 px
Rectangle:9595 ms, 921600 px
Rectangle:15019 ms, 921600 px
Rectangle:20827 ms, 921600 px
```
What we did next, was to remove the draw callbacks and see where that oflload cames from, so it is not from drawing.
Results without the draw callbacks (remember this is v8 version):
```
Rectangle:3131 us, 921600 px
Rectangle:5171 us, 921600 px
Rectangle:9595 us, 921600 px
Rectangle:15019 us, 921600 px
```
So it is clead that drawing operations (by CPU) takes around 4-5 ms, but where is the extra time came from? And why is so difference between frames? Do you have any idea why this pattern?

Thanks,
Nicu.

### To Reproduce
Environment:

SDK 2.14.0
LVGL v8.3.9
MCUXpresso - Debug build
Optimization level -O3
Frame buffer in SDRAM
App binary and source images in FLASH / OCRAM
`LV_COLOR_DEPTH = 16` and `LV_DISP_DEF_REFR_PERIOD = 30`
Src images are the same (cogweels) but resized to 128x128 pixels
Double buffer mode
CPU only - GPU is disabled!

### Expected behavior

### Screenshots or video

nicusorcitu, 2023-09-25:
[0001-Log-each-frame-time-in-us.patch](https://github.com/lvgl/lvgl/files/12712512/0001-Log-each-frame-time-in-us.patch)
[Time_per_frame_results_draw.txt](https://github.com/lvgl/lvgl/files/12712514/Time_per_frame_results_draw.txt)
[Time_per_frame_results_no_draw.txt](https://github.com/lvgl/lvgl/files/12712515/Time_per_frame_results_no_draw.txt)

kisvegabor, 2023-09-25:
Hi,

How often do you call `lv_timer_handler`? Is there a delay after that? If so, please remove it.

Please also set `LV_DISP_DEF_REFR_PERIOD` to `1` in `lv_conf.h`

nicusorcitu, 2023-09-25:
> How often do you call lv_timer_handler? Is there a delay after that? If so, please remove it.
Yes, there is a 5 sec delay.
```
    for (;;)
    {
        lv_task_handler();
        vTaskDelay(5);
    }
```

> Please also set LV_DISP_DEF_REFR_PERIOD to 1 in lv_conf.h
>
Setting it to 1, does not mean that refresh_timer handler will happen at each 1ms? Is not loading the CPU even more instead of running the lv_disp_refr_timer at 30msec?

kisvegabor, 2023-09-25:
> Yes, there is a 5 sec delay.

I hope it's not 5 sec, but 5 OS ticks (5-10 ms)  :slightly_smiling_face: 

> Setting it to 1, does not mean that refresh_timer handler will happen at each 1ms? Is not loading the CPU even more instead of running the lv_disp_refr_timer at 30msec?

By setting it to 1, LVGL will check if there is something to draw in each ms. As drawing takes more than 1 ms time, it means LVGL will draw as fast as possible. So yes, it's a higher CPU load but result in the best possible FPS.

nicusorcitu, 2023-09-25:
I am sorry, is 5 msec :) But how can this generate the pattern that repeats over again?
```
Rectangle:3131 us, 921600 px
Rectangle:5171 us, 921600 px
Rectangle:9595 us, 921600 px
Rectangle:15019 us, 921600 px
```

> So yes, it's a higher CPU load but result in the best possible FPS.
>
Ok, the problem here is not the FPS, but why do I see this big difference on frames?

kisvegabor, 2023-09-25:
Can you send a Pull request with your changes in the benchmark? Not to merge it just to be sure we are working with the same code base. After that I can also test if I see the same pattern as well.

nicusorcitu, 2023-09-25:
Yes, I will create a PR against 8.3 branch. I will not include the high resolution timer changes (unless you want to run our NXP SDK) because you can still use the msec timer, at the end the difference is big enough to be detectable in msec. Is that fine?

nicusorcitu, 2023-09-25:
Well, I just checked and basically the only patch that matters is the one attached to this ticket.

nicusorcitu, 2023-09-26:
Hi @kisvegabor 
I just run the benchmark of v9 + vglite, currently from flash only. Some results are better than v8, some are worst. I need to understand what happens because I do not trust the results (sometimes I see they are identical during consecutive tests).

I fixed one issue (stride uninitialized for layer blending).
So far I see two more issues:
1. The letters are not drawn correctly by vglite, the top of the letter is not correct - the first 2 lines from each letter seems is messed up. The bottom looks correct. Need to investigate why.
2. The substract tests are not working. I can fix it from my side, GPU uses all the time BLEN_SRC_OVER so I have to handle the SUBRACT as well. The problem I see here is that only `lv_draw_image_dsc_t` have the field:
`lv_blend_mode_t blend_mode : 4;`
So, how should I handle the tests like `sub_arc_cb()` or `sub_line_cb()` when there is no `blend_mode` on descriptors `lv_draw_arc_dsc_t` or `lv_draw_line_dsc_t`?

kisvegabor, 2023-09-26:
Git couldn't apply that  patch. I've tried it manually, and saw that there are other parts too in your code which are not part of LVGL originally. Can you attach the whole file?

> I need to understand what happens because I do not trust the results (sometimes I see they are identical during consecutive tests).

At least for v9 we need to develop a rock solid benchmarking method to be sure we won't mislead anyone (including ourselves). Fortunately  in v9 we have some very useful display events to save timestamps at various stages of the rendering.

> I fixed one issue (stride uninitialized for layer blending).

I know that it's difficult to send PRs for you due to company policies, so if you send the patch I can merge is manually. 

> The letters are not drawn correctly by vglite, the top of the letter is not correct - the first 2 lines from each letter seems is messed up. The bottom looks correct. Need to investigate why.

Can you `printf` a letter pixel by pixel to see if LVGL passes a correct bitmap?

> So, how should I handle the tests like sub_arc_cb() or sub_line_cb() when there is no blend_mode on descriptors lv_draw_arc_dsc_t or lv_draw_line_dsc_t?

For the sake of simplicity only images can be blended with non-normal blend mode. If an arc has non normal blend mode setting, LVGL will create layer, and you can blend the layer as an image with the given blend mode. 
Note that, for `opa_layered` and `blend_mode` LVGL will slice the layer to `LV_DRAW_SW_LAYER_SIMPLE_BUF_SIZE` sized chunks to save some memory.

nicusorcitu, 2023-09-27:
Please see the custom PR I opened, it was needed one more patch before to add CPU load to the benchmark (which is not needed in our case). But here there are the changes to see how much does it spend each frame from each test.

nicusorcitu, 2023-09-27:
> I know that it's difficult to send PRs for you due to company policies, so if you send the patch I can merge is manually.
>
I will create one more PR when the fixes are more enough. For patch fixes the policy is less restrictive.

nicusorcitu, 2023-09-27:
> For the sake of simplicity only images can be blended with non-normal blend mode. If an arc has non normal blend mode setting, LVGL will create layer, and you can blend the layer as an image with the given blend mode.
Note that, for opa_layered and blend_mode LVGL will slice the layer to LV_DRAW_SW_LAYER_SIMPLE_BUF_SIZE sized chunks to save some memory.
>
Actually this is not true. And I believe neither the approach is not correct.
1. I did a small test:
```
static void sub_img_argb_cb(void)
{
    lv_style_reset(&style_common);
    lv_style_set_image_opa(&style_common, scene_with_opa ? LV_OPA_20 : LV_OPA_COVER);
    lv_style_set_blend_mode(&style_common, LV_BLEND_MODE_SUBTRACTIVE);
    img_create(&style_common, &img_cogwheel_argb8888_128, false, false);
}
```
First issue that I see, with CPU it does nothing on the screen. I think it is not supported yet the blending modes?

Then I see 6 tasks being executed:
- blend SRC_OVER to layer 128x48
- blend layer 128x48 SUBSTRACT to FB
- blend SRC_OVER to layer 128x48
- blend layer 128x48 SUBSTRACT to FB
- blend SRC_OVER to layer 128x32
- blend layer 128x32 SUBSTRACT to FB

So, we end up with 6 tasks when this could have been completed in just one task by the GPU without any additional allocations for layers. And this is the image blending which should support blending modes directly.

2. I checked again, and `lv_draw_line_dsc_t` have also the `blend_mode`, so blend mode can be handled directly without any use of additional layer.

3. All the other objects, rectangle, arcs can also have the blend mode on their descriptor so we can do it in a single shot. What do you think?
Thanks, Nicu.

kisvegabor, 2023-09-27:
>  Please see the custom PR I opened, it was needed one more patch before to add CPU load to the benchmark (which is not needed in our case). But here there are the changes to see how much does it spend each frame from each test.

I also wee the jitter in render time even on PC (~1-8 ms).  I've tried it out on an STM MCU and saw the same issue. I've changed `#define SCENE_TIME      1000000      /*ms*/`, due to to this the rectangles weren't moving for a lot of frames,. I've also enabled `lv_obj_invalidate(lv_scr_act());` in line 843. I hoped a more consistent result, but it was the same.

However, when made the flush_cb return immediately, I got a stable result. 
```c
static void tft_flush_cb(lv_disp_drv_t * drv, const lv_area_t * area, lv_color_t * color_p)
{
    lv_disp_flush_ready(drv);
    return;
}
```

Can you confirm it's the same on your end too?

---

> First issue that I see, with CPU only, on this test the screen looks empty. I think CPU does not supported yet all blending modes?

I doesn't well for me either. I'll check it out.

> I checked again, and lv_draw_line_dsc_t have also the blend_mode, so blend mode can be handled directly without any use of additional layer.

It seems I just forgot to remove it :) 

> All the other objects, rectangle, arcs can also have the blend mode on their descriptor so we can do it in a single shot. What do you think?

Maybe we can do the same what we can do with `style_opa` (just scale the `bg_opa`, `border_opa`, etc with this value) and `opa_layer` (create a new layer and lend it with the given opacity). With `blend_mode` we can also have a `blend_mode_layer` option. What do you think?

nicusorcitu, 2023-09-27:
> It seems I just forgot to remove it :)
>
I am actually thinking to add for the others too. We can use directly the blend mode while drawing things:
https://github.com/lvgl/lvgl/blob/8c2fc7ec4ba4b86d9e82475ff37c9f2e7e042aac/src/draw/nxp/vglite/lv_draw_vglite_line.c#L147

>  With blend_mode we can also have a blend_mode_layer option. What do you think?
>
Honestly I do not like at all the idea with layers. I would leave the implementation of it only in the units that does not support a certain task. I signaled from the begginning that is memory consumption is big (we limited but it came with a slice layer mechanism) and it also is fragmenting the task into multiple steps.
See the example I just posted with draw_img task that for an image of128x128 it will create 3 layers. The image blendings + layer blendings will be done in 6 tasks. We could have done it in a single task with no need of layer at all.

kisvegabor, 2023-09-27:
Maybe I was not clear about my idea about `blend_mode_layer`. 
So
- `blend_mode`: Add it to all `draw_dsc`s (line, fill, image, etc) and the draw_unit can decide if it can draw it or not. SW will not take draw task (other than images) with non normal blend_mode  
- `blend_mode_layered` if set, create a layer and blend the image with the set blend mode

I also don't like the idea of layers due to the larger memory consumption however in some cases they produce a different result than applying blend mode or opa widget by widget.  See [this](https://github.com/lvgl/lvgl/pull/4415#issuecomment-1676423724) for example.

Right now layers are user for 4 things:
- opa: the user can decide if they want to create a layer by using `opa_layered` style property
- blend_mode: same as a opa with  
- rotation and scale: it would be difficult to manage it widget by widget, but we can add a `*_layered` property for these as well. 
- masking and post processing: the layer can be clipped, masked or modified anyhow.  In VG-Lite, Can we handle e.g. clipping a widget and all its children to a rounded rectangle mask?

What do you think?

nicusorcitu, 2023-09-27:
> Can you confirm it's the same on your end too?
>
My flush callback is:
```
static void DEMO_FlushDisplay(lv_disp_t *disp, const lv_area_t *area, uint8_t *color_p)
{
    DEMO_FLUSH_DCACHE();

    g_dc.ops->setFrameBuffer(&g_dc, 0, (void *)color_p);

    DEMO_WaitBufferSwitchOff();

    /* IMPORTANT!!!
     * Inform the graphics library that you are ready with the flushing*/
    lv_disp_flush_ready(disp);
}
```
`DEMO_WaitBufferSwitchOff` is a function that waits for a semaphore to be released from display controller IRQHandler.
I believe the IRQHandler gets executed constantly at each 33ms (because of 30Hz refresh) and maybe that is the reason of those patterns!? I.E. Display refresh might happen at different moment time of the frame drawing time.

kisvegabor, 2023-09-27:
And do you really see less jitter with an empty flush callback?

nicusorcitu, 2023-09-28:
> And do you really see less jitter with an empty flush callback?
>
I run the results again from flush (so the timestamp can be bigger that previous which were from OCRAM), but the effect shall be the same:
1. Default config:
`#define LV_DISP_DEF_REFR_PERIOD      30`

Delay 5ms:
```
static void AppTask(void *param)
{
   ...
    lv_demo_benchmark();

    for (;;)
    {
        lv_task_handler();
        vTaskDelay(5);
    }
}
```

FlushDisplay waits for a semaphore to be released from display controller IRQHandler.
```
static void DEMO_FlushDisplay(lv_disp_t *disp, const lv_area_t *area, uint8_t *color_p)
{
    DEMO_FLUSH_DCACHE();

    g_dc.ops->setFrameBuffer(&g_dc, 0, (void *)color_p);

    DEMO_WaitBufferSwitchOff();

    /* IMPORTANT!!!
     * Inform the graphics library that you are ready with the flushing*/
    lv_disp_flush_ready(disp);
}
```

The periodicity result is:
```
Rectangle:12483 us, 921600 px
Rectangle:14902 us, 921600 px
Rectangle:20383 us, 921600 px
Rectangle:24830 us, 921600 px
```
2. Use 1 ms timer and remove delay:
`#define LV_DISP_DEF_REFR_PERIOD      1`

No delay
```
static void AppTask(void *param)
{
   ...
    lv_demo_benchmark();

    for (;;)
    {
        lv_task_handler();
        //vTaskDelay(5);
    }
}
```

The periodicity result is:
```
Rectangle:16510 us, 921600 px
Rectangle:17326 us, 921600 px
```

3. Same as 2, plus remove the semaphore waiting:
```
static void DEMO_FlushDisplay(lv_disp_t *disp, const lv_area_t *area, uint8_t *color_p)
{
    /* IMPORTANT!!!
     * Inform the graphics library that you are ready with the flushing*/
    lv_disp_flush_ready(disp);
}
```
NO periodicity result:
```
Rectangle:13244 us, 921600 px
Rectangle:12532 us, 921600 px
Rectangle:12530 us, 921600 px
Rectangle:12235 us, 921600 px
Rectangle:11921 us, 921600 px
Rectangle:12369 us, 921600 px
Rectangle:12245 us, 921600 px
Rectangle:11735 us, 921600 px
Rectangle:12747 us, 921600 px
Rectangle:12447 us, 921600 px
Rectangle:13430 us, 921600 px
Rectangle:12287 us, 921600 px
```

nicusorcitu, 2023-09-28:
As a conclusion:
Display controller gets an interrupt each 33ms, no matter what (because its 30HZ). In the IRQ handler it will release a semaphore stating that buffers have been switched.
During our frame process, we call the `display->flush_cb()` (in my case it is `DEMO_FlushDisplay()`) which waits for that semaphore to be released from DC IRQ handler.
Because the moment in time is different, the waiting for semaphore creates a pattern - sometimes you need to wait longer, sometimes shorter.

kisvegabor, 2023-09-28:
So finally it's not an LVGL issue , right?

nicusorcitu, 2023-09-28:
Let me measure the time spend in that `flush_cb()` and I will conclude.

nicusorcitu, 2023-09-28:
I saved to an array (to exclude the LOG time) the first 200 execution times from "wait for sempahore" then I print them out. Below is just the first part of it. So there a big time waiting for DC IRQ. The values are in us.
```
DEMO_WaitBufferSwitchOff elaps: 7353
DEMO_WaitBufferSwitchOff elaps: 12038
DEMO_WaitBufferSwitchOff elaps: 17468
DEMO_WaitBufferSwitchOff elaps: 15930
DEMO_WaitBufferSwitchOff elaps: 17006
DEMO_WaitBufferSwitchOff elaps: 15983
DEMO_WaitBufferSwitchOff elaps: 17255
DEMO_WaitBufferSwitchOff elaps: 17208
DEMO_WaitBufferSwitchOff elaps: 17237
DEMO_WaitBufferSwitchOff elaps: 884
DEMO_WaitBufferSwitchOff elaps: 3458
DEMO_WaitBufferSwitchOff elaps: 6995
DEMO_WaitBufferSwitchOff elaps: 12298
DEMO_WaitBufferSwitchOff elaps: 17355
DEMO_WaitBufferSwitchOff elaps: 17214
DEMO_WaitBufferSwitchOff elaps: 17201
DEMO_WaitBufferSwitchOff elaps: 16340
DEMO_WaitBufferSwitchOff elaps: 15956
DEMO_WaitBufferSwitchOff elaps: 16140
DEMO_WaitBufferSwitchOff elaps: 16862
DEMO_WaitBufferSwitchOff elaps: 17456
```

nicusorcitu, 2023-09-28:
So this is not LVLG upstream issue. We need to think how to make a valuable benchmark for lvgl v9.
RENDER only shall mean no DC IRQ wait.

kisvegabor, 2023-09-29:
Yes, in v9 RENDER_ONLY uses an[ empty flush_cb](https://github.com/lvgl/lvgl/blob/master/demos/benchmark/lv_demo_benchmark.c#L930) so DC IRQ shouldn't ne involved.  

Adding a polling to the flush_cb instead of using a semaphore would be slower but made the results cleaner.

nicusorcitu, 2023-10-02:
I will close this topic as well as the PR and I will open a separate discussion for `blend_mode_layer` as this is an important topic on where shall layers be used or not.

closed 2023-10-02 by nicusorcitu

nicusorcitu, 2023-10-06:
> > The letters are not drawn correctly by vglite, the top of the letter is not correct - the first 2 lines from each letter seems is messed up. The bottom looks correct. Need to investigate why.
>>
> Can you printf a letter pixel by pixel to see if LVGL passes a correct bitmap?
>
I figured out the root cause. It is because of the CPU cache. I am clearing the CPU cache before the CPU compose the buffers with each mask.
```
void lv_draw_buf_invalidate_cache(lv_draw_buf_t  * draw_buf, const char * area)
```
I would change the definition of this function to:
```
void lv_draw_buf_invalidate_cache(void *addr, uint32_t size)
```
So that I can call it also for buffers which are not allocated with draw_buf APIs.

nicusorcitu, 2023-10-06:
@kisvegabor

kisvegabor, 2023-10-08:
I see the problem, however by having only `void * buf, size_t size` it's more complicated to invalidate only a smaller rectangle inside a memory block.

Anyway, it's still possible and probably it's not a common task, so I updated the API as you have suggested: e12ac1ae9

I started to think about maybe we don't need `lv_draw_buf_t` as all, but make the operation work directly on `void * buf`.
Now only these functions uses `lv_draw_buf_t`:
```c
typedef uint32_t (*lv_draw_buf_get_stride_cb)(const lv_draw_buf_t * draw_buf);
typedef void * (*lv_draw_buf_go_to_xy_cb)(lv_draw_buf_t * draw_buf, lv_coord_t x, lv_coord_t y);
typedef void (*lv_draw_buf_clear_cb)(lv_draw_buf_t * draw_buf, const lv_area_t * a);
```
where we could pass a `buf` and a `stride` directly as well. What do you think?

nicusorcitu, 2023-10-09:
> however by having only void * buf, size_t size it's more complicated to invalidate only a smaller rectangle inside a memory block.
>
I am thinking to use the functionality of `go_to_x_y` so that I jump with the buf pointer as close as possible to the area [x1, y1] (keeping the alignment requirement). The size will be only the size of the area.

> In these we could pass a buf and a stride directly as well.
>
We need the `color_format` too as argument. Please keep in mind that VGLite use its own variant of CPU's `lv_color_format_get_size()` which is `vglite_get_px_size()` that returns the value in bits (to cover the index formats as well).

nicusorcitu, 2023-10-09:
> I started to think about maybe we don't need lv_draw_buf_t at all, but make the operation work directly on void * buf.
>
I would say:
1. Either we keep the `lv_draw_buf_t` but we need to make use of it all over the code. E.G. on fonts we will have to modify [this descriptor](https://github.com/lvgl/lvgl/blob/a7d03e09ef1c8ab0d435561583cfb90d00c7e9e1/src/draw/lv_draw_label.c#L413) to have a `lv_draw_buf_t` pointer and not simply a void * buffer like now.
2. We get rid of `lv_draw_buf_t` and use only void * and other arguments, as some of the APIs are already independent.
Please also not the inconsistency between the two buf operations: [buf_clear](https://github.com/lvgl/lvgl/blob/a7d03e09ef1c8ab0d435561583cfb90d00c7e9e1/src/draw/lv_draw_buf.c#L198) and [buf_copy](https://github.com/lvgl/lvgl/blob/a7d03e09ef1c8ab0d435561583cfb90d00c7e9e1/src/draw/lv_draw_buf.c#L207)

kisvegabor, 2023-10-09:
Thank you for the feedback! 

I'll make some experiments this week, and we will see how it goes.

nicusorcitu, 2023-10-10:
The only issue that bothers me now is that in case of fonts, the buffer is aligned to 8 bytes (which is enough for VGLite constraints).
But if I want to clean and invalidate data cache of that buffer, the buffer needs to be aligned to 32 bytes (ARM constraints).
If I change the draw_buf allocator so the buffer gets alligned all the time to 32 bytes, it will create a bigger buffer for each letter. Suppose we have a letter of 9x10 pixels, it will create a buffer of 32x10. Not sure how to handle this situation.
BTW, in this code [here](https://github.com/lvgl/lvgl/blob/9bf50c65cfc99549584c881fdd158d7ffdc5f09e/src/draw/lv_draw_label.c#L403) I think you shall not use the hard code alignment, but use the allocator from draw_buf.

nicusorcitu, 2023-10-10:
I just add this piece of code to clean and invalidate a specific area from a buffer.
```
void clean_invalidate_dcache(void ** buf, lv_area_t * area, uint32_t stride, lv_color_format_t cf)
{
    uint8_t ** buf_u8 = (uint8_t **)buf;
    uint8_t align_bytes = 32; // ARM require a 32 byte aligned address
    uint8_t bits_per_pixel = vglite_get_px_size(cf);

    uint16_t align_pixels = align_bytes * 8 / bits_per_pixel;
    uint16_t offset_x = 0;

    if(area->x1 >= (lv_coord_t)(area->x1 % align_pixels)) {
        uint16_t shift_x = area->x1 - (area->x1 % align_pixels);

        offset_x = area->x1 - shift_x;
        *buf_u8 += (shift_x * bits_per_pixel) / 8;
    }

    if(area->y1) {
        uint16_t shift_y = area->y1;

        *buf_u8 += shift_y * stride;
    }

    uint16_t size = (offset_x + lv_area_get_width(area)) * lv_area_get_height(area) * 8 / bits_per_pixel;

    lv_draw_buf_invalidate_cache(*buf_u8, size);
}
```

By looking at it, I think the cleanest way would be [solution 1](https://github.com/lvgl/lvgl/issues/4597#issuecomment-1752387379) to keep the `lv_draw_buf_t` but make use of it all over in the code.
In this way we can keep the old definition:
```
void lv_draw_buf_invalidate_cache(lv_draw_buf_t  * draw_buf, const char * area)
```

But it is up to you, I can handle it either way.

kisvegabor, 2023-10-10:
> The only issue that bothers me now is that in case of fonts, the buffer is aligned to 8 bytes (which is enough for VGLite constraints).
But if I want to clean and invalidate data cache of that buffer, the buffer needs to be aligned to 32 bytes (ARM constraints).

As there is only one letter is drawn at a time I think it doesn't really matter if the buffer is larger a little bit. 

> BTW, in this code [here](https://github.com/lvgl/lvgl/blob/9bf50c65cfc99549584c881fdd158d7ffdc5f09e/src/draw/lv_draw_label.c#L403) I think you shall not use the hard code alignment, but use the allocator from draw_buf.

It's really a leftover hack, I'm fixing it now :slightly_smiling_face: 

> But it is up to you, I can handle it either way.

I got cold a little bit, so I had less energy/opportunity to work today, but I'm planning to look into the removal of `lv_draw_buf_t` tomorrow in more detail.

kisvegabor, 2023-10-10:
I've just realized what was my original intention with `bitmap_size = (bitmap_size + 63) & (~63);`. It's used to avoid many allocations if the next buffer is just slightly larger.

Without it if we needed 57, 58, 59, and 60 bytes for the buffer we needed perform 4 allocations. With the round up we allocate 64 instead of 57 once and it will be good for 60 too.

nicusorcitu, 2023-10-11:
No problem, take care!

kisvegabor, 2023-10-11:
I've removed the `lv_draw_buf_t` in 8deea7004. I'm quite happy the result. What do you think?

nicusorcitu, 2023-10-12:
It looks cleaner. I will update my code too. Thanks.

nicusorcitu, 2023-10-12:
Please make the function definition:
`void lv_draw_buf_invalidate_cache(void * buf, uint32_t stride, lv_color_format_t color_format, const lv_area_t * area);`
Otherwise I have to make my own wrapper over the existing definition.

kisvegabor, 2023-10-12:
Updated here: d975e398d

nicusorcitu, 2023-10-13:
`static void buf_clear(void * buf, uint32_t stride, lv_color_format_t color_format, const lv_area_t * a);`
We need to know the buf width and buf height along with buf stride provided. Those width and height can be taken from the area.

kisvegabor, 2023-10-13:
I don't understand the problem. Let's say the buffer is an ARGB8888 300 x 400 buffer, and the area is {10; 20} {100;200}. 

By knowing the stride and color format you can go to the 10;20 coordinate of the area and clear 90 pixels, go the next line , an repeat it 180 times.

Am I missing something?

nicusorcitu, 2023-10-13:
It might work to set to the buffer the "wrong" width and height (taken from the area) as the GPU shall have the stride and should be enough. But it would be a hack. I might face some driver issues.

kisvegabor, 2023-10-16:
> It might work to set to the buffer the "wrong" width and height (taken from the area) as the GPU shall have the stride and should be enough. But it would be a hack. I might face some driver issues.

That was my other idea as well. Just set the `width` to `stride`, and `height` to `area.y2`. But I admit that it's hacky.

So to solve correctly you need this, right?
```c
static void buf_clear(void * buf, uint32_t width, uint32_t height, lv_color_format_t color_format, const lv_area_t * area);
```

nicusorcitu, 2023-10-17:
> So to solve correctly you need this, right?
> `static void buf_clear(void * buf, uint32_t width, uint32_t height, lv_color_format_t color_format, const lv_area_t * area);`
>
Correct.
Looking now at buf_copy I do not have the width and height of the src buf and dest buf either! As it is, buf_copy will only work if the areas are starting from 0,0 in each buffer.

Not sure how to fix it right... it will create too many arguments. Maybe for buf_copy and buf_clear we need a structure for buffers...

kisvegabor, 2023-10-17:
In case of `buf_copy` it seems we can have this:
```c
void lv_draw_buf_copy(void * dest_buf, const lv_area_t * dest_area_full, const lv_area_t * dest_area_target,
                      void * src_buf, const lv_area_t * src_area_full, const lv_area_t * src_area_target, 
                      lv_color_format_t color_format);

```

nicusorcitu, 2023-10-17:
We can have like this:
```
static void buf_clear(void * buf, const lv_area_t * buf_area, lv_color_format_t color_format, const lv_area_t * area_to_clear);
```
```
void lv_draw_buf_copy(void * dest_buf, const lv_area_t * dest_buf_area, const lv_area_t * dest_area_to_copy,
                      void * src_buf, const lv_area_t * src_buf_area, const lv_area_t * src_area_to_copy, 
                      lv_color_format_t color_format);
```
The stride can be take from width and color format. You might rename the arguments per will.

kisvegabor, 2023-10-17:
What I don't like in it is that from `lv_area_t * buf_area` we need just the width and height. Having x1, x2 ,y1 ,y2 is overkill and misleading.  So it could be:

```c
void lv_draw_buf_copy(void * dest_buf, uint32_t w, uint32_t h, const lv_area_t * dest_area_to_copy,
                      void * src_buf,  uint32_t w, uint32_t h, const lv_area_t * src_area_to_copy, 
                      lv_color_format_t color_format);
```
But it's 9 parameters. I prefer max 3. :smile:  Anyway, we can make an exception here if you also agree.

nicusorcitu, 2023-10-17:
Yes, better with `w` and `h`. We can leave as you suggested and we will see how to change that in the future.

nicusorcitu, 2023-10-17:
> But it's 9 parameters. I prefer max 3.
>
If we have used draw_buf * we would have had 4 arguments instead of 9 for `buf_copy` and just 2 args for `buf_clear` :)

kisvegabor, 2023-10-18:
> If we have used draw_buf * we would have had 4 arguments instead of 9 for buf_copy and just 2 args for buf_clear :)

I've just removed it, don't make me cry. :smile:  

However we can really have a type to store buffers ina  generic way if needed. And we can use it only for these few functions when it can make the API much simpler. 

However how would it look like?
```c
lv_draw_buf_init(&draw_buf1, buf, w, h, cf);
lv_draw_buf_init(&draw_buf2, buf, w, h, cf);
lv_area_init(&scr_area, 20, 20, 100, 100);
lv_area_init(&dect_area, 120, 120, 200, 200);

buf_copy(&draw_buf1, &src_area, &draw_buf1, &dest_area);
```

It doesn't seem so much better in the end. :slightly_frowning_face:

nicusorcitu, 2023-10-18:
The overhead is the same by executing few more functions instead of copying on the stack few more arguments.

kisvegabor, 2023-10-18:
I think the overhead is not a big deal, as the copying it self will take way longer. 

As none of the options seem obviously better, let's go with 9 parameter way. Do you agree?

nicusorcitu, 2023-10-18:
I think is ok. I will also update the code after upstream changes are complete.