lvgl/lvgl #4599: Recolor when wrapping text on another line

rlagace-cld, 2023-09-25:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Only the first line displayed use the color parameter when the text is wrapped because it is too long.

### To Reproduce
```c
lv_obj_t* column_create(lv_obj_t* parent)
{
    lv_obj_t* o = lv_obj_create(parent);
    lv_obj_set_style_bg_color(o, lv_color_black(), 0);
    lv_obj_set_style_border_color(o, lv_color_black(), 0);
    lv_obj_set_style_outline_color(o, lv_color_black(), 0);
    lv_obj_set_width(o, lv_pct(100));
    lv_obj_set_flex_flow(o, LV_FLEX_FLOW_ROW);
    lv_obj_set_flex_align(o, LV_FLEX_ALIGN_SPACE_AROUND, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER);
    lv_obj_remove_style(o, NULL, (int)LV_PART_SCROLLBAR | (int)LV_STATE_ANY);
    return o;
}

lv_obj_t* title_create(lv_obj_t* parent, const char* text)
{
    lv_obj_t* o = lv_label_create(parent);
    lv_obj_set_style_text_color(o, lv_color_white(), 0);
    lv_label_set_text(o, text);
    lv_label_set_long_mode(o, LV_LABEL_LONG_WRAP);
    lv_label_set_recolor(o, true);
    lv_obj_set_style_text_font(o, &lv_font_montserrat_20, 0);
    lv_obj_set_width(o, lv_pct(100));
    lv_obj_set_style_text_align(o, LV_TEXT_ALIGN_CENTER, 0);
    return o;
}

void test()
{
    lv_obj_t * screen = column_create(NULL);
    lv_obj_t * title = title_create(screen, "#ff0000 Bug with recolor when wrapping the text in v8.3.10#");
    lv_scr_load(screen);
}
```

### Expected behavior
All the text lines should change color

### Screenshots or video
[image]

C47D, 2023-09-25:
Hi, 

I was trying to replicate it in `master` and noticed it's not possible to color the font:
[image]

Then I went to the v9 docs and the text in the example is also all black:
[image]

Docs for v8.3
[image]

But editing it to be similar to your snippet also fails:
```python
##### startup script #####

#!/opt/bin/lv_micropython -i

import lvgl as lv

import display_driver

##### main script #####

#
# Show line wrap, re-color, line align and text scrolling.
#
label1 = lv.label(lv.scr_act())
label1.set_long_mode(lv.label.LONG.WRAP)      # Break the long lines*/
label1.set_recolor(True)                      # Enable re-coloring by commands in the text
label1.set_text("#0000ff Re-color words of a label, align the lines to the center and  wrap long text automatically.#")
label1.set_width(150)                         # Set smaller width to make the lines wrap
label1.set_style_text_align(lv.ALIGN.CENTER, 0)
label1.align(lv.ALIGN.CENTER, 0, -40)

label2 = lv.label(lv.scr_act())
label2.set_long_mode(lv.label.LONG.SCROLL_CIRCULAR) # Circular scroll
label2.set_width(150)
label2.set_text("It is a circularly scrolling text. ")
label2.align(lv.ALIGN.CENTER, 0, 40)
```
[image]

Any ideas @kisvegabor

kisvegabor, 2023-09-26:
I've removed this feature in v9, because it was too limited. It can be replaced by `lv_span` which has much more features.

However adding a not in v8's docs would be really great. Could you send a PR?

PGNetHun, 2023-09-26:
> I've removed this feature in v9, because it was too limited. It can be replaced by `lv_span` which has much more features.

Oh, I also wondered where label recolor feature is gone :-)
Then we don't need in v9 the `lv_label_set_recolor()` function either, do we?
https://github.com/lvgl/lvgl/blob/master/src/widgets/label/lv_label.c#L208

(I can remove that in my next "cleanup commit", just confirm it or not)

rlagace-cld, 2023-09-29:
Is it possible to add a spangroup in a lv_list?
From what I see in the documentation, it's only possible to add a button or a label in a lv_list.

kisvegabor, 2023-10-02:
Yes, you can add anything to the list. Just create a button or an `lv_obj` with `lv_pct(100) width. 

You can also create an empty button with `lv_list_add_button(list, NULL, NULL)` where you can create any widgets.

rlagace-cld, 2023-10-03:
Thanks, I will close this issue, I was able to rework my code tu use spangroup instead of using recolor in a label inside a list.

closed 2023-10-03 by rlagace-cld

jianglianfang, 2023-11-08:
What if lv_label_set_text_fmt(label_time, "it is %02d - #470033 %02d# now", hour, min)?

PR #4766 (2023-11-08): chore(lv_label): remove codes for recolor feature
### Description of the feature or fix

#4599

### Checkpoints
- [x] Run `code-format.py` from the scripts folder. [astyle](http://astyle.sourceforge.net/install.html) needs to be installed.
- [ ] Update the [Documentation](https://github.com/lvgl/lvgl/tree/master/docs) if needed
- [ ] Add [Examples](https://github.com/lvgl/lvgl/tree/master/examples) if relevant. 
- [ ] Add [Tests](https://github.com/lvgl/lvgl/blob/master/tests/README.md) if applicable.
- [ ] If you added new options to `lv_conf_template.h` run [lv_conf_internal_gen.py](https:// [... cut]