lvgl/lvgl #4602: image setsize: When I set the image size, it always has a mosaic effect. If I use setsize instead of setzoom to achieve the scaling effect, it will achieve the same effect as reactnative.

SUPPH, 2023-09-26:
### Introduce the problem

When I use avatar.setWidth(100);, it always has a mosaic effect. How can I achieve the scaling effect? I'd be grateful if anyone could tell me.
### Examples and cases

### Suggested solution

SUPPH, 2023-09-26:
reactnative example: https://reactnative.dev/docs/image

kisvegabor, 2023-09-26:
In v8, it's really tiled automatically if the widget size is larger. 
Howevem, in LVGL v9 (under development) we will add image sizing modes, similar to CSS.

SUPPH, 2023-09-26:
@kisvegabor Ok, thanks for your answer

SUPPH, 2023-09-27:
@kisvegabor May I ask why there is an operation of adding and subtracting coordinates by 1 in the lv_img_set_zoom function?I would be grateful if you could answer
`a.x1 += obj->coords.x1 - 1;
    a.y1 += obj->coords.y1 - 1;
    a.x2 += obj->coords.x1 + 1;
    a.y2 += obj->coords.y1 + 1;`

kisvegabor, 2023-09-27:
It just makes the area to invalidate slightly larger to avoid rounding errors.

SUPPH, 2023-09-27:
Got it, thank you very much

closed 2023-10-19 by lvgl-bot