lvgl/lvgl #4605: lv_dropdown sends event for the wrong item id

Undef-a, 2023-09-26:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version 
  - I'm packaging a LVGL using package for a distro. I can't upgrade lvgl without upgrading that package. However It looks to me like the relevant code between the two LVGL versions hasn't changed.
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
Selecting an item in an lv_dropdown results in the final item being set. This occurs because the y point sent to `get_id_on_point` from `list_release_handler` is 0, resulting in an opt of 65534. As a result the error case in `get_id_on_point` is triggered and the final option of the dropdown list is always set.

Some debug prints of this occuring can be seen below:

```
[Warn]	(258.231, +0)	 list_press_handler: ### CALLING get_id_on_point from list_press_handler: y=137
 	(in lv_dropdown.c line #1099)
[Warn]	(258.231, +0)	 get_id_on_point: LINE HEIGHT IS 41
 	(in lv_dropdown.c line #1119)
[Warn]	(258.231, +0)	 get_id_on_point: Y POINT IS 55
 	(in lv_dropdown.c line #1120)
[Warn]	(258.231, +0)	 get_id_on_point: OPT IS 1
 	(in lv_dropdown.c line #1123)
[Warn]	(258.231, +0)	 list_press_handler: ### pr_opt_id set to 1
 	(in lv_dropdown.c line #1101)
[Info]	(258.236, +0)	 indev_proc_press: pressed at x:247 y:137 	(in lv_indev.c line #822)
[Info]	(258.241, +0)	 indev_proc_press: pressed at x:247 y:137 	(in lv_indev.c line #822)
[Info]	(258.250, +3)	 libinput_poll_worker: libinput: event read 	(in libinput.c line #344)
[Info]	(258.256, +0)	 indev_proc_release: released 	(in lv_indev.c line #970)
[Warn]	(258.257, +0)	 list_release_handler: CALLED list_release_handler

 	(in lv_dropdown.c line #1047)
[Warn]	(258.257, +0)	 lv_indev_get_point: EVERYTHING WENT WELL, GETTING CORRECT POINT
 	(in lv_indev.c line #235)
[Warn]	(258.257, +0)	 list_release_handler: IF WE JUST USE pr_opt_id? 1
 	(in lv_dropdown.c line #1066)
[Warn]	(258.257, +0)	 list_release_handler: SETTING DROPDOWN POINT BASED ON Y VALUE: 0
 	(in lv_dropdown.c line #1067)
[Warn]	(258.257, +0)	 get_id_on_point: LINE HEIGHT IS 41
 	(in lv_dropdown.c line #1119)
[Warn]	(258.257, +0)	 get_id_on_point: Y POINT IS -82
 	(in lv_dropdown.c line #1120)
[Warn]	(258.257, +0)	 get_id_on_point: OPT IS 65534
 	(in lv_dropdown.c line #1123)
[Warn]	(258.257, +0)	 get_id_on_point: OPT WAS BIGGER THAN dropdown->option_cnt: o: 65534, c: 4
 	(in lv_dropdown.c line #1126)
[Warn]	(258.257, +0)	 list_release_handler: IMMEDIATELY AFTER get_id_on_point Y is 0 and sel_opt_id is 3
 	(in lv_dropdown.c line #1069)
[Warn]	(258.257, +0)	 list_release_handler: sel_opt_id is 3
 	(in lv_dropdown.c line #1078)
[Warn]	(258.257, +0)	 list_release_handler: VALUE CHANGED CALLED FROM list_release_handler with index 3 	(in lv_dropdown.c line #1080)
[Warn]	(258.257, +0)	 list_release_handler: END CALL list_release_handler
```

Note that the call to `get_id_on_point` from `list_press_handler` in the first section is correct and results in the correct option being selected.

As a workaround I've currently patched `list_release_handler` to leach off the value set in `list_press_handler` with the following patch:
```
-    if(lv_indev_get_type(indev) == LV_INDEV_TYPE_POINTER || lv_indev_get_type(indev) == LV_INDEV_TYPE_BUTTON) {
-        lv_point_t p;
-        lv_indev_get_point(indev, &p);
-        dropdown->sel_opt_id     = get_id_on_point(dropdown_obj, p.y);
-        dropdown->sel_opt_id_orig = dropdown->sel_opt_id;
+    if(indev && (lv_indev_get_type(indev) == LV_INDEV_TYPE_POINTER || lv_indev_get_type(indev) == LV_INDEV_TYPE_BUTTON)) {
+        /* lv_point_t p; */
+        /* lv_indev_get_point(indev, &p); */
+        /* LV_LOG_WARN("IF WE JUST USE pr_opt_id? %d\n", dropdown->pr_opt_id); */
+        /* LV_LOG_WARN("SETTING DROPDOWN POINT BASED ON Y VALUE: %d\n", p.y); */
+        /* dropdown->sel_opt_id     = get_id_on_point(dropdown_obj, p.y); */
+        /* LV_LOG_WARN("IMMEDIATELY AFTER get_id_on_point Y is %d and sel_opt_id is %d\n", p.y, dropdown->sel_opt_id); */
+        /* dropdown->sel_opt_id_orig = dropdown->sel_opt_id; */
+        dropdown->sel_opt_id = dropdown->pr_opt_id;
+        dropdown->sel_opt_id_orig = dropdown->pr_opt_id;
     }
```
This is obviously not a correct solution, but does result in correct output for this case.

### To Reproduce
This issue is likely somewhat platform dependent. It was originally reported as https://gitlab.com/cherrypicker/unl0kr/-/issues/45, but that project's maintainer is unable to reproduce it on postmarketOS. On Debian it happens 100% of the time. The Debian package (with bug) is available at https://salsa.debian.org/DebianOnMobile-team/unl0kr.

### Expected behavior
The correct touch release Y point should be returned by `lv_indev_get_point` and the correct option should be returned by `list_release_handler`.

Undef-a, 2023-09-27:
I believe it's just directly using libinput via lv_drivers: https://gitlab.com/cherrypicker/unl0kr/-/blob/master/indev.c?ref_type=heads

kisvegabor, 2023-09-28:
I think the issue  is in `libinput_read_state` (I haven't found the implementation). Just be sure to not set the `data->point.x` and `data->point.y` when the touchpad is released.

I hope it will solve the problem.

Johennes, 2023-09-30:
Thanks both for investigating this. Unl0kr is using the lv_drivers fork from https://github.com/lvgl/lv_drivers/pull/280. Looking at the code again I think we might be missing to set `point` on touch release events as you say: https://github.com/calebccff/lv_drivers/blob/caleb/unl0kr-disp-input-fixes/indev/libinput.c#L586

If you look at the diff in https://github.com/lvgl/lv_drivers/pull/280/files, the old code didn't do this either though. So I'm slightly unsure why this regressed. :thinking:

Johennes, 2023-09-30:
Actually I think I see what's missing. The old code did automatically inherit the `point` value from the previous event (touch down or touch motion) because it used `state->most_recent_touch_point`. I've tried to fix it with https://github.com/calebccff/lv_drivers/commit/7e3135bcae37cbd03c2f003fcc96278671bd8632 / https://gitlab.com/cherrypicker/unl0kr/-/merge_requests/23. If that fixes the problem, we can close this issue here.

Johennes, 2023-10-01:
Looks like that fixed. So we can close this issue. Thanks both for the help.

closed 2023-10-02 by kisvegabor