lvgl/lvgl #4609: img_draw_core issue with temporary buffer

anaGrad, 2023-09-27:
### Perform all steps below and tick them with [x]
- [ ] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [ ] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [ ] Update lvgl to the latest version
- [ ] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

With vglite enabled I ran a test with an image 128x128 rotated and recolored.
```
static void img_rgb_recolor_rot_cb(void)
{
    lv_style_reset(&style_common);
    lv_style_set_image_opa(&style_common, scene_with_opa ? LV_OPA_20 : LV_OPA_COVER);
    lv_style_set_image_recolor_opa(&style_common, LV_OPA_20);
    img_create(&style_common, &img_cogwheel_rgb565_128, true, false);
}
static void img_create(lv_style_t * style, const void * src, bool rotate, bool zoom)
{
    lv_obj_t * obj = lv_img_create(scene_bg);
    lv_obj_remove_style_all(obj);
    lv_obj_add_style(obj, style, 0);
    lv_img_set_src(obj, src);
    lv_obj_set_style_image_recolor(obj, lv_palette_main(LV_PALETTE_ORANGE), 0);

    if(rotate) {
        /*Rotation degrees. Pivot is center.*/
        lv_img_set_angle(obj, ANGLE);
    }
    if(zoom) {
        /*Zoom in half size*/
        lv_img_set_zoom(obj, LV_ZOOM_NONE * 0.6);
    }

    lv_obj_set_pos(obj, LCD_WIDTH / 2, LCD_HEIGHT / 2);
}
```
The image task (rotate and recolor) i taken by the CPU but is not handled correctly. The result is below.
[image]

The root cause is the CPU use the draw_buf callbacks of the VgLite, hence in this code:
https://github.com/lvgl/lvgl/blob/8c2fc7ec4ba4b86d9e82475ff37c9f2e7e042aac/src/draw/sw/lv_draw_sw_img.c#L289
The stride for blend_area is being calculated by vglite callback, but the CPU is not handling it correctly.

### To Reproduce

In order to reproduce easily the issue, simply use the width_to_stride Vglite implementation.
E.G. for the width of 180px it will return the alignment to 192px * px_size.

### Expected behavior

### Screenshots or video

nicusorcitu, 2023-09-28:
Hi @kisvegabor 
This is an issue we saw on CPU side while using the draw_buf callbacks of the GPU (vglite). I have checked and the _width_to_stride() from vglite does report correctly the stride. It looks like the CPU can not handle images with stride alignment other than 1 byte?!
I do not fully understand the code from
https://github.com/lvgl/lvgl/blob/8c2fc7ec4ba4b86d9e82475ff37c9f2e7e042aac/src/draw/sw/lv_draw_sw_img.c#L284C33-L284C33
And also I am not sure why the allocation is not made via the draw_buf allocator:
https://github.com/lvgl/lvgl/blob/8c2fc7ec4ba4b86d9e82475ff37c9f2e7e042aac/src/draw/sw/lv_draw_sw_img.c#L295

Thanks.

kisvegabor, 2023-09-28:
I added `LV_DRAW_BUF_STRIDE_ALIGN  64` to the unit tests and found several issues. 
The issue you have mentioned should be fixed, but I couldn't fix all the issues yet.

anaGrad, 2023-10-02:
I run the tests again with latest fixes and having in lv_conf:
```
/*Align the start address of draw_buf addresses to this bytes*/
#define LV_DRAW_BUF_ALIGN                       64
```

CPU Only will crash while rotating RGB 45 degree:
```
(gdb) bt                                                                                                            
#0  MemManage_Handler () at ../startup/startup_mimxrt1176_cm7.c:943
#1  <signal handler called>
#2  0x3003c012 in remove_free_block (control=0x80384648 <work_mem_int>, block=0x80387c8c <work_mem_int+13892>, fl=25, sl=0) at ../lvgl/lvgl/src/stdlib/builtin/lv_tlsf.c:592
#3  0x3003c178 in block_remove (control=0x80384648 <work_mem_int>, block=0x80387c8c <work_mem_int+13892>) at ../lvgl/lvgl/src/stdlib/builtin/lv_tlsf.c:637
#4  0x3003c340 in block_merge_next (control=0x80384648 <work_mem_int>, block=0x80386688 <work_mem_int+8256>) at ../lvgl/lvgl/src/stdlib/builtin/lv_tlsf.c:707
#5  0x3003c6c6 in lv_tlsf_free (tlsf=0x80384648 <work_mem_int>, ptr=0x80386690 <work_mem_int+8264>) at ../lvgl/lvgl/src/stdlib/builtin/lv_tlsf.c:1171
#6  0x3001221c in lv_free_core (p=0x80386690 <work_mem_int+8264>) at ../lvgl/lvgl/src/stdlib/builtin/lv_mem_core_builtin.c:177
#7  0x30012af4 in lv_free (data=0x80386690 <work_mem_int+8264>) at ../lvgl/lvgl/src/stdlib/lv_mem.c:109
#8  0x3004d7aa in buf_free (buf=0x80386690 <work_mem_int+8264>) at ../lvgl/lvgl/src/draw/lv_draw_buf.c:165
#9  0x3001a580 in lv_draw_buf_free (buf=0x80386690 <work_mem_int+8264>) at ../lvgl/lvgl/src/draw/lv_draw_buf.c:96
#10 0x3001843e in img_draw_core (draw_unit=0x8038536c <work_mem_int+3364>, draw_dsc=0x80386590 <work_mem_int+8008>, draw_area=0x8040dddc <ucHeap+38244>, src=0x8040dda4 <ucHeap+38188>, sup=0x8040dd98 <ucHeap+38176>, img_coords=0x80386666 <work_mem_int+8222>) at ../lvgl/lvgl/src/draw/sw/lv_draw_sw_img.c:410
#11 0x30048f7e in lv_draw_sw_image (draw_unit=0x8038536c <work_mem_int+3364>, draw_dsc=0x80386590 <work_mem_int+8008>, coords=0x80386666 <work_mem_int+8222>) at ../lvgl/lvgl/src/draw/sw/lv_draw_sw_img.c:185
#12 0x30017b26 in execute_drawing (u=0x8038536c <work_mem_int+3364>) at ../lvgl/lvgl/src/draw/sw/lv_draw_sw.c:157
#13 0x30044afe in render_thread_cb (ptr=0x8038536c <work_mem_int+3364>) at ../lvgl/lvgl/src/draw/sw/lv_draw_sw.c:124
#14 0x3003c8da in prvRunThread (pxArg=0x80385388 <work_mem_int+3392>) at ../lvgl/lvgl/src/osal/lv_freertos.c:312
#15 0x30022904 in pxPortInitialiseStack (pxTopOfStack=0x80385388 <work_mem_int+3392>, pxCode=0xa5a5a5a5, pvParameters=0x80385388 <work_mem_int+3392>) at ../freertos/freertos-kernel/portable/GCC/ARM_CM4F/port.c:221
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

CPU + PXP will crash while recoloring and rotate at 90 degree:
```
(gdb) bt
#0  MemManage_Handler () at ../startup/startup_mimxrt1176_cm7.c:943
#1  <signal handler called>
#2  0x3003d9a2 in remove_free_block (control=0x80384648 <work_mem_int>, block=0x80387ce4 <work_mem_int+13980>, fl=12, sl=30) at ../lvgl/lvgl/src/stdlib/builtin/lv_tlsf.c:592
#3  0x3003ddea in block_locate_free (control=0x80384648 <work_mem_int>, size=5716) at ../lvgl/lvgl/src/stdlib/builtin/lv_tlsf.c:776
#4  0x3003dff4 in lv_tlsf_malloc (tlsf=0x80384648 <work_mem_int>, size=5715) at ../lvgl/lvgl/src/stdlib/builtin/lv_tlsf.c:1102
#5  0x300122c0 in lv_malloc_core (size=5715) at ../lvgl/lvgl/src/stdlib/builtin/lv_mem_core_builtin.c:145
#6  0x30012be0 in lv_malloc (size=5715) at ../lvgl/lvgl/src/stdlib/lv_mem.c:76
#7  0x30017f92 in img_draw_core (draw_unit=0x8038536c <work_mem_int+3364>, draw_dsc=0x80386670 <work_mem_int+8232>, draw_area=0x8040dde4 <ucHeap+38244>, src=0x8040ddac <ucHeap+38188>, sup=0x8040dda0 <ucHeap+38176>, img_coords=0x803866ba <work_mem_int+8306>) at ../lvgl/lvgl/src/draw/sw/lv_draw_sw_img.c:292
#8  0x3004a90e in lv_draw_sw_image (draw_unit=0x8038536c <work_mem_int+3364>, draw_dsc=0x80386670 <work_mem_int+8232>, coords=0x803866ba <work_mem_int+8306>) at ../lvgl/lvgl/src/draw/sw/lv_draw_sw_img.c:185
#9  0x30017c3e in execute_drawing (u=0x8038536c <work_mem_int+3364>) at ../lvgl/lvgl/src/draw/sw/lv_draw_sw.c:157
#10 0x3004648e in render_thread_cb (ptr=0x8038536c <work_mem_int+3364>) at ../lvgl/lvgl/src/draw/sw/lv_draw_sw.c:124
#11 0x3003e26a in prvRunThread (pxArg=0x80385388 <work_mem_int+3392>) at ../lvgl/lvgl/src/osal/lv_freertos.c:312
#12 0x30023ca4 in pxPortInitialiseStack (pxTopOfStack=0x80385388 <work_mem_int+3392>, pxCode=0xa5a5a5a5, pvParameters=0x80385388 <work_mem_int+3392>) at ../freertos/freertos-kernel/portable/GCC/ARM_CM4F/port.c:221
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) [Switching to thread 5 (Thread 4)]
#5  0x300122c0 in lv_malloc_core (size=5715) at ../lvgl/lvgl/src/stdlib/builtin/lv_mem_core_builtin.c:145
145         void * p = lv_tlsf_malloc(state.tlsf, size);
[Switching to thread 5 (Thread 4)]
#6  0x30012be0 in lv_malloc (size=5715) at ../lvgl/lvgl/src/stdlib/lv_mem.c:76
76          void * alloc = lv_malloc_core(size);
[Switching to thread 5 (Thread 4)]
#7  0x30017f92 in img_draw_core (draw_unit=0x8038536c <work_mem_int+3364>, draw_dsc=0x80386670 <work_mem_int+8232>, draw_area=0x8040dde4 <ucHeap+38244>, src=0x8040ddac <ucHeap+38188>, sup=0x8040dda0 <ucHeap+38176>, img_coords=0x803866ba <work_mem_int+8306>) at ../lvgl/lvgl/src/draw/sw/lv_draw_sw_img.c:292
292             uint8_t * tmp_buf = lv_malloc(buf_stride * buf_h);
[Switching to thread 5 (Thread 4)]
[... cut ...] 176>, img_coords=0x803866ba <work_mem_int+8306>) at ../lvgl/lvgl/src/draw/sw/lv_draw_sw_img.c:292
292             uint8_t * tmp_buf = lv_malloc(buf_stride * buf_h);
(gdb) p buf_stride
$7 = 381
(gdb) p blend_w
$8 = 127
(gdb) p cf_final
$9 = 20 '\024'
(gdb) p blend_h
$10 = 128
(gdb) 
```
Both core dumps shows a memory corruption in `lv_draw_sw_image`, the second crash seems to be because of a wrong clipping area (127x128px) which will end up in a wrong `blend_area` leading to a wrong `buf_stride`.

anaGrad, 2023-10-02:
Another problem is seen while drawing images with CPU only. I have the following test:

```
static void img_alpha8_cb(void)
{
    lv_style_reset(&style_common);
    /* Use a background color (without opacity) to see the transparent pixels */
    lv_style_set_bg_color(&style_common, lv_palette_main(LV_PALETTE_ORANGE));
    lv_style_set_bg_opa(&style_common, LV_OPA_COVER);
    lv_style_set_image_opa(&style_common, scene_with_opa ? LV_OPA_20 : LV_OPA_COVER);
    img_create(&style_common, &img_cogwheel_a8_128);
}
```

```
static lv_obj_t * img_create(lv_style_t * style, const void * src)
{
    lv_obj_t * obj = lv_img_create(scene_bg);
    lv_obj_remove_style_all(obj);
    lv_obj_add_style(obj, style, 0);
    lv_img_set_src(obj, src);

    lv_obj_set_pos(obj, 0, 50);
    return obj;
}
```

Upon testing, with buf align set like so:
```
/*Align the start address of draw_buf addresses to this bytes*/
#define LV_DRAW_BUF_ALIGN                       1
```
The result is the one expected, with the image at the very left limit of the display:
[image]

But when I change in lv_conf the value of buf align like so:
```
/*Align the start address of draw_buf addresses to this bytes*/
#define LV_DRAW_BUF_ALIGN                       64
```
I am left with a margin of what I think is 32 px:
[image]

kisvegabor, 2023-10-02:
Can you send the images (as C array) that you have in the mentioned issues?

anaGrad, 2023-10-03:
Sure thing, these are the two images mentioned:

[images.zip](https://github.com/lvgl/lvgl/files/12790371/images.zip)

kisvegabor, 2023-10-03:
Thank you!

> CPU Only will crash while rotating RGB 45 degree:

I've fixed it in: 2debd1d16

> I am left with a margin of what I think is 32 px:

I can't reproduce it. I have these settings (tried others as well), but the image is on the very left.
```
#define LV_DRAW_BUF_STRIDE_ALIGN                16          /*Multiple of these Bytes*/
#define LV_DRAW_BUF_ALIGN                         64
```

nicusorcitu, 2023-10-03:
Please try with
```
#define LV_DRAW_BUF_STRIDE_ALIGN             1          /*Multiple of these Bytes*/
#define LV_DRAW_BUF_ALIGN                         64
```

anaGrad, 2023-10-03:
The second issue is gone, I can't reproduce it anymore, but the crash doesn't seem to be fixed:

[image]

nicusorcitu, 2023-10-04:
I can reproduce the crash very easy (CPU only) with recolor and opa test.

kisvegabor, 2023-10-05:
This is my simplified test code, but it still works for me:
```c
  LV_IMAGE_DECLARE(img_cogwheel_rgb565_128);
  lv_obj_t * obj = lv_img_create(lv_scr_act());
  lv_obj_remove_style_all(obj);
  lv_obj_set_style_image_opa(obj, LV_OPA_20, 0);
  lv_obj_set_style_image_recolor(obj, lv_color_hex(0xff0000), 0);
  lv_obj_set_style_image_recolor_opa(obj, LV_OPA_60, 0);
  lv_img_set_src(obj, &img_cogwheel_rgb565_128);
  lv_obj_center(obj);
```

```c
#define LV_COLOR_DEPTH 16
#define LV_DRAW_BUF_STRIDE_ALIGN                32          /*Multiple of these Bytes*/
#define LV_DRAW_BUF_ALIGN                       64
```

Am I missing somerthing?

nicusorcitu, 2023-10-05:
I have the same, except the `LV_DRAW_BUF_STRIDE_ALIGN` which is 1.

kisvegabor, 2023-10-05:
Still works. Have you tried my exact code?

anaGrad, 2023-10-05:
I tried the exact code, and it still crashes, but this time, it crashes in lv_draw_buf_free(), when deallocating tmp_buf.

[image]

kisvegabor, 2023-10-05:
Normally I use the stdlib malloc as I thought with ASAN it catches all out of bound error, double frees, etc. I've tried to reproduce the issue with LVGL's built in malloc and interestingly I saw the crash. It was a meaningful lesson to me. 
The problem was the I used `lv_malloc` with `lv_draw_buf_free`. I think it hasn't caused issues with stdlib malloc because it aligned the allocated memory to 64 byte anyway.

Finally I fixed it here: a7d03e09e 

We need to make our CI work with some  != 1 STRIDE and ALIGN settings to catch this issues automatically. I'm working on it.

nicusorcitu, 2023-10-06:
I saw the `lv_malloc()` usage and then `lv_draw_buf_free()` which basically is just the `lv_free()`. I still do not understand why it crashed and why it gets fixed now.
When I checked, the pointer returned by `lv_malloc()` indeed was not aligned. But the same pointer unaligned was passed to the `lv_free()` and it crashed. Why?

kisvegabor, 2023-10-06:
I called `lv_malloc(0x40)`, let's say it returned the  `0x0104` address (so the buffer is between `0x0104`..`0x0144`). I called `lv_draw_buf_align()` which aligned the address to   `0x0110` (to 16). LVGL assumed that there is a buffer between `0x0110` and `0x150`, however it wasn't as the buffers ended at `0x0144`.

When I used `lv_draw_buf_malloc` is allocated a larger buffer to have some space for the alignment as well, so `0x0104..0x0150`.

On PC, the stdlib malloc returned e.g. `0x00123400` so the memory was aligned "by accident" and the crash remained hidden.

nicusorcitu, 2023-10-06:
> The problem was not with lv_free but with the buffer was written out of bounds and the memory descriptors was messed up, therefor free crashed.
>
This is why I did not understood why lv_free fails. I would expect a next lv_malloc() to crash.
Thanks for your answer!
@anaGrad please verify the crash with the latest code.

anaGrad, 2023-10-06:
It is fixed, no more crashes. Thank you!

kisvegabor, 2023-10-06:
Fantastic, thank you! 
I hope we can add ALIGN and STRIDE to our CI soon.