lvgl/lvgl #4611: Opacity on ANIM_FADE v8.3.10

faxe1008, 2023-09-28:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug
On v8.3.10 the commit [6548ea0f291be8a97afb3c4d7dcabbe465ae5a04](https://github.com/lvgl/lvgl/commit/6548ea0f291be8a97afb3c4d7dcabbe465ae5a04) introduced a regression regarding opacity during screen loads with ANIM_FADE.

### To Reproduce
```c
static lv_obj_t *splash_screen;
static lv_obj_t *main_screen;

static void ui_event_splash_screen(lv_event_t *e)
{
	lv_scr_load_anim(main_screen, LV_SCR_LOAD_ANIM_FADE_ON, 500, 3000, false);
}

int main(void)
{
	lv_disp_t *dispp = lv_disp_get_default();
	lv_theme_t *theme =
		lv_theme_default_init(dispp, lv_palette_main(LV_PALETTE_BLUE),
				      lv_palette_main(LV_PALETTE_RED), true, LV_FONT_DEFAULT);
	lv_disp_set_theme(dispp, theme);

	main_screen = lv_obj_create(NULL);
	lv_obj_t *ui_panelarc = lv_obj_create(main_screen);
	lv_obj_set_width(ui_panelarc, 186);
	lv_obj_set_height(ui_panelarc, 208);
	lv_obj_set_align(ui_panelarc, LV_ALIGN_CENTER);
	lv_obj_set_flex_align(ui_panelarc, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER,
			      LV_FLEX_ALIGN_CENTER);
	lv_obj_clear_flag(ui_panelarc, LV_OBJ_FLAG_SCROLLABLE); /// Flags
	lv_obj_set_style_bg_opa(ui_panelarc, 0, LV_PART_MAIN | LV_STATE_DEFAULT);
	lv_obj_set_style_border_side(ui_panelarc, LV_BORDER_SIDE_NONE,
				     LV_PART_MAIN | LV_STATE_DEFAULT);

	lv_obj_t *ui_arcselection = lv_arc_create(ui_panelarc);
	lv_obj_set_width(ui_arcselection, 150);
	lv_obj_set_height(ui_arcselection, lv_pct(88));
	lv_obj_set_align(ui_arcselection, LV_ALIGN_LEFT_MID);
	lv_obj_set_flex_align(ui_arcselection, LV_FLEX_ALIGN_CENTER, LV_FLEX_ALIGN_CENTER,
			      LV_FLEX_ALIGN_CENTER);

	splash_screen = lv_obj_create(NULL);
	lv_obj_add_event_cb(splash_screen, ui_event_splash_screen, LV_EVENT_SCREEN_LOADED, NULL);
	lv_disp_load_scr(splash_screen);

	while (1) {
		k_msleep(lv_task_handler());
	}
}

```

### Expected behavior
There should not be any white flashing during the animation.
Can be fixed by reverting a check that the commit has removed:
```
diff --git a/src/core/lv_obj_draw.c b/src/core/lv_obj_draw.c
index e0b95c523..efde0a0f1 100644
--- a/src/core/lv_obj_draw.c
+++ b/src/core/lv_obj_draw.c
@@ -180,12 +180,14 @@ void lv_obj_init_draw_rect_dsc(lv_obj_t * obj, uint32_t part, lv_draw_rect_dsc_t
     }
 #endif

-    if(opa < LV_OPA_MAX) {
-        draw_dsc->bg_opa = (opa * draw_dsc->shadow_opa) >> 8;
-        draw_dsc->bg_img_opa = (opa * draw_dsc->shadow_opa) >> 8;
-        draw_dsc->border_opa = (opa * draw_dsc->shadow_opa) >> 8;
-        draw_dsc->outline_opa = (opa * draw_dsc->shadow_opa) >> 8;
-        draw_dsc->shadow_opa = (opa * draw_dsc->shadow_opa) >> 8;
+    if(part != LV_PART_MAIN) {
+        if(opa < LV_OPA_MAX) {
+            draw_dsc->bg_opa = (opa * draw_dsc->shadow_opa) >> 8;
+            draw_dsc->bg_img_opa = (opa * draw_dsc->shadow_opa) >> 8;
+            draw_dsc->border_opa = (opa * draw_dsc->shadow_opa) >> 8;
+            draw_dsc->outline_opa = (opa * draw_dsc->shadow_opa) >> 8;
+            draw_dsc->shadow_opa = (opa * draw_dsc->shadow_opa) >> 8;
+        }
     }
 }
```

### Screenshots or video
https://github.com/lvgl/lvgl/assets/5813055/f9deba9c-a827-4118-855e-a4e510290f94

kisvegabor, 2023-10-02:
Now I see the PR, sorry :slightly_smiling_face:

faxe1008, 2023-10-02:
Will test this tomorrow, thanks for addressing it 👍

faxe1008, 2023-10-04:
Indeed the fix resolves the issue. Thanks for addressing 👍 

https://github.com/lvgl/lvgl/assets/5813055/696e0bfa-87b7-48de-b116-08b7a6f0d172

closed 2023-10-04 by faxe1008

kisvegabor, 2023-10-04:
Cool, thank you for the feedback!

commit 7a8fcbfd (2023-10-02): fix(draw): fix scaling rectangle parts with opa

fixes #4611