lvgl/lvgl #4618: Clarifying the layer usage in v9

nicusorcitu, 2023-10-02:
> For the sake of simplicity only images can be blended with non-normal blend mode. If an arc has non normal blend mode setting, LVGL will create layer, and you can blend the layer as an image with the given blend mode.
Note that, for opa_layered and blend_mode LVGL will slice the layer to LV_DRAW_SW_LAYER_SIMPLE_BUF_SIZE sized chunks to save some memory.
>
Actually this is not true. And I believe neither the approach is not correct.
1. I did a small test on v9:
```
static void sub_img_argb_cb(void)
{
    lv_style_reset(&style_common);
    lv_style_set_image_opa(&style_common, scene_with_opa ? LV_OPA_20 : LV_OPA_COVER);
    lv_style_set_blend_mode(&style_common, LV_BLEND_MODE_SUBTRACTIVE);
    img_create(&style_common, &img_cogwheel_argb8888_128, false, false);
}
```
First issue that I see, with CPU only, on this test the screen looks empty. I think CPU does not supported yet all blending modes?

Then I see 6 tasks being executed (as you mentioned the layers are slices):
- blend SRC_OVER to layer 128x48
- blend layer 128x48 SUBSTRACT to FB
- blend SRC_OVER to layer 128x48
- blend layer 128x48 SUBSTRACT to FB
- blend SRC_OVER to layer 128x32
- blend layer 128x32 SUBSTRACT to FB

So, we end up with 6 tasks when this could have been completed in just one task by the GPU without any additional allocations for layers. And this is the image blending which should support blending modes directly.

I fixed the `SUBSTRACTIVE` mode blending on VGLite, so these 6 tasks are done correctly, except that the 3 layer slices are a bit shifted at the end.
If I change the `LV_DRAW_SW_LAYER_SIMPLE_BUF_SIZE` to 64k, it will be needed a single layer and the final result is correct.

2. I checked again, and `lv_draw_line_dsc_t` have also the `blend_mode`, so blend mode can be handled directly without any use of additional layer.

3. All the other objects, rectangle, arcs can also have the blend mode on their descriptor so we can do it in a single shot. What do you think?
Thanks, Nicu.

_Originally posted by @nicusorcitu in https://github.com/lvgl/lvgl/issues/4597#issuecomment-1736863697_

kisvegabor, 2023-10-02:
My idea is to have
- `blend_mode`: Add it to all` draw_dsc`s (line, fill, image, etc) and the `draw_unit` can decide if it can draw it or not. SW will not take draw task (other than images) with non normal `blend_mode`
- `blend_mode_layered` if set, create a layer and blend the image with the set blend mode

I also don't like the idea of layers due to the larger memory consumption however in some cases they produce a different result than applying blend mode or opa widget by widget. See https://github.com/lvgl/lvgl/pull/4415#issuecomment-1676423724 for example.

Right now layers are user for 4 things:
- opa: the user can decide if they want to create a layer by using `opa_layered` style property
- blend_mode: same as a opa
- rotation and scale: it would be difficult to manage it widget by widget, but we can add a *_layered property for these as well.
- masking and post processing: the layer can be clipped, masked or modified anyhow. In VG-Lite, Can we handle e.g. clipping a widget and all its children to a rounded rectangle mask?

What do you think?

nicusorcitu, 2023-10-03:
Let me understand how this works:
1. As GPU supports `blend_mode`, it can handle the tasks directly (line, fill, image, arcs...)
BUT
If `blend_mode_layered` is set, a layer will be created no matter of 1.

Is that correct?

kisvegabor, 2023-10-04:
> Let me understand how this works:

Yes, that's how I meant. 

> BTW, a short question. The LVGL build fails all the time because there are 3 files which are twice in the library:
This is just one example:

Upps, removed here:  aed8473d1

kisvegabor, 2023-10-19:
Not stale

kisvegabor, 2023-11-03:
Not stale

nicusorcitu, 2023-11-15:
What represent this buf_area?

https://github.com/lvgl/lvgl/blob/628fb98b523d24089d339db3292a8181286b7560/src/draw/lv_draw.c#L332

1. Is the total area of the buffer (including full framebuffer or layer buffer)?
2. Is the area which is about to be written?
3. Is the area source that is about to be blended during layer task?

Basically, I have to invalidate the cache in two situations:
1. During any task processing, the destination layer buffer (only the area which is being written) need to be invalidated here:
https://github.com/lvgl/lvgl/blob/628fb98b523d24089d339db3292a8181286b7560/src/draw/nxp/vglite/lv_draw_vglite.c#L291

2. During processing layer task, I have to invalidate the source layer as well.
https://github.com/lvgl/lvgl/blob/628fb98b523d24089d339db3292a8181286b7560/src/draw/nxp/vglite/lv_draw_vglite_layer.c#L49

kisvegabor, 2023-11-15:
`buf_area` is the absolute coordinates where the layer is on the display. E.g. if it's the "main layer" of the display widgets are drawn on a smaller area. The area where the actual drawing will happen is the intersection of `task->area` and `draw_unit->clip_area`.

nicusorcitu, 2023-11-15:
OK, I see an issue while using layer buffers. Initially I thought is from the cache, but it looks the rootcause is on the layer slices mechanism.

I am running the benchmark scene `containers_with_opa_layer_cb`  with VGLite. I am using a single card. The John's picture is not blended because there is no stride in the image header - you can ignore that for now.

With default config:
```
#define LV_MEM_SIZE (256 * 1024U)          /*[bytes]*/
#define  LV_LAYER_MAX_MEMORY_USAGE             150       /*[kB]*/
#define LV_DRAW_SW_LAYER_SIMPLE_BUF_SIZE          (24 * 1024)   /*[bytes]*/
```
The output is like this:
[image]

If I change the config to not use the layer slices but a big portion of the memory:
```
#define LV_MEM_SIZE (2048 * 1024U)          /*[bytes]*/
#define  LV_LAYER_MAX_MEMORY_USAGE             1024       /*[kB]*/
#define LV_DRAW_SW_LAYER_SIMPLE_BUF_SIZE          (512 * 1024)   /*[bytes]*/
```
Then everything looks fine:
[image]

kisvegabor, 2023-11-15:
Could you try the output of this with? `lv_demo_render(LV_DEMO_RENDER_SCENE_LAYER_NORMAL, LV_OPA_COVER);` with `LV_DRAW_SW_LAYER_SIMPLE_BUF_SIZE          (2 * 1024)`

You can also enable `LV_USE_LAYER_DEBUG`. It looks like this for me:

[image]

Reds are ARGB8888 layers, the green one is RGB565 layer. You can see that the green was created from 2 parts.

Are you 100% sure that it's not a caching issue? Those thin horizontal lenes on the sides of the blue button looks like typical cache remnants.

nicusorcitu, 2023-11-15:
It looks partially wrong:

[image]

kisvegabor, 2023-11-16:
Note that in same cases `scale_x` and `scale_y` are different. It seems you have used only e.g. `scale_x` for both. In this case the result image can be out of the bounding box. See the last but one in the first row:
[image]

Or the first one in the second row:
[image]

Anyway, the simple opa test looks good:
[image]  [image]

Have you tried with full cache invalidation and/or cleaning?

nicusorcitu, 2023-11-16:
> Note that in same cases scale_x and scale_y are different.
>
Yes, I know, it was just a temp fix until doing it right.

> Have you tried with full cache invalidation and/or cleaning?
>
That was also in my mind, to do full clean and invalidate cache before every GPU run. (like in v8). I will post the updates once I have them.

nicusorcitu, 2023-11-17:
I think is a mix of issues: I used full invalidation/clear of cache before every GPU run and the output looks like this (all cards are enabled)
[image]

I will investigate the clear cache and see where do I miss one.

Wondering: how is the text rendered in case it is on layers edges? Is the clipping area used?

kisvegabor, 2023-11-18:
Do you blend the letters with VG-Lite as well? In the screenshot the letters are not aligned to the base line but to the top. E.g. "h" moved a few pixels down compared to "o". 
[image]

`lv_draw_label_iterate_letters` should set the absolute coordinates properly for `draw_letter_cb` in `glyph_draw_dsc->letter_coords`. Could you check this part?

nicusorcitu, 2023-11-20:
yes. the letters are drawn by VGLite as well.

I do not believe is only the coordinates, the letters are not only drawn to wrong coordinates. The letter looks "duplicated" on the upper side.

I think what happens is the layer slice "cuts" the letters in two parts (top and bottom). This is why I asked if the clipping area is set correctly to those letters so that the lower part to math exactly with the upper part drawn on a different layer.

nicusorcitu, 2023-11-20:
I will investigate anyway and see where the issue is.

kisvegabor, 2023-11-20:
Could you send the most recent version VG-Lite's letter drawing algorithm? Maybe I can help finding the issue.

nicusorcitu, 2023-11-20:
Yes, of course I can. But I would not like you to lose time investigating the GPU issues. I am trying to fix by myself and only signal only when there is an upstream problem.

I enabled LV_USE_LAYER_DEBUG and as I expected, the label is on the edge of the layers.
The top of the letters are drawn once to the red layer, and on the green layer it looks like the letters are not overlapped (they should be clipped) but instead they are shifted to fit to the green layer.

The question I have is: Why some layers are RGB while others are ARGB?

kisvegabor, 2023-11-20:
It's interesting, I'll check it out as well. Green (RGB) layers should be created in the middle where there are no corners. Red (ARGB) layers are created when there  are "holes", e.g. on the corners. So there shouldn't be 3 red layers at the bottom.

nicusorcitu, 2023-11-21:
1. I have it fixed the issue with label and invalidate cache area, so now the benchmark passes that step. It shall still be investigated why there are so many layers created.
2. But I run into another issue, while running the last widget test, it pass through the first round of switching "Profile" to "Analytics" and "Shop" and it goes "Profile" tab again where it get stuck.
[image]

The root cause is the wrong coordinates for `t->area` for an image to be blitted:
```
(gdb) p *t
$2 = {next = 0x80387e10 <work_mem_int+13988>, type = LV_DRAW_TASK_TYPE_IMAGE, area = {x1 = -117, y1 = 159, x2 = 42, y2 = 312}, clip_area_original = {x1 = 0, y1 = 0, x2 = 0, y2 = 0}, clip_area = {x1 = 0, y1 = 159, x2 = 42, y2 = 312}, state = 2, draw_dsc = 0x8038f91c <work_mem_int+45488>, preferred_draw_unit_id = 2 '\002', preference_score = 80 'P'}
(gdb) p *layer
$3 = {buf_unaligned = 0x0 <lv_log>, buf = 0x80000620 <s_frameBuffer>, buf_stride = 1440, buf_area = {x1 = 0, y1 = 0, x2 = 719, y2 = 1279}, color_format = 18 '\022', _clip_area = {x1 = 417, y1 = 255, x2 = 522, y2 = 300}, draw_task_head = 0x8038f974 <work_mem_int+45576>, parent = 0x0 <lv_log>, next = 0x0 <lv_log>, all_tasks_added = false, user_data = 0x0 <lv_log>}
(gdb) p *draw_unit
$4 = {next = 0x8038550c <work_mem_int+3488>, target_layer = 0x80385880 <work_mem_int+4372>, clip_area = 0x8038f99c <work_mem_int+45616>, dispatch_cb = 0x300622cd <_vglite_dispatch>, evaluate_cb = 0x300288c5 <_vglite_evaluate>, delete_cb = 0x30062359 <_vglite_delete>}
(gdb) 
```

kisvegabor, 2023-11-21:
Have you seen the LVGL logo in the top left corner of the header in the first round?

nicusorcitu, 2023-11-22:
No, I did not.

I also figured out that are a lot of assets, source images that have stride = 0. Sometimes the GPU fails, sometime did not. So I assume I could have been also a wrong accessing memory if we passed stride = 0 to the GPU (maybe it wrote some other memory areas).
Once the GPU ignores all the images without the stride, everything looks fine.

So, existing issues would be:
1. Images used in all DEMOS shall be stride aligned. We would have replace all C arrays accordingly.
2. Too many layers are created in some circumstances.

kisvegabor, 2023-11-23:
> Images used in all DEMOs should be stride aligned. We would have replace upstream all C arrays accordingly. Otherwise we can not run the DEMOs with GPU enabled.

True, we need to update those, but:
- In the `image_draw_dsc_t` -> `header` the stride should be set by LVGL if it wasn't set in the original image.
- The draw unit shall not take the images it cannot draw due to stride or memory alignment issues.

> Too many layers are created in some circumstances. I think it might be because of the extra "Connect" button - which is another rounded rectangle.

The is no layer created for rounded buttons. It created for the slices of the card in case of `opa_layered` test and for the whole card in case of the `rotation` test.  

What is the value of `LV_DRAW_SW_LAYER_SIMPLE_BUF_SIZE`?

nicusorcitu, 2023-11-23:
> In the image_draw_dsc_t -> header the stride should be set by LVGL if it wasn't set in the original image.
>
It is not enough to just set the stride in LVGL code:
`image.header.stride = lv_draw_buf_width_to_stride(width, color_format);`
It does not mean it will work, because the C array shall contain also the stride padding.

> The draw unit shall not take the images it cannot draw due to stride or memory alignment issues.
>
I do checks, and it does not take the image that does not math the stride. It wan an issue with stride=0 and I fixed it.
But I want to test the benchmark with the VGLite, so I want to see it blending all the images and see the performance. Not to just pass them to the CPU because the stride padding is not in the image source.

>What is the value of LV_DRAW_SW_LAYER_SIMPLE_BUF_SIZE?
>
24*1024

kisvegabor, 2023-11-24:
> It is not enough to just set the stride in LVGL code. If we set:
image.header.stride = lv_draw_buf_width_to_stride(width, color_format);
It does not mean it will work, because the C array shall contain also the stride padding.

You are right. Let's consider it a temporary solution while we update all the images. I'll take care of that on Monday by using 64 byte stride align for each.

XuNeo, 2023-11-25:
Ideally, we can gather all generated C-array/BIN images to a single folder, so a simple script could update them from PNGs repeatedly with different stride(parameters).

Still it's weird a decoded image has stride == 0, since it's updated to a non-zero value after decoding.

https://github.com/lvgl/lvgl/blob/3d3c0cbf86939742fab6ed1e9ec1a347ee516387/src/draw/lv_image_decoder.c#L145C9-L145C9

kisvegabor, 2023-12-11:
It should be resolved by https://github.com/lvgl/lvgl/pull/4883#issuecomment-1833030123

closed 2023-12-11 by kisvegabor