lvgl/lvgl #4622: Add OpenGL/SDL support layer [Sponsored]

kisvegabor, 2023-10-03:
I think v9 is ready to make some experiments with OpenGL/SDL again.

To keep things simpler I would like to follow a different approach than in v8. It would be great to create a generic layer (or so) which uses SW render to draw the primitives, cache them in OpenGL/SDL as textures, and just blend the cached textures. I imagine it as a simple API which can be used by OpenGL, OpenGL ES, NanoVG, SDL, Vulkan, etc.

In some cases where we are quite sure that the GPU can handle a draw task very fast we can ignore this caching mechanism. For example:
- Image drawing
- Image transformation
- Rectangle fill
- Triangle drawing

What do you think about it? 

FYI, the summary of the changes in LVGL v9 in general can be found in https://github.com/lvgl/lvgl/issues/4011 and the description of the new rendering architecture is https://github.com/lvgl/lvgl/issues/4016#issue-1599634031.

---

I would be happy give 1000 USD for developing 
- the abstraction layer 
- implement a selected engine (SDL, OpenGL, etc) as a reference
- use SW render for almost all tasks
- use the engine for a few draw tasks

cc @mariotaku

mariotaku, 2023-10-03:
I would like to use SDL renderer to implement this, and it will work similar to v8.3 as follows:

* Very basic shapes (solid rectangles without radius, bitmap etc), will be rendered by SDL directly.
* Complex elements like texts, shadows, lines and polygons will be drawn in memory by software, then uploaded as texture
* Software rendering can be done in separated threads, while uploading/rendering must be happened in the main thread

kisvegabor, 2023-10-03:
> Very basic shapes (solid rectangles without radius, bitmap etc), will be rendered by SDL directly.

It's fine. In a given `draw_task` SDL can decide whether to use SDL_fill (or similar) directly, or get the texture from the cache and blend it as  bitmap.
 
> Software rendering can be done in separated threads, while uploading/rendering must be happened in the main thread

:+1: 

---

So it sounds good, but it's very important to have a generic layer on top of which any GPU engine can be easily built. 

Mainly it means that we should have a few public functions to:
1. draw into a buffer:
   1. allocate a buffer
   2. make LVGL draw into that buffer
   3. convert the buffer into a texture
   4. save the texture into the cache
2. get the texture from the cache  

For "draw into a buffer" see [lv_canvas](https://github.com/lvgl/lvgl/blob/master/src/widgets/canvas/lv_canvas.h) and [its examples](https://github.com/lvgl/lvgl/tree/master/examples/widgets/canvas) as a reference. The canvas uses the `lv_draw` functions directly to draw into a buffer.

For "save the rendered image into the cache" we have added a generic cache manager to LVGL. See [lv_cache.c](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_cache.c) and [lv_cache_builtin.c](https://github.com/lvgl/lvgl/blob/master/src/misc/lv_cache_builtin.c). This cache is used by the image decoders too and I'm planning to use it for glyph bitmap caching as well in [Tiny_TTF](https://docs.lvgl.io/master/libs/tiny_ttf.html). If you find that a feature is missing from the current cache implementation, please let me know, and let's extend it, instead of create a new cache manager. 

For buffer allocation please use the [lv_draw_buf](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw_buf.h) module as it allows for the user to overwrite how large buffer allocation works.

Probably some parts of LVGL v9 won't be compatible with SDL as it is. For these cases please comment here to discuss how to solve those issues.

kisvegabor, 2023-10-05:
@mariotaku In order to to verify LVGL's architecture it would be very important to add these features before releasing v9  in December. So could you tell me, according to your estimate when can we have a proof of concept SDL driver?

mariotaku, 2023-10-06:
@kisvegabor I think I can add it before early December

kisvegabor, 2023-10-06:
It feel a little bit late to me. What would be required to have a proof of concept in early November?

mariotaku, 2023-10-06:
@kisvegabor I'll be poking around the code this weekend and share with you what I feel is missing and better estimation.

kisvegabor, 2023-10-06:
Than you so much! :pray: :partying_face:

mariotaku, 2023-10-09:
@kisvegabor 
I have something want to change about this task.

1. Replace `_lv_draw_buf_init_handlers` to create `SDL_Texture` instead of allocating memories
1.1. `lv_draw_buf_malloc` will need to know geometry of the buffer instead of size of bytes.
2. Make `lv_draw_sdl` to replace all the `lv_draw_sw` implementations, as the buffer handlers doesn't support writing to memory anymore
2.1. In this case, I wish `lv_draw_buf_t` could have class and different implementations, so I can draw in memory then copy it to the GPU

mariotaku, 2023-10-09:
Considering the number of changes needed I'm afraid it will take relatively long time.

kisvegabor, 2023-10-11:
Maybe I'm wrong, but this is how I imagine it:
1. We need to set a texture for in `lv_display_t` as a target buffer. Here `lv_display_set_buffers()` can be sued as it needs `void *` buffer(s).
2. During rendering LVGL uses SW rendering to get the bitmap e.g. of a rounded rectangle. Once you have the bitmap, you can create a Texture with the bitmap a cache it. So you don't need to interact with `lv_draw_buf_...` at all. It will be used only internally in SW render.
3. [lv_draw_layer_alloc_buf](https://github.com/lvgl/lvgl/blob/master/src/draw/lv_draw.c#L327C8-L327C31) is really problematic, as it uses `lv_draw_buf_alloc` to create a new layer, but we needed a Texture here. Note that it's used in special cases e.g. when a widget with all its children is rotated. In these cases LVGL creates a snapshot about the widget and blends it as an image by applying transformation. So all we need to have a Texture based version for this function, but let's solve it later when the other things are working.

So the gist is that, I think you don't have to do anything with `lv_draw_buf_...`.

mariotaku, 2023-10-14:
@kisvegabor This concept sounds good to me. Unfortunately I have made some estimations and decided that I can't make it on time, due to other upcoming events. I think I'll be pretty busy to contribute for a few months 😢 
I would happily assist someone who wants to take this over and wish they can succeed and get sponsorship!

kisvegabor, 2023-10-16:
No worries, thank you for letting me know. I have a few things to finalize in the upcoming weeks, but after the - if there no applicant -  I'll  look into it myself.

JAremko, 2023-10-28:
There are currently no plans to develop newer versions of `OpenGL` or `OpenGL ES` as they have been superseded by Vulkan. Contemporary SOCs, such as i.MX8, already support Vulkan. 
WebGL also getting replaced by WebGPU. Given the significant time investment required to implement full GPU support, it might be better to focus on these newer APIs. 🤔

JAremko, 2023-10-28:
By **full GPU support**, I mean a system that leans on GPU to create a highly efficient and dynamic GUI while freeing CPU for other tasks:

-  Create an instanced array of screen-aligned quads. Each quad would represent a GUI element, allowing efficient rendering of multiple elements. Render individual GUI elements (quads) with alpha blending, enabling them to be moved or rotated (animated) without the need to recreate their textures on CPU.  For text, glyphs, and general shapes, using [Multi-channel Signed Distance Fields (MSDF)](https://github.com/Chlumsky/msdfgen) would also allow for scaling objects without the need to recreate textures.

- Texture atlases/arrays to reduce number of texture bindings and draw calls. 

- These APIs are designed with multi-threading in mind.

kisvegabor, 2023-10-29:
Hey,

Thanks for chiming in! 

I'm not expert on SDL/OpenGL/etc, but I know LVGL very well :slightly_smiling_face: I could add an SDL texture caching in a quite straight forward way. See [here](https://github.com/lvgl/lvgl/tree/feat/sdl-render).

In summary:
- It's not finished yet. I hacked a few parts of LVGL and haven't cleaned it up yet
- It uses the software render to cache SDL textures. 
- It caches everything into a texture, even the most simple rectangle fills and images
- If there is a cache hit the texture will be retrieved and copied into the parent texture.  
- It supports LVGL layers. That is when LVGL opens a new layer (e.g. to rotate a group of widgets) an SDL texture will be created too, and when the layer is ready that texture will be merged to the parent layer/texture.

Issues/optimizations ideas:
- Replace simple fills with `SDL_FillRect`
- Render all primitives with opa = 0xff and use AlphaMod to the blend with the current opacity.
- How to handle images?
  - E.g. PNG images are also cached in LVGL. If we create a texture from a cached image it's not trivial how will we invalidate the cached texture too. 
  - Would it be slow to create texture from images in each rendering cycle? So that we could use always the currently active and valid image source?
  - Can we use SDL e.g. scale only vertically and rotate? See the last column [here](https://github.com/lvgl/lvgl/blob/master/tests/ref_imgs/draw/render/demo_render_image%20normal_opa_255.png)
  - Can we use SDL for image recoloring? See [here](https://github.com/lvgl/lvgl/blob/master/tests/ref_imgs/draw/render/demo_render_image_recolor_opa_128.png).
- How to open a OpenGL context? It should be possible to have a texture where any OpenGL thing can be rendered. I think it's more of a question to myself to how to handle it in LVGL.
- It only compares the pointer of the string, but `strcmp` should be used. In some cases (tabview, buttonmatrix, keyboard) it results in extra cache misses.
- Both `LV_DISPLAY_RENDER_MODE_FULL` (update the whole screen even if only  a singe pixel changed) and `LV_DISPLAY_RENDER_MODE_DIRECT` (update only the changed areas) are working.
- How to utilize parallel rendering?

I'm not sure which GPU APIs should we support, but I think
- SDL: it very high level and solves a lot of platform specific issues
- OpenGL ES: for MPUs
- Vulkan: for modern GPUs
- WebGPU: we can suppose that web browsers will support it

kisvegabor, 2023-11-13:
Not stale, I'll try to finalize it on the the weekend.

kisvegabor, 2023-11-19:
I've added a basic SDL backend [here](https://github.com/lvgl/lvgl/tree/master/src/draw/sdl).

It caches SW rendered bitmaps in textures.

Limitations of SDL renderer:
- not all draw task types are supported
- image transformation is not supported
- some textures (e.g. button matrix texts) are not detected as cached
- images are not cached
- images are supported only from path 

As it proves that LVGL v9 is capable of handling SDL based rendering I close this issue. As highlighted above there are many room for optimizations.

Feel free to open new PRs or issues if you are interested in contributing.

closed 2023-11-19 by kisvegabor