lvgl/lvgl #4623: Label not being drawn

Jomy10, 2023-10-04:
### Perform all steps below and tick them with [x]
- [x] Read the [FAQ](https://docs.lvgl.io/master/intro/index.html#faq)
- [x] Check the related part of the [Documentation](https://docs.lvgl.io/)
- [x] Update lvgl to the latest version (release/v8.3)
- [x] Reproduce the issue in a [Simulator](https://docs.lvgl.io/latest/en/html/get-started/pc-simulator.html)

### Describe the bug

Labels do not show on the screen after calling lv_label_create like so:
```c
lv_obj_t* label = lv_label_create(lv_scr_act());
lv_label_set_text(label, "Aikatsu!");
lv_obj_set_style_text_color(lv_scr_act(), lv_color_hex(0x00ff00), LV_PART_MAIN);
lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);
```

### To Reproduce

```c
#include <lvgl.h>
#include <stdlib.h>
#include <string.h>
#include <log.h>
#include <SDL2/SDL.h>
#include <stdio.h>
#include "sleep.h"
#include <pthread.h>

static void lvgl_logger(const char* str) {
  printf("%s\n", str);
}

#define WIDTH 800
#define HEIGHT 400
static lv_disp_draw_buf_t draw_buf;
static lv_color_t buf[WIDTH * HEIGHT / 10];

static SDL_Renderer* renderer;
static SDL_Window* window;
static bool shouldRerender = false;

void display_flush(lv_disp_drv_t* disp, const lv_area_t* area, lv_color_t* color_p) {
  int32_t x, y;
  for (y = area->y1; y <= area->y2; y++) {
    for (x = area->x1; x <= area->x2; x++) {
      SDL_SetRenderDrawColor(renderer, color_p->ch.red, color_p->ch.green, color_p->ch.blue, color_p->ch.alpha);
      SDL_RenderDrawPoint(renderer, x, y);
    }
  }
  lv_disp_flush_ready(disp);
  shouldRerender = true;
  log_info("Display flush ready");
}

void input_read(lv_indev_drv_t* indev_drv, lv_indev_data_t* data) {}

void* tick_thread (void *args) {
  log_info("Starting tick trhead...");
  while(1) {
    sleep_ms(5);
    lv_tick_inc(5);
  }
}

int main(int argc, char** argv) {
  lv_init();
  lv_disp_draw_buf_init(&draw_buf, buf, NULL, WIDTH * HEIGHT / 10);

  // init display
  static lv_disp_drv_t disp_drv; // display driver descriptor
  lv_disp_drv_init(&disp_drv);
  disp_drv.flush_cb = display_flush;
  disp_drv.draw_buf = &draw_buf;
  disp_drv.hor_res = WIDTH;
  disp_drv.ver_res = HEIGHT;
  lv_disp_drv_register(&disp_drv);

  static lv_indev_drv_t indev_drv; // input device driver descriptor
  lv_indev_drv_init(&indev_drv);
  indev_drv.type = LV_INDEV_TYPE_POINTER;
  indev_drv.read_cb = input_read;
  lv_indev_drv_register(&indev_drv);

  lv_log_register_print_cb(lvgl_logger);

  SDL_Init(SDL_INIT_VIDEO);
  SDL_CreateWindowAndRenderer(WIDTH, HEIGHT, 0, &window, &renderer);

  lv_obj_set_style_bg_color(lv_scr_act(), lv_color_hex(0x003a57), LV_PART_MAIN);
  lv_obj_t* label = lv_label_create(lv_scr_act());
  lv_label_set_text(label, "Aikatsu!");
  lv_obj_set_style_text_color(lv_scr_act(), lv_color_hex(0x00ff00), LV_PART_MAIN);
  lv_obj_align(label, LV_ALIGN_CENTER, 0, 0);
  
  pthread_t thread_id;
  pthread_create(&thread_id, NULL, tick_thread, NULL);

  SDL_Event event;
  while (true) {
    if (SDL_PollEvent(&event) && event.type == SDL_QUIT) break;

    lv_timer_handler();

    if (shouldRerender){
      SDL_RenderPresent(renderer);
      shouldRerender = false;
    }

    sleep_ms(5);
  }

  SDL_DestroyRenderer(renderer);
  SDL_DestroyWindow(window);
  SDL_Quit();

  lv_deinit();

  return 0;
}
```

### Expected behavior

Expected a label to show up on the middle of the screen.
The background fill function does work.

### Screenshots or video

Output of my program:
[image]

kisvegabor, 2023-10-06:
Here you always set the draw color to the color of the first pixel:
```c
  for (y = area->y1; y <= area->y2; y++) {
    for (x = area->x1; x <= area->x2; x++) {
      SDL_SetRenderDrawColor(renderer, color_p->ch.red, color_p->ch.green, color_p->ch.blue, color_p->ch.alpha);
      SDL_RenderDrawPoint(renderer, x, y);
    }
  }
```

I think it should be:
```c
  for (y = area->y1; y <= area->y2; y++) {
    for (x = area->x1; x <= area->x2; x++) {
      SDL_SetRenderDrawColor(renderer, color_p->ch.red, color_p->ch.green, color_p->ch.blue, color_p->ch.alpha);
      SDL_RenderDrawPoint(renderer, x, y);
      color_p++;  //Go to the next pixel
    }
  }
```

Jomy10, 2023-10-06:
Yes ofcourse. That was a dumb mistake, but one I could not figure out.
Thank you for pointing it out!

closed 2023-10-06 by Jomy10

kisvegabor, 2023-10-06:
Happy to help! 
It's hard to find these mistakes in your own code :smile: